realsenseai/librealsense #10421: Illegal Instruction when processing point cloud

calvin-bunge-ndts, 2022-04-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Windows 10  | 
| Kernel Version (Linux Only)    |    | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0                        | 
| Language                            |  C++                   | 
| Segment			|           | 

### Issue Description

When I run the following code:
`rs2::pointcloud pc;`
`rs2::points points;`
`auto frames = pipe.wait_for_frames();`
`auto depth = frames.get_depth_frame();`
`points = pc.calculate(depth);`

I get the following exception:
`Message: `
`Exception Code: C000001D`
`Stack Trace: `
`filter::process() line 354`
`pointcloud::calculate() line 433`
`myCode line xxx`

I compiled the repository using CMAKE with the following command:
`cmake .. -DBUILD_SHARED_LIBS=FALSE -DBUILD_WITH_STATIC_CRT=FALSE -DBUILD_WITH_TM2=FALSE -DCMAKE_BUILD_TYPE=Release`

Interestingly, if I run the same code on a laptop where the test user has admin permissions, there is no problem. I cannot figure out what could be causing this issue.

calvin-bunge-ndts, 2022-04-19:
Using the lib files provided by the SDK installer gives the same result.

MartyG-RealSense, 2022-04-20:
Hi @calvin-bunge-ndts  Could you first check whether access to the camera hardware and the ability of applications to access the camera are both set to On, using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789 please?

MartyG-RealSense, 2022-04-20:
Can you confirm whether you have tried installing the RealSense SDK using the **Intel.RealSense.SDK-WIN10** installer file in the 'Assets' file list on the SDK **Releases** page, please as you mentioned an installer above.

https://github.com/IntelRealSense/librealsense/releases

[image]

If you have used this installer, please try running the **rs-pointcloud.exe** pre-built example program, as it uses the **pc.calculate** method of generating a pointcloud like your own script does.  You can find this program with the following procedure:

1.  Right-click on the RealSense Viewer launch shortcut icon placed on the Windows desktop by the SDK installer and select the **Open file location** menu option.  This will open the SDK **tools** folder containing the pre-built launchable example .exe files.

2.  Find the file named **rs-pointcloud.exe** in the folder and double-leftclick on the file to launch the program.

[image]

The source code of the rs-pointcloud C++ example is here:

https://github.com/IntelRealSense/librealsense/blob/master/examples/pointcloud/rs-pointcloud.cpp

Here is the rs-pointcloud.exe example running with an L515.

[image]

calvin-bunge-ndts, 2022-04-20:
I can confirm that I installed the Windows 10 SDK. **rs-pointcloud.exe** does not work, both the version I compiled and the pre-built version installed by the sdk. A console window and an application window open for a second then close. There is no output when I start the program from the command line. Other programs in the tools folder work, like **rs-depth-quality.exe** and **rs-capture.exe**.

MartyG-RealSense, 2022-04-20:
Could you update the graphics drivers on your Windows computer please to eliminate the possibility that there is a problem with the OpenGL graphics support on the computer that is provided by the graphics drivers?

calvin-bunge-ndts, 2022-04-20:
I have the most recent graphics driver installed, however my graphics card is an AMD Radeon HD 5450, which is considered legacy hardware. What versions of OpenGL need to be supported for this library to work?

MartyG-RealSense, 2022-04-20:
As far as I am aware, the contents of the graphics driver is more important than the type of graphics hardware in regard to OpenGL support (though hardware specification makes a difference in terms of performance).  

If the latest driver for a particular model of graphics hardware has modern OpenGL support incorporated into it then that should be sufficient.  AMD Radeon is a major video brand and so should have up to date OpenGL support in its drivers.

The RealSense SDK makes use of an open-source implementation of OpenGL called **GLFW3**

https://github.com/glfw/glfw

calvin-bunge-ndts, 2022-04-20:
How modern does the OpenGL support need to be? The latest driver for my card is from 2015.

MartyG-RealSense, 2022-04-20:
If your card has not received a new driver update since 2015 then it does seem likely that any OpenGL support included in the driver could be substantially out of date.  The RealSense SDK 2.0 was first released in late 2017.

As rs-depth-quality.exe and rs-capture.exe work on your computer though, it does suggest that the SDK's graphics are working correctly despite the age of your video card though.  It is strange that code based on pc.calculate should be so specifically affected when other graphical RealSense applications are not.

calvin-bunge-ndts, 2022-04-20:
Yes, it is strange. In my initial post I mentioned that the code works on a multiple test computers where the test users have administrative permissions. I was able to get my IT administrator to run the program on my computer as an admin and the issue persists, so I can rule out permissions as the source of the problem.

MartyG-RealSense, 2022-04-20:
I will continue to look at the issue tomorrow.  Thanks very much for your patience!

MartyG-RealSense, 2022-04-21:
Do you still experience the error if you incorporate a pipeline and a pipe.start() into the script?  I note that you were not defining a **pipe** pipeline in your script but were calling the pipe reference in your wait_for_frames() line.

```
rs2::pointcloud pc;
rs2::points points;
rs2::pipeline pipe;
pipe.start();
auto frames = pipe.wait_for_frames();
auto depth = frames.get_depth_frame();
points = pc.calculate(depth);
```

calvin-bunge-ndts, 2022-04-21:
Yes I do start the pipeline before using it when I experience the issue. The pipeline object is passed into the function that creates the point cloud and is initialized in a separate function. Here is the c++ pipeline initialization:
```
rs2::config depthConfig;
depthConfig.enable_stream(RS2_STREAM_DEPTH, 320, 240, RS2_FORMAT_Z16, 30);
rs2::pipeline pipe;
pipe.start(depthConfig);
```
I have also tried using c# to collect a point cloud and got the same illegal instruction error. Here is the c# code:
```
var pipe = new Pipeline();
var pc = new Intel.RealSense.PointCloud();
var vertices = new Intel.RealSense.Math.Vertex[points.Count];

pipe.Start();

using (var frames = pipe.WaitForFrames())
using (var depth = frames.DepthFrame)
using (var points = pc.Process(depth).As<Points>())
{
    points.CopyVertices(vertices);
}
```
The line `using (var points = pc.Process(depth).As<Points>())` is the source of error in c#

MartyG-RealSense, 2022-04-21:
Your C# pointcloud code seems to be based on the SDK's **C# Cookbook** 'Generics' example code.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#generics-support

I note that the Cookbook version uses **var pc = new PointCloud();** instead of `var pc = new Intel.RealSense.PointCloud();`

Looking for a  problem and a solution in code may be an investigative wrong-direction though if the same code works fine on multiple PCs but just not on one particular PC.

calvin-bunge-ndts, 2022-04-21:
I agree.

MartyG-RealSense, 2022-04-28:
Hi @calvin-bunge-ndts  Do you require further assistance with this case, please?  Thanks!

calvin-bunge-ndts, 2022-04-28:
Yes, I would like to know why the pointcloud's calculate method causes errors on some computers and not others. If there are hardware requirements, I would like to know what those are.

calvin-bunge-ndts, 2022-04-28:
The CPU on the problem computer is an AMD Phenom II X6 1035T.

MartyG-RealSense, 2022-04-28:
I don't have CPU compatibility reports for L515.  In regard to the 400 Series cameras, there have been a few cases where they have been reported to be able to work with AMD Ryzen Threadripper CPUs in the past.  There are very few past cases where use of AMD processors has been reported though so I do not have compatibility data about non-Threadripper CPUs.

In general though, Intel and Arm processors are the focus of RealSense.

closed 2022-05-10 by MartyG-RealSense