realsenseai/librealsense #10424: Error with createOniFrame (invalid frame)

PotatoDroid5, 2022-04-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17) | 
| Platform                                 | PC |
| SDK Version                            |  { 2.5.0 }                          | 
| Language                            |  {C++ }                          | 

### Issue Description
For the context, I am currently trying to replace Asus Xtion cameras by Realsense D455 in a project. Everything is configured to run with OpenNI2. I installed librealsense from the github, the examples provided with librealsense (rs-align, rs-hello-realsense,...) work. The examples of OpenNI2 aso work well (like SimpleViewer,...) with the D455, so the realsense drivers seem to be installed correctly.

The next step was to run my C++ application with the camera D455 rather than the Xtion one, and this is the error message I'm having. I tried to search what could be the cause, but this message doesn't seem very common. Any idea about what could cause this? 

> [RS2] restartPipeline 
> [RS2] startPipeline 
> [RS2] rs2_create_config
> [RS2] rs2_config_enable_device 117122250989
> [RS2] ENABLE STREAM type=2 sensorId=1 streamId=0 1280x800 @15
> [RS2] ENABLE STREAM type=1 sensorId=0 streamId=0 848x480 @60
> [RS2] rs2_create_pipeline
> [RS2] rs2_pipeline_start_with_config
> [RS2] STARTED
> 117122250989-Unexpected color format
> [RS2] ERROR at FILE /ext_libs/librealsense/wrappers/openni2/src/Rs2Device.cpp LINE 634 FUNC createOniFrame
> 	invalid frame: rsSize=2048000 oniSize=814080

I built librealsense with this configuration:
`cmake .. -DBUILD_OPENNI2_BINDINGS:BOOL=ON`
And also tried to add `-DFORCE_LIBUVC:BOOL=ON` and `-DBUILD_WITH_CUDA:BOOL=ON`, but it didn't change anything.

MartyG-RealSense, 2022-04-20:
Hi @PotatoDroid5  **-DBUILD_WITH_CUDA:BOOL=ON** will only work with computers or computing devices (including Nvidia Jetson boards) that are equipped with an Nvidia graphics GPU.

May I ask first whether your camera is on a USB 3.2 connection, as using the camera with a USB 2.1 port will restrict the supported modes of a RealSense camera.  For example, a D455 will support 1280x720 at 15 FPS for the RGB stream when used with USB 2.1 but not 1280x800, whilst 1280x800 RGB is supported on USB 3.2.

[image]

[image]

PotatoDroid5, 2022-04-21:
Hi @MartyG-RealSense
My computer has an Nvidia GPU, so I thought it could help with the performance to set this parameter to True.

As for the USB port, I have 3.0 ports. So the problem may come from the stream dimension chosen by default?

MartyG-RealSense, 2022-04-21:
A D455 camera operating on a USB 3 connection has the defaults of 848x480 depth and 1280x720 RGB.  The default profile is applied when the pipeline is started if a custom stream configuration is not defined in a script.

closed 2022-04-22 by PotatoDroid5

MartyG-RealSense, 2022-04-22:
That's great news, @PotatoDroid5 - thanks so much for the update and for sharing the details of your solution with the RealSense community for the benefit of other OpenNI2 users!