realsenseai/librealsense #10425: D435/SR3305 question:  How do Whitebalance color temperature relates  red and blue channel gains of  OV2740 sensor

hernot, 2022-04-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 / D400 } | 
| Firmware Version                   |  | 
| Operating System & Version |     | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 |   |
| SDK Version                            |                           | 
| Language                            |                            | 
| Segment			|                            | 

### Issue Description
Have just a quick questoins Abuout how Whitebalance Color-Temperature is related to red and blue color channel gains in OV2740 color camera used on the D435 and SR300 Models. At which color temperature the gains for red and blue are approximately equal? Is three any rule of thumb how red channel and blue channel gain change when incrementing or decrementing Color-Temperature via lib real sense SDK.

MartyG-RealSense, 2022-04-21:
Hi @hernot  The default color balance value is **4600** on both the 400 Series and SR30x cameras, which is in the middle of the supported minimum-maximum white balance value range.  Increasing the white balance above the default makes colors stronger and warmer, whilst reducing it below the default progressively shifts the image towards a blue color tint.  This is illustrated in the images below, captured with a D415 camera model with an OV2740 RGB sensor.

**2800**

[image]

**5500*

[image]

The older SR30x camera models do not use the OV2740 imager for the color sensor.  It is an unnamed sensor whose details are listed below.

[image]

hernot, 2022-04-21:
Thanks. 
Strange as i head the impression that also SR300 uses the same model never the less the two have the same color temp range 2500K(2800K) to 6500K so i do assume that it has similar default balance at which blue green and red can be assumed to be sort of balanced

MartyG-RealSense, 2022-04-21:
Yes, the color behaviour is similar.  Here are the same 2800 and 5500 values on an SR300.

**2800**

[image]

**5500**

[image]

closed 2022-04-21 by hernot