realsenseai/librealsense #10426: Zoom option fort realsense SDK

ilyak93, 2022-04-21:
Is there generally an adjustable  option to control the zoom of the camera, i.e like toggling between 100% (original view) and more, like 150% for example is like x1.5 view and e.t.c ?  

It's not very critical for my work and I'm some skeptical that it exists, because its probably would need calibration (different extrinsics) to different zoom modes of the depth info and e.t.c.

MartyG-RealSense, 2022-04-21:
Hi @ilyak93  You can zoom in and out on pointclouds using the mouse scroll-wheel.  The RealSense Viewer's 3D pointcloud mode and SDK pointcloud examples for C++ (rs-pointcloud) and Python (opencv_pointcloud_viewer.py) have this function.

In the Viewer you can also toggle between windowed and fullscreen modes with the **F8** keyboard key, though whether you achieve an enlarged view may depend on the current resolution that your computer screen is displaying at.

ilyak93, 2022-04-21:
@MartyG-RealSense, I meant for something else a bit. Like for example I have another camera that I place near the realsense and I also calibrate them both for some mutual filming. But the other camera view field looks like it zoomed relatively to the realsense camera. So I wondered If there is some option in `rs2::config` or something to zoom in the realsense camera, so the rgb and depth images will be more similar to the other camera field of view (so it will be easy to calibrate the different cameras), like playing with it zooming in to different zoom levels.

MartyG-RealSense, 2022-04-21:
Is the other camera a non-RealSense camera, please?

ilyak93, 2022-04-21:
@MartyG-RealSense, exactly.

MartyG-RealSense, 2022-04-21:
So this is in relation to your recent case at https://github.com/IntelRealSense/librealsense/issues/10363 where you were aiming to align a RealSense and non-RealSense camera, yes?

This seems to be the kind of situation where an image such as Apriltag or ArUco tag, or a chessboard, could be used to calibrate the position of multiple cameras relative to each other.  An example of this is the SDK's Python example program **box_dimensioner_multicam** that auto-calibrates camera positions when the program launches, or a demo in January 2020 (in the YouTube link below) where Intel used Apriltag images to calibrate the position of nine L515 cameras relative to each other.

https://www.youtube.com/watch?v=UzIfn667abE

***********

My understanding though is that if software_device is being used then it should just be a matter of providing the intrinsics and extrinsics of both cameras, as described in https://github.com/IntelRealSense/librealsense/issues/4202

ilyak93, 2022-04-21:
@MartyG-RealSense, yes this is the case. But I wanted manually to help the calibration by making the two cameras see as much as similar scene, by manually zooming in the real sense. I already make some partial success in calibrating them and I noticed that maybe zooming in the realsense field of view could improve it.

MartyG-RealSense, 2022-04-21:
You could possibly downscale the image resolution without changing the defined stream resolution if you use the **Decimation** post-processing filter.  It automatically rescales the depth image whilst preserving the image's aspect ratio.

https://dev.intelrealsense.com/docs/post-processing-filters

closed 2022-05-04 by MartyG-RealSense