realsenseai/librealsense #10427: Timestamps may reset periodically.

kk2487, 2022-04-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version  | Ubuntu 20.04  | 
| Kernel Version (Linux Only)   | 5.13.0-39-generic                                         | 
| Platform                                 | PC |
| SDK Version                          |  V2.50.0                          | 
| Language                            |  {C/C++/Python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

I am using the ROS to publish the topic by 
`roslaunch realsense2_camera rs_camera.launch`
It's successful to publish but show some warning.
**[ WARN] [1650506621.054099370]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.**

How to fix this problem？

MartyG-RealSense, 2022-04-21:
Hi @kk2487  I note that you are using kernel 5.13.0-39-generic.  At the time of writing this, librealsense only supports a maximum of kernel 5.11 in SDK 2.50.0.  A 5.11 kernel version that has been recommended for stability by a few RealSense users is **5.11.0-34-generic**

Kernels newer than supported ones can work but there may be unpredictable consequences in regard to stability.  If your project is not able to change kernel version then you can instead build librealsense from source code with the **RSUSB backend** method.  This installation method bypasses the kernel and so is not dependent on Linux versions or kernel versions and does not require patching.

Do you still experience the warning if you set Global Time to false in your roslaunch instruction?

**roslaunch realsense2_camera rs_camera.launch global_time_enabled:=false**

kk2487, 2022-04-21:
thanks for your reply.

when I try 
`roslaunch realsense2_camera rs_camera.launch global_time_enabled:=false`
It also shows **frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.**

type c does have one direction that will fail.

I have another question.
When I use the L515 to do SLAM project with rtabmap in ROS, there are often video or IMU topic interruptions here.
Is this **cv_bridge** related?

MartyG-RealSense, 2022-04-21:
Your hardware clock warning may be due to an L515-specific RealSense ROS issue described at  https://github.com/IntelRealSense/realsense-ros/issues/2179#issuecomment-981079781

During SLAM navigation, is the camera moving or turning quickly?  This could result in point cloud matches being broken, as described in Intel's D435i SLAM guide that makes use of rtabmap_ros

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i#personalize-rviz

To quote the section linked to above:

"The built-in IMU can only keep track for a very short time.  Moving or turning too quickly will break the sequence of successful point cloud matches and will result in the system losing track.  It could happen that the system will recover immediately if stopped moving but if not, the longer the time passed since the break, the farther away it will drift from the correct position. 
 The odds for recovery get very slim, very quickly".

kk2487, 2022-04-22:
Thank you very much.
Adding 
  **<rosparam>
      /camera/motion_module/global_time_enabled: true
      /camera/l500_depth_sensor/global_time_enabled: true
      /camera/rgb_camera/global_time_enabled: true
  </rosparam>**
in my lauch file effectively solved my problem.

It turns red in the beginning when I start on rtabmap. 
after clicking reset odometry it can successfully find  feature points **just one frame**
and it will fail again at the next moment
During testing L515 was placed on the desktop without moving

MartyG-RealSense, 2022-04-22:
Would it be possible to post your RealSense ROS wrapper log from the terminal after launch has completed with the line **RealSense Node is Up!** to show any warning messages (such as **control_transfer returned**) that might be being generated after launch when you are experiencing the video and IMU topic problems?

MartyG-RealSense, 2022-04-28:
Hi @kk2487  Do you require further assistance with this case, please?  Thanks!

kk2487, 2022-04-28:
Hello, no other questions for now, thank you for your help

closed 2022-04-28 by kk2487

MartyG-RealSense, 2022-04-28:
You are very welcome, @kk2487 - thanks very much for the update!

JoaoVitorBranco, 2022-12-08:
Thanks for the answer, Marty! I will try that. If I have more doubts, I will contact again.