realsenseai/librealsense #10428: With firmware updated, L515 goes wrong after working for a while, only plugging again solves problem transiently.

weizihao30, 2022-04-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (01.05.08.01) | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  (5.4.0-107-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  {2.50.0 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
My L515 once went out of order, and the firmware has been updated with help of technical aftersales service.
However, there still appears error shown below after photographing for a while. This problem can be fixed transiently by plugging again, but it will still appear later.
[image]

MartyG-RealSense, 2022-04-21:
Hi @weizihao30  As it is a UVCIOC error, it may be a kernel related glitch, in which case you could try building librealsense with the **RSUSB backend** installation method if you have not done so already to see whether it improves stability.  An RSUSB build of librealsense bypasses the kernel and so is not dependent on Linux versions or kernel versions and does not require patching.

https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692 has instructions for building librealsense from source code with the RSUSB backend.

You could alternatively test whether the error still occurs if using a camera configuration preset other than Max Range.  For example, the **Short Range** preset will reduce the Receiver Gain.

weizihao30, 2022-04-21:
Thanks a lot for your reply. Do you mean that I try to build a librealsense manually with source code? In fact,  I use the L515 to get object's 3D coordinate through my own code. I develop my program using python and pyrealsense2. I wonder whether building a librealsense works for my program.
Besides,  could you please explain the meaning of "a camera configuration preset other than Max Range"? 

 

 
#9931 (comment) has instructions for building librealsense from source code with the RSUSB backend.

MartyG-RealSense, 2022-04-21:
Pyrealsense2 is a Python compatibility 'wrapper' that runs on the librealsense SDK.  You cannot use pyrealsense2 without librealsense first being installed.

Yes, the RSUSB Backend method involves building librealsense from source code with CMake.

The RealSense SDK has a range of **Visual Presets** that, when selected, set a specific range of values for the camera's configuration.  Each of these presets will have different effects depending on the balance of the values on particular options.

[image]

You can use one of the built-in presets shown above or create your own custom configuration.

You can download the official PDF user guide for the L515 at the link below.  Page 9 of the guide explains the differences between the built-in presets.

https://support.intelrealsense.com/hc/en-us/articles/360051646094-Intel-RealSense-LiDAR-Camera-L515-User-Guide

closed 2022-05-04 by MartyG-RealSense