realsenseai/librealsense #10431: Capturing and exporting raw file

NguyenKhanh27, 2022-04-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win10  |                           | 
| Platform                                 | PC/|
| Language                            | Python}                          | 

### Issue Description
Im newbie to IntelRealsense camera. Is there anyway to capture and save raw file (image) so that I can later use it to extract depth data. I work with Python for the project, Thanks for any helps, I appreciate!

MartyG-RealSense, 2022-04-22:
Hi @NguyenKhanh27  An example of Python code for exporting depth to .raw that was verified as correct by a RealSense team member is at https://github.com/IntelRealSense/realsense-ros/issues/1312

For C++ users there is an example provided at https://github.com/IntelRealSense/librealsense/issues/1485#issuecomment-379703868

NguyenKhanh27, 2022-04-23:
> Hi @NguyenKhanh27 An example of Python code for exporting depth to .raw that was verified as correct by a RealSense team member is at [IntelRealSense/realsense-ros#1312](https://github.com/IntelRealSense/realsense-ros/issues/1312)
> 
> For C++ users there is an example provided at [#1485 (comment)](https://github.com/IntelRealSense/librealsense/issues/1485#issuecomment-379703868)

Thanks @MartyG-RealSense for replying my problem, the mentioned topic seems not to have the way to save depth to raw, it was focused onto how to read the bag file mostly :<
From what I have searched for, the librealsense has rs_convert tool for C++ to capture and save as .raw file, Is there anyways to implement it in Python? thanks for any help, I ll would appreciate!

MartyG-RealSense, 2022-04-23:
It looks as though the Python script is reading the topic **/aligned_depth_to_color/image_raw** from a bag file and saving it as a depth image with the .raw file extension image via **cv.imwrite**.

[image]

NguyenKhanh27, 2022-04-23:
[image]
The topic owner said Open CV couldnt save .raw extension so yea, I m still try to get raw image through python, just know Librealsense has it regarding C++ only, but thanks for spending effort, Supporter <3

MartyG-RealSense, 2022-04-23:
Exporting to PNG is a much more common method of exporting RealSense depth data to an image file but yes, most of the depth information is lost when saving as a PNG, whereas the .raw format preserves the depth information.

I researched converting to .raw in OpenCV and Python with imwrite but did not find any useful leads either or other methods in librealsense for .raw export.  

An alternate approach in Python to preserve the depth information during export may be to save it as an **.npy** file, which can then be imported into other Python applications such as Pytorch.  A librealsense Python script for doing so is at https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225

NguyenKhanh27, 2022-04-27:
> Exporting to PNG is a much more common method of exporting RealSense depth data to an image file but yes, most of the depth information is lost when saving as a PNG, whereas the .raw format preserves the depth information.
> 
> I researched converting to .raw in OpenCV and Python with imwrite but did not find any useful leads either or other methods in librealsense for .raw export.
> 
> An alternate approach in Python to preserve the depth information during export may be to save it as an **.npy** file, which can then be imported into other Python applications such as PyTorch. A librealsense Python script for doing so is at [#4934 (comment)](https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225)

Yes I could do it, but just further question, has there been any post relating to process the NPY file for depth information? If there is, please mention for it. Thank you sm!

MartyG-RealSense, 2022-04-28:
https://github.com/gdlg/panoramic-depth-estimation/issues/3 - whilst not a librealsense case - advises loading an .npy file containing depth data with the numpy instruction **np.load** to retrieve the depth values.

https://numpy.org/doc/stable/reference/generated/numpy.load.html

NguyenKhanh27, 2022-04-29:
> [gdlg/panoramic-depth-estimation#3](https://github.com/gdlg/panoramic-depth-estimation/issues/3) - whilst not a librealsense case - advises loading an .npy file containing depth data with the numpy instruction **np.load** to retrieve the depth values.
> 
> https://numpy.org/doc/stable/reference/generated/numpy.load.html

Im not yet found the way to work with npy file since it save a 3x3 matrices for each pixel and it seems to save 848x160 instead of 848x480, but thank you so much for the support !

MartyG-RealSense, 2022-05-04:
Hi @NguyenKhanh27  Do you require further assistance with this case, please?  Thanks!

NguyenKhanh27, 2022-05-04:
> Hi @NguyenKhanh27 Do you require further assistance with this case, please? Thanks!

Hi, Yes I'm done with the case and thanks for the help! Topic closed

closed 2022-05-04 by NguyenKhanh27

MartyG-RealSense, 2022-05-04:
You are very welcome.  Thanks for the update!

ZeroHeading, 2022-05-26:
> Yes I could do it, but just further question, has there been any post relating to process the NPY file for depth information? If there is, please mention for it. Thank you sm!
hi, would please tell me how to convert .bag file into .png in python? thanks very much!

NguyenKhanh27, 2022-05-26:
> Exporting to PNG is a much more common method of exporting RealSense depth data to an image file but yes, most of the depth information is lost when saving as a PNG, whereas the .raw format [...]

I have used numpy load for that,
```
sample1 = np.load('C:\\Users\\Khanh27\\PycharmProjects\\SeniorCamTest\\Orange\\Orange22\\Raw\\Orange22_4.npy')
d=sample1[y][x] #d would be the depth value
```
And when I save raw value to .npy file, I save depth_image only, not depth_color_image:
`frame = pipeLine.wait_for_frames()
 depth_frame = frame.get_depth_frame()
depth_image = np.asanyarray(depth_frame.get_data())`

I didnt work with .bag file tho so yea homie x.x