realsenseai/librealsense #10433: Announcement: RealSense Depth Module D401 + Vision Processor D4 Board Bundle (10 Pack) in stock at the RealSense Store

MartyG-RealSense, 2022-04-22:
Hi everyone,

The official online RealSense Store is now stocking a bundle containing 10 each of the **RealSense Depth Module D401** (the camera board inside the RealSense D405 camera model) and the **Vision Processor D4 Board V4** 

https://store.intelrealsense.com/intelr-realsensetm-depth-module-d401-vision-processor-d4-board-bundle-10-pack.html

closed 2022-04-25 by MartyG-RealSense

michael-nc, 2022-05-05:
Hey @MartyG-RealSense, sorry for the double posting but we are kind of in a rush for this. So, I am interested in buying the D401 + Vision Processor D4 Board bundle. It seems like I would need some sort of ribbon cable to connect them. On the Intel store page, it seems like none of the ribbon cables are in stock, do you have any idea when they might be back in stock? If not soon, is there an alternative way to purchasing these ribbon cables or is there some other generic ribbon cables I can use?

MartyG-RealSense, 2022-05-06:
No problem at all, @michael-nc - I posted a reply at your other question.

https://support.intelrealsense.com/hc/en-us/community/posts/5717080982931/comments/6036176471187