realsenseai/librealsense #10436: How to create your own independent project that uses opencv (apt-get), librealsense (build from source), openvino (build from source)

cassini-fly, 2022-04-23:
Hi guys,

Thanks for this great repo! lots of useful stuff...
I was able to build and run openvino examples (dnn and face) with a realsense camera, but I want to know more about how to create a new C++ project's cmake. Note that I installed openvino and librealsense both from source (since I needed to containerise and could not find/execute any easier/better way).

Thanks!
### Issue Description
 Let's say I have the following headers that I need to use:
 
```
#include <iostream>
#include <librealsense2/rs.hpp>     // Include RealSense Cross Platform API
#include "cv-helpers.hpp"         // frame_to_mat
#include <opencv2/core/utils/filesystem.hpp>   // glob

#include <rs-vino/object-detection.h>
#include <rs-vino/detected-object.h>

#include <easylogging++.h>
#ifdef BUILD_SHARED_LIBS
// With static linkage, ELPP is initialized by librealsense, so doing it here will
// create errors. When we're using the shared .so/.dll, the two are separate and we have
// to initialize ours if we want to use the APIs!
INITIALIZE_EASYLOGGINGPP
#endif

#include <rs-vino/openvino-helpers.h>

#include <chrono>

```
I tried looking at dnn example in openvino ([this](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/openvino/dnn/CMakeLists.txt)).But there are multiple cmakelists in previous directories: librealsense, wrapper, openvino and are linked vertically.  Is there a proper way to do this - How to create your own independent project that uses opencv, librealsense, openvino?
In other words, If create a new project with a new cmake and try to copy the needed parts from the cmake mentioned above, I will run into relative path issue. What's the proper way of doing this?

Note. I can build and run all of the examples in librealsense while i have librealsense and openvino built from source.

MartyG-RealSense, 2022-04-23:
Hi @cassini-fly  The link below has a DNN project that makes use of librealsense, CMake, OpenVINO and OpenCV and has very detailed installation instructions.

https://github.com/PINTO0309/MobileNet-SSD-RealSense

The project is based around use of the **Intel Neural Compute Stick (NCS)**, a device that plugs into a computer's USB port alongside the RealSense camera to accelerate vision computing applications.

Of the 9 examples provided on the page linked to above, Example 9 does not require the Neural Compute Stick though.

[image]

cassini-fly, 2022-04-25:
@MartyG-RealSense thanks for the reply. I looked for a cmakelist.txt in the repository you linked, haven't found any. Am I looking in the right place?

MartyG-RealSense, 2022-04-26:
I couldn't find a CMakeLists.txt file or work out how to build the program either, unfortunately.  The tutorial also sems to be centered around using a Raspberry Pi board as the computing device.

closed 2022-05-10 by MartyG-RealSense