realsenseai/librealsense #10439: Release to Ubuntu 22.4 / jammy

doisyg, 2022-04-23:
Now that ubuntu 22.04 is out, would it be possible to get updated binaries or if they are the same to update the debian repo ?

MartyG-RealSense, 2022-04-23:
Hi @doisyg  The librealsense SDK currently officially supports up to Ubuntu 20.04 (Focal) at the time of writing this.  As 21.04 (Hirsuite) support is not yet added, I cannot provide an estimate for when support for 22.04 (Jammy) and Debian packages for it may be added to the SDK.

You should be able to attempt to build the SDK from source code with CMake for Jammy immediately though if the SDK is built with the **RSUSB backend** method, as that type of librealsense build is not dependent on Linux versions or kernel versions and does not require patching

MartyG-RealSense, 2022-05-04:
Hi @doisyg  Do you require further assistance with this case, please?  Thanks!

doisyg, 2022-05-04:
A roadmap for a proper release to the current Ubuntu LTS (22.04) would be a minimum. I believe there are very few adaptation to make for the tooling executables to work (i.e realsense-viewer). The library itself works just fine by forcing the installation of the focal deb:
[image]

dheera, 2022-05-06:
> Intel do not provide schedules and roadmaps in advance for possible supported RealSense features in the future, however.

Why not?

MartyG-RealSense, 2022-05-06:
I do understand your frustration but it has been the long-standing policy since the 400 Series cameras were launched at the end of 2017 that specific release dates and roadmaps for updates are not published.  Instead, updates (new features, bug fixes, API changes, firmwares, etc) are announced in the release notes of new RealSense SDK versions when they arrive.  SDK version releases are also not announced in advance.

MartyG-RealSense, 2022-05-19:
Case closed due to no further comments received.

doisyg, 2022-05-19:
It doesn't make sense to close this issue until realsense is properly released to Ubuntu 22.04. And this is the good place to communicate about the progress, schedule, beta and blocking points I believe.

MartyG-RealSense, 2022-05-19:
A closed issue is still findable in a web search so that other RealSense users can comment on the subject if they wish to.  As there is not a time estimate available for implementation of official Jammy support into librealsense, keeping the case open would not be useful.  It is a subject that is likely to be returned to in future anyway as Jammy use becomes more widespread in the RealSense community.

doisyg, 2022-05-19:
Well, if it is going to be released for Jammy, this issue should be left open until it is. There is no problem in leaving issues open for a long time, it is actually good practice, signaling to the community that it is being addressed, for instance: https://github.com/ros-planning/navigation2/issues/1391
Closing the issue, on the opposite, maintain the doubt that it will never be done. And if that's the case, I need to plan and organize (I have more than 300 D435 in production to maintain), and I guess it is the same for many other people using D400 cameras in production. 
The realsense support improved compared to the early days of R200s but its history is a subject of stress for many products relying on it (I am sure @SteveMacenski can relate) and your response (and the overall realsense communication)  is not at all reassuring. To the point that when starting a new project, many people (including myself) are looking for alternatives.

MartyG-RealSense, 2022-05-19:
Unfortunately I cannot offer guarantees for possible future features.  If you require certainty in your planning then I would recommend assuming that a feature is not on its way until a time when it actually is announced, which will typically not be in advance but instead when a new librealsense SDK release arrives.  My understanding is that Q3 2022 is the current target for the next SDK release.

As mentioned earlier, Intel does not release development roadmaps for RealSense in the style of the quoted ros-planning one.

For stable long-term support I would recommend only using SDK and firmware pairings that Intel has categorized as **Production** (suitable for critical applications and commercial products). The most recent Production configuration at the time of writing this is SDK 2.50.0 and firmware 5.13.0.50.

https://dev.intelrealsense.com/docs/recommended-production-camera-configurations

SteveMacenski, 2022-05-19:
To @doisyg ’s comment, that attitude is incredibly problematic and why most companies are looking to Realsense alternatives even though they dont want to.  Hardware is great, software support is some of the worst I’ve ever run into…

smorillon, 2022-05-20:
This problem is not solved but closed!?
Can you reopen this issue so that people don't look for solutions in this thread when there are none?

MartyG-RealSense, 2022-05-20:
There is no information that I can provide about future Jammy support.

However, as there is a demonstrated demand for future information expressed in this discussion by more than one RealSense user, as a compromise I will re-open it and label the case as **Enhancement** to signify that it should be kept open for a long-term period.

davetcoleman, 2022-06-05:
Hi, I represent a company very well established in the ROS 2 community (PickNik). We've worked with 70+ companies, [including Intel](https://twitter.com/picknikrobotics/status/1080866094074814464), helping them best use ROS with their robots, and we frequently recommend and sell Realsense cameras into our client projects. However we are switching to Jammy and a lack of driver support is a pretty big deal for us.

> that attitude is incredibly problematic and why most companies are looking to Realsense alternatives even though they dont want to. Hardware is great, software support is some of the worst I’ve ever run into… it should take a day to release for 22.04 but Intel really can’t spend a day on it?

I generally agree with @SteveMacenski's sentiment here.

Nir-Az, 2022-06-29:
Ubuntu 22.04 `librealsense` Debians & ROS2 Humble RS camera node support are on the future roadmap.
Stay tuned...

pepisg, 2022-06-30:
Hi. I wrote to Intel Realsense's sales team to ask for a timeline on the release (the company I work for relies on realsense cameras for the main product as well). He told me the release was planned for December 31st, but that they would try their best to push it forward. Sadly it seems that it will still be a long time :(

doisyg, 2022-06-30:
> Ubuntu 22.04 `librealsense` Debians & ROS2 Humble RS camera node support are on the future roadmap. Stay tuned...

This i reassuring and then

> He told me the release was planned for December 31st

this is alarming

There is almost nothing to do. As described above, it is already working on Humble/Jammy by forcing the install of the Focal packets (at least for the D435s). Just add Jammy to the deb repo.
What's missing is maybe the kernel patch for the metadata, but for many of us it is not needed (and it was already the case for the lastest hwe kernel in Focal).
And a recompilation of the realsense-viewer GUI.
If the timeline is really 31st of December, I am going to end up doing it and creating an alternate deb repo.

I don't understand how complex stacks like Moveit 2 or nav2 are ready very soon after the Humble release, and something as trivial as this is taking months.

mengyui, 2022-07-01:
There is a draft kernel patch in the PR https://github.com/IntelRealSense/librealsense/pull/10625
It should enable metadata on Ubuntu using kernel 5.13/5.15.

BTW, I have a pre-built package for testing in [here](https://github.com/mengyui/librealsense2-dkms/releases).

aagrawa3-dev, 2022-07-09:
Hi Realsense team,
We strive to migrate to Ros2-humble (which is based on Ubuntu2204).

### relasense SDK 
As suggested, we tried compiling relasense SDK on U2204. 
https://github.com/IntelRealSense/librealsense.git
It is compiled.

### realsense_viewer:
Also works but we see this error(in red box) which does not go away even after applying udev rules from librelasense repo & rebooting machine. 
**Please suggest solution for it:**
[image]

### ros2-wrapper:
The ros2-wrapper for realsense seems not working when compiled on ros2-humble.
Its compiled but a small patch in makefile was needed. 
**Is it expected and can this patch be released officially?**

```
diff --git a/realsense2_camera/CMakeLists.txt b/realsense2_camera/CMakeLists.txt
index e55490d..90cfbf3 100644
--- a/realsense2_camera/CMakeLists.txt
+++ b/realsense2_camera/CMakeLists.txt
@@ -146,6 +146,10 @@ elseif("$ENV{ROS_DISTRO}" STREQUAL "rolling")
   message(STATUS "Build for ROS2 Rolling")
   set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -DROLLING")
   set(SOURCES "${SOURCES}" src/ros_param_backend_foxy.cpp)
+elseif("$ENV{ROS_DISTRO}" STREQUAL "humble")
+  message(STATUS "Build for ROS2 Humble")
+  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -DHUMBLE")
+  set(SOURCES "${SOURCES}" src/ros_param_backend_foxy.cpp)
 else()
   message(FATAL_ERROR "Unsupported ROS Distribution: " "$ENV{ROS_DISTRO}")
 endif()
```

On launch of the node, we get below error:
```
$ ros2 launch realsense2_camera rs_launch.py 
[INFO] [launch]: All log files can be found below /home/eiforamr/.ros/ros_log/2022-07-09-10-58-09-036872-demohost-109
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [110]
[realsense2_camera_node-1] [INFO] [1657357089.768480259] [camera.camera]: RealSense ROS v4.0.4
[realsense2_camera_node-1] [INFO] [1657357089.768578837] [camera.camera]: Built with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1657357089.768600116] [camera.camera]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1657357089.842162413] [camera.camera]: Device with serial number 044322073299 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1657357089.842220304] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1657357089.842235453] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1657357089.842389614] [camera.camera]: Device with port number 2-3 was found.
[realsense2_camera_node-1] [INFO] [1657357089.842407956] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1657357089.848333463] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1657357089.849051888] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1657357089.850315853] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1657357089.851342552] [camera.camera]: Device Serial No: 044322073299
[realsense2_camera_node-1] [INFO] [1657357089.851644524] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1657357089.852256396] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1657357089.852528712] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1657357089.852777950] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[realsense2_camera_node-1]   what():  parameter 'depth_module.auto_gain_limit_toggle' has invalid type: cannot undeclare an statically typed parameter
[ERROR] [realsense2_camera_node-1]: process has died [pid 110, exit code -6, cmd '/home/eiforamr/ros2_ws/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_umzve47l'].
```
**Do you know how to solve this issue?**

MartyG-RealSense, 2022-07-09:
Thanks very much @aagrawa3-dev for your makefile patch for the beta wrapper!  I have passed the details to the ROS2 wrapper development team.

The launch error that you reported was previously experienced by a RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/issues/2245 and they suggested setting a **dynamic_typing** flag as a fix.

In regard to missing udev rules, please try installing them with the instructions below.

```
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger
```

dadaroce, 2022-07-12:
Hello there, using the dkms prebuild provided by @MartyG-RealSense and some tricky configurations, we was able to use the relasense-ros package (Ubuntu 22.04 and ROS2 Humble).
```
RUN apt-get update && apt-get install --no-install-recommends -y \
    libssl-dev  \
    libgl1-mesa-dev \
    libgtk-3-dev \
    libglfw3-dev \
    libglu1-mesa-dev \
    libusb-1.0-0-dev \
    pkg-config \
    python3 \
    python3-dev \
    python3-distutils \
    dkms \
    --reinstall \
    # Install realsense DKMS
    && wget -nc https://github.com/mengyui/librealsense2-dkms/releases/download/initial-support-for-kernel-5.15/librealsense2-dkms-dkms_1.3.14_amd64.deb \
    && dpkg -i librealsense2-dkms-dkms_1.3.14_amd64.deb \
    && rm librealsense2-dkms-dkms_1.3.14_amd64.deb \
    && wget -nc 'http://keyserver.ubuntu.com/pks/lookup?op=get&search=0xF6E65AC044F831AC80A06380C8B3A55A6F3EFCDE' -O librealsense.key \
    && gpg --no-default-keyring --keyring ./librealsense_keyring.gpg --import librealsense.key \
    && gpg --no-default-keyring --keyring ./librealsense_keyring.gpg --export > librealsense.gpg \
    && mv librealsense.gpg /etc/apt/trusted.gpg.d/ \
    && rm *.gpg \
    && cp /etc/apt/sources.list /etc/apt/sources.list.bkp \
    && echo "deb https://librealsense.intel.com/Debian/apt-repo focal main" >> /etc/apt/sources.list \
    # To install libssl1.1, and another dependencies
    && echo "deb http://archive.ubuntu.com/ubuntu focal universe" >> /etc/apt/sources.list \
    && echo "deb http://archive.ubuntu.com/ubuntu focal-updates universe" >> /etc/apt/sources.list \
    && echo "deb http://archive.ubuntu.com/ubuntu focal main restricted" >> /etc/apt/sources.list \
    && echo "deb http://archive.ubuntu.com/ubuntu focal-updates main restricted" >> /etc/apt/sources.list \
    && echo "deb http://archive.ubuntu.com/ubuntu focal multiverse" >> /etc/apt/sources.list \
    && echo "deb http://archive.ubuntu.com/ubuntu focal-updates multiverse" >> /etc/apt/sources.list \
    && apt update && apt-get install -y \ 
    librealsense2-utils \
    librealsense2-dev \
    librealsense2-udev-rules \
    # Restore the Ubuntu Repositories Version
    && rm /etc/apt/sources.list \
    && mv /etc/apt/sources.list.bkp /etc/apt/sources.list \
    # ------------------------------
    && apt autoremove && apt clean -y \ 
    && rm -rf /var/lib/apt/lists/*
```

MartyG-RealSense, 2022-07-12:
Thanks so much @dadaroce for sharing the news of your success and the details of your procedure!

Nir-Az, 2022-07-17:
**Update:** Building from source the `RS camera ROS2 wrapper` now support Humble distro (See PR: https://github.com/IntelRealSense/realsense-ros/pull/2406)

Kernel patched for 5.15 is not included, so the support is currently without full metadata.

Still on the roadmap:
* Jammy Debians
* Kernel patches for 5.15

aagrawa3-dev, 2022-07-17:
> **Update:** Building from source the `RS camera ROS2 wrapper` now support Humble distro (See PR: [IntelRealSense/realsense-ros#2406](https://github.com/IntelRealSense/realsense-ros/pull/2406))
> 
> Kernel patched for 5.15 is not included, so the support is currently without full metadata.
> 
> Still on the roadmap:
> 
>     * Jammy Debians
> 
>     * Kernel patches for 5.15

Hi @Nir-Az  Thanks for info. But this patch is merged to branch "[IntelRealSense:ros2-beta](https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta)" !!

Is this branch going to be main branch for humble upgrade?  When is it planned to merge to development branch?

Nir-Az, 2022-07-18:
Hi @aagrawa3-dev ,

I know it might be confusing but I will try to clarify the `realsense-ros` repo branches, 
We develop/maintain ROS1 + ROS2 on this repo, on different branches.
If you notice on the `development` branch [readme](https://github.com/IntelRealSense/realsense-ros/blob/development/README.md), it said that this is the ROS1 branch.
```
This version supports Kinetic, Melodic and Noetic distributions.

For running in ROS2 environment please switch to the [ros2 branch](https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta).
```

The ROS2 branch is [`ros2-beta`](https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta)
Meaning only `ros2-beta` branch can support ROS2-Humble as ROS1 latest support is Noetic distro.

Hope it helps..

bikcrum, 2022-07-26:
Try to build from source.

```
sudo apt install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev libusb-1.0-0-dev
mkdir -p ~/repos && cd ~/repos
git clone https://github.com/IntelRealSense/librealsense
mkdir -p librealsense/build && cd librealsense/build
cmake .. -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release
make -j$(nproc)
sudo make install
```

AustinDeric, 2022-09-04:
I am unable to compile it on 22.04.  here is a way to reproduce my error: 

Dockerfile: 
```
FROM ubuntu:22.04
RUN apt update && apt install -y libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev libusb-1.0-0-dev git build-essential cmake libssl-dev
RUN mkdir -p ~/repos && cd ~/repos
RUN git clone https://github.com/IntelRealSense/librealsense
RUN mkdir -p librealsense/build
WORKDIR /librealsense/build
RUN cmake ..
RUN make &&  make install
```

error:
```
#12 778.6 [ 35%] Building CXX object CMakeFiles/realsense2.dir/src/libusb/context-libusb.cpp.o
#12 778.7 In file included from /librealsense/src/libusb/context-libusb.h:10,
#12 778.7                  from /librealsense/src/libusb/context-libusb.cpp:4:
#12 778.7 /librealsense/src/libusb/libusb.h:10:82: error: extended character ‘ is not valid in an identifier
#12 778.7    10 | /usr/include/libusb-1.0/libusb.h:736:4: warning: ISO C++ forbids zero-size array ‘dev_capability_data’ [-Wpedantic]
#12 778.7       |                                                                                  ^
#12 778.7 /librealsense/src/libusb/libusb.h:10:82: error: extended character ’ is not valid in an identifier
#12 778.7 /librealsense/src/libusb/libusb.h:13:82: error: extended character ‘ is not valid in an identifier
#12 778.7    13 | /usr/include/libusb-1.0/libusb.h:767:4: warning: ISO C++ forbids zero-size array ‘dev_capability’ [-Wpedantic]
#12 778.7       |                                                                                  ^
#12 778.7 /librealsense/src/libusb/libusb.h:13:82: error: extended character ’ is not valid in an identifier
#12 778.7 /librealsense/src/libusb/libusb.h:18:83: error: extended character ‘ is not valid in an identifier
#12 778.7    18 | /usr/include/libusb-1.0/libusb.h:1260:4: warning: ISO C++ forbids zero-size array ‘iso_packet_desc’ [-Wpedantic]
#12 778.7       |                                                                                   ^
#12 778.7 /librealsense/src/libusb/libusb.h:18:83: error: extended character ’ is not valid in an identifier
#12 780.2 make[2]: *** [CMakeFiles/realsense2.dir/build.make:1280: CMakeFiles/realsense2.dir/src/libusb/context-libusb.cpp.o] Error 1
#12 780.2 make[1]: *** [CMakeFiles/Makefile2:916: CMakeFiles/realsense2.dir/all] Error 2
#12 780.2 make: *** [Makefile:136: all] Error 2
```

Nir-Az, 2022-09-05:
Hi @AustinDeric ,

This error is raised when building with GCC >= 10
Thanks to @chris-durand a [fix PR](https://github.com/IntelRealSense/librealsense/pull/10818) was merged into development branch yesterday (@MartyG-RealSense  FYI). 
To get it, retry running the docker image but before adjust your docker file to use development branch

Change this:
```
RUN git clone https://github.com/IntelRealSense/librealsense
```
to this:
```
RUN git clone -b development https://github.com/IntelRealSense/librealsense
```

Good luck!

MartyG-RealSense, 2022-09-07:
Thanks so much @GuintherKovalski for sharing!  :)

dHofmeister, 2022-10-14:
Dear @GuintherKovalski, thank you for your contribution. 

I'm running into an issue with reproducing the dockerfile as you have added a wildcard` linux-headers-* ` I doubt this is intended? Could you specify which ones we need? Adding all of them costs: 

```
After this operation, 67,0 GB of additional disk space will be used.
Do you want to continue? [Y/n] 
```

> RUN apt-get update && apt-get install -y \ 
>     sudo \ 
>     cmake \
>     git \
>     curl \
>     libopencv-dev \ 
>     libjpeg-dev \ 
>     libpng-dev \ 
>     libtiff-dev \  
>     libgtk2.0-dev \ 
>     python-numpy \ 
>     python-pycurl \ 
>     keyboard-configuration \
>     libgl1-mesa-dev \
>     libglu1-mesa-dev \
>     libusb-1.0-0-dev \
>     libglfw3-dev \
>     freeglut3 \
>     at \
>     freeglut3-dev \
>     nano \
>     curl \
>     libglfw3-dev \
>     libgl1-mesa-dev \
>     libglu1-mesa-dev \
>     at \ 
>     udev \
>     linux-headers-* \
>     libusb-1.0-0-dev 
> ```

dHofmeister, 2022-10-14:
I built a different version. 
It compiles ROS2 + Ubuntu 22 + Realsense and it's working. 

Source files / Reference files: 
https://github.com/mengyui/librealsense2-dkms/commit/15562dc0a501a4aa67e087f579c69a0ecc85f621
https://github.com/mengyui/librealsense2-dkms/releases/tag/initial-support-for-kernel-5.15

```
FROM ros:humble-ros-core-jammy

# https://github.com/mengyui/librealsense2-dkms/commit/15562dc0a501a4aa67e087f579c69a0ecc85f621
# https://github.com/mengyui/librealsense2-dkms/releases/tag/initial-support-for-kernel-5.15

ARG DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get install -y --no-install-recommends \
    ca-certificates \
    wget \
    dkms \
    git \
    libusb-1.0-0-dev \
    libudev-dev \
    libglfw3-dev \
    pkg-config \
    libgtk-3-dev \
    at \
    unzip \
    libpopt0 \
    rsync \
    libglu1 \
    libglu1-mesa \
    python3-colcon-common-extensions \
    python3-rosdep \
    build-essential \
    nano \
    usbutils \
    && rm -rf /var/lib/apt/lists/*

RUN mkdir -p /invisible/src
WORKDIR /invisible

COPY realsense /invisible/src/realsense

RUN cd /invisible/src/realsense && \
    dpkg -i librealsense2-dkms-dkms_1.3.14_amd64.deb && \
    dpkg -i libssl1.1_1.1.1f-1ubuntu2.16_amd64.deb && \
    dpkg -i librealsense2-udev-rules_2.50.0-0~realsense0.6128_amd64.deb && \
    dpkg -i librealsense2_2.50.0-0~realsense0.6128_amd64.deb && \
    dpkg -i librealsense2-gl_2.50.0-0~realsense0.6128_amd64.deb && \
    dpkg -i librealsense2-net_2.50.0-0~realsense0.6128_amd64.deb && \
    dpkg -i librealsense2-utils_2.50.0-0~realsense0.6128_amd64.deb && \
    dpkg -i librealsense2-dev_2.50.0-0~realsense0.6128_amd64.deb

RUN rosdep init && \
    rosdep update

RUN apt-get update  && \
    rosdep install -i --from-path src --rosdistro $ROS_DISTRO --skip-keys=librealsense2 -y

RUN . /opt/ros/$ROS_DISTRO/setup.sh && colcon build

CMD ["sleep", "infinity"]
```
 
There is one `COPY realsense /invisible/src/realsense` argument. 

This folder contains a raw copy-paste of the github repository: 
`git clone https://github.com/IntelRealSense/realsense-ros.git -b ros2-beta`

and all the unpacked .deb files from `https://github.com/mengyui/librealsense2-dkms/releases/tag/initial-support-for-kernel-5.15`

I had to go this route with cloning the repo because there is a compile-time error. The github repo searches for: 

```
find_package(realsense2 2.51.1)
if(NOT realsense2_FOUND)
    message(FATAL_ERROR "\n\n Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases\n\n")
endif()
```
There is a versioning mismatch somewhere which I couldnt trace. But removing the release version requirement makes it build and run. 

```
find_package(realsense2)
if(NOT realsense2_FOUND)
    message(FATAL_ERROR "\n\n Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases\n\n")
endif()
```
cheers

Benjamin-Tan, 2022-10-17:
Hi, is there any plan or work in progress that try to resolve the build error of arm64 build in ros humble? https://build.ros2.org/job/Hsrc_uJ__librealsense2__ubuntu_jammy__source/

Nir-Az, 2022-10-17:
Hi @Benjamin-Tan,
We have released an initial release to ROS2 Humble.

This release builds as expected at all x86 environments foxy/galactic/humble and also works at arm64 for foxy & galactic.
Somehow the arm64 build at the buildfarm fails.
We are trying for quiet some time now to fix it and we still havn't found why it happens.
We did check building it on an arm64 machine and it works.
Something about the Humble arm64 buildfarm cause it to fail and we still don't know what.

If you have some idea's about what is special about that build against foxy/galactic builds please share 😀

Benjamin-Tan, 2022-10-17:
Hi @Nir-Az, thanks for the information and i just manage to see your ROS Discourse [post](https://discourse.ros.org/t/humble-build-farm-job-on-arm-fails-on-first-release/27413/1). 

Hi @tfoote, thanks for the help last time on the advice for humble release for pinocchio package. Would you happen to know what might cause this issue of librealsense failing on Humble arm64 build farm as detailed in the ROS Discourse post?

tfoote, 2022-10-18:
Likely there's a library with different content between the two builds. I would look on a working platform where you can find the missing symbols (find the appropriate `.so` file and then see if the symbols are in the libraries on the humble arm64 builds or if they are present see why those libraries aren't being linked on that build.

Nir-Az, 2022-10-18:
The library that the librealsense2 link complains about is our internal library called `fw` / `fwlib`.
We build on several OSs and several platform and never saw this issue.
I tried running the build farm locally (It took a whole day.. why does it compile with -j1??)
and it reproduced.. but when the build fails the container is automatically removed..
We will keep investigating it..

justinyue1643, 2022-10-24:
I wasn't sure if this is related. When trying to the add repository, I get this error:
[image]

I'm following the installation instructions from [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md).

MartyG-RealSense, 2022-10-24:
Hi @justinyue1643  The librealsense SDK does not yet officially support Ubuntu 22.04 Jammy.  An unofficial test patch and pre-built package is available at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106

sheaffej, 2022-10-25:
@dHofmeister thank you very much for sharing your Dockerfile in https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1279130578

It worked for me also, and I was able to build the image on an Intel NUC8 running Ubuntu 18.04 on the host, and Docker server 20.10.20 (Note: I had to upgrade Docker to 20.10.x to solve some apt-get update errors with Ubuntu 22.04).

I also converted your `COPY` statement into the steps you describe, adding a `sed` to modify the CMakeLists.txt file as you describe. This worked well for me!

```
# COPY realsense /invisible/src/realsense
RUN cd /invisible/src/realsense && \
    git clone https://github.com/IntelRealSense/realsense-ros.git -b ros2-beta && \
    wget https://github.com/mengyui/librealsense2-dkms/releases/download/initial-support-for-kernel-5.15/librealsense2-dkms-dkms_1.3.14_amd64.deb && \
    wget https://github.com/mengyui/librealsense2-dkms/releases/download/initial-support-for-kernel-5.15/libssl1.1_1.1.1f-1ubuntu2.16_amd64.deb && \
    wget https://github.com/mengyui/librealsense2-dkms/releases/download/initial-support-for-kernel-5.15/PREBUILT-DEB-PACKAGE-librealsense2_2.50.0-0.realsense0.6128_amd64.zip && \
    unzip PREBUILT-DEB-PACKAGE-librealsense2_2.50.0-0.realsense0.6128_amd64.zip && \
    rm -f PREBUILT-DEB-PACKAGE-librealsense2_2.50.0-0.realsense0.6128_amd64.zip && \
    cd realsense-ros/realsense2_camera && \
    sed -i 's/find_package(realsense2 2.51.1)/find_package(realsense2)/' CMakeLists.txt
```

Thank you again!

tfoote, 2022-11-23:
I found this reference where link time optimization support was added to dpkg-buildpackage: https://wiki.debian.org/ToolChain/LTO  "Starting with dpkg 1.21.0, dpkg-buildflags introduced a new optimization area, which adds the necessary flags to enable LTO."

And OpenSUSE enabled it by default in 2019: https://www.phoronix.com/news/GCC-9-LTO-Tumbleweed-Firefox I can't find an announcement of Debian doing the same, but I assume they followed suit. Ubuntu's statement here is that it's enabled for basically all architectures now. https://wiki.ubuntu.com/ToolChain/LTO

You could override the LTO settings of the build by patching the debian rules file template in the release repository. But the cleaning and stripping unreferenced content is generally helpful to avoid system bloat. These sorts of optimizations have been great at allowing our number of packages to grow, but the installation size to shrink so I'd strongly prefer to find a way to reference them properly so they don't get stripped rather than just disable the cleanup.

Nir-Az, 2022-11-23:
@tfoote thanks,
We currently overcome it by forcing no-lto or the problematic library build.
IMO LTO is wrong on his decision for cropping this functions as they are in use.
I confirm that Ubuntu 22 Debian build added the lto flag and not the build farm (it was hard to catch that)

I will look at the patch you suggested if we have another issue with this library cause we cannot spend this amount of time investigating issues that do not interrupting the native build..

Thanks

MartyG-RealSense, 2022-12-07:
Hi everyone,

Version **2.53.1** of the librealsense SDK has been released, featuring support for Ubuntu 2020 kernels 5.13 and 5.15, and Ubuntu 2022 with kernel 5.15.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1

The SDK installation instruction pages have been updated to reflect this new support.

**Package build**
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

**Source code build**
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

tonynajjar, 2022-12-07:
> Hi everyone,
>
> Version **2.53.1** of the librealsense SDK has been released, featuring support for Ubuntu 2020 kernels 5.13 and 5.15, and Ubuntu 2022 with kernel 5.15. [...]

Thanks for the update @MartyG-RealSense, where can I find realsense firmware 5.14.0.0? It's not [here](https://dev.intelrealsense.com/docs/firmware-releases)

MartyG-RealSense, 2022-12-08:
Hi again @tonynajjar  Firmware 5.14.0.0 is now available on the firmware archive page.  Thanks very much for your patience!

https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

MartyG-RealSense, 2022-12-19:
As support for Ubuntu 2020 kernels 5.13 and 5.15, and Ubuntu 2022 with kernel 5.15 was released in SDK **2.53.1**, does anyone require further support with this issue please?  Thanks!

MartyG-RealSense, 2022-12-30:
Case closed due to kernel 5.13 and 5.15 support implemented in SDK version 2.53.1 and no further comments received after 2 weeks.

closed 2022-12-30 by MartyG-RealSense