realsenseai/librealsense #10440: How to Download intel RealSense sdk in macOS?

isharizh, 2022-04-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2022-04-24:
Hi @isharizh  If you are able to use Brew then the **brew install librealsense** installation method should provide an easy way to install the RealSense SDK on both older Intel-based Macs and modern Apple Silicon M1 models.

https://formulae.brew.sh/formula/librealsense

If you wish to build the SDK from source code then the official MacOS installation instructions can be found at the link below.  These instructions are best suited for Intel-based Mac models rather than M1.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

A highly detailed and up to date MacOS installation guide created by a RealSense user that covers MacOS Monterey and Apple Silicon M1 has also been published.

https://lightbuzz.com/realsense-macos/

MartyG-RealSense, 2022-05-10:
Case closed due to no further comments received.

closed 2022-05-10 by MartyG-RealSense

kwihl, 2022-06-13:
> Hi @isharizh If you are able to use Brew then the **brew install librealsense** installation method should provide an easy way to install the RealSense SDK on both older Intel-based Macs and modern Apple Silicon M1 models.
> 
> https://formulae.brew.sh/formula/librealsense
> 
> If you wish to build the SDK from source code then the official MacOS installation instructions can be found at the link below. These instructions are best suited for Intel-based Mac models rather than M1.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md
> 
> Also, a highly detailed and up to date MacOS installation guide has been published by a RealSense user that covers MacOS Monterey and Apple Silicon M1 as well as older MacOS versions such as Big Sur and Intel-based Mac models.
> 
> https://lightbuzz.com/realsense-macos/

Hello Marty (sorry if I'm breaking etiquette by commenting on a closed issue), I'm in a similar situation where I've successfully managed to follow Vangos Pterneas excellent guide to build the realsense & libusb binaries, but now I'm completely stuck. I'm on MacOS monterey 12.4, and I have no clue where to start looking for info on how to, let's say, run the example projects in the realsense2 repo in a simple non-IDE setup (vs code).

I'm currently bogged down with linker errors and cmake errors, and all the documentation I can find seems to assume that you've installed via homebrew and the compiler just knows how to locate the package etc (the example project itself for instance just uses an include statement with no absolute/relative path, and the cmakelists doesn't reference any binary). I'd be super grateful if you know _any_ resources to point me to on how to proceed - this might really just be down to my inexperience with C++ and CMake.

kwihl, 2022-06-15:
Hello again @MartyG-RealSense . Thank you for your help, however this project (getting started) is also entirely adapted to a windows environment. I've been trying to get this to work on my M1 mac machine for 2 weeks now and at the end I just gave up and switched to an old Windows one, and then it took 5 minutes. It would be really helpful if there were any examples on how to actually _use_ the binaries on MacOS rather than just build them, otherwise the plethora of surrounding information isn't really helpful. To me it seems weird that the repo releases are even marked as mac compatible, considering how utterly mysterious the installation process seems, unless you're on Catalina which is 5 years old.

Regardless, thank you very much for pointing me in a good direction. I might try again to build it on mac at a later stage and if I have any success I'll make sure to document the process.

MartyG-RealSense, 2022-06-15:
Whilst the RealSense SDK has support for MacOS, official development support for Mac stopped at MacOS High Sierra, though the SDK unofficially continues to work with newer MacOS versions.  The RealSense Mac community is self-supporting and the best solutions always come from Mac-owning community members supporting each other.

If you are searching for the location of the SDK's built binary versions of examples or tools on MacOS then you could try searching for the program filename using one of the methods suggested at the link below.

https://osxdaily.com/2021/06/21/how-find-files-mac/

As a general rule, you can find the filename of an example program by left-clicking on its folder name at the link below.  For example, motion is rs-motion, capture is rs-capture, hello-realsense is rs-hello-realsense, etc.  The launch filename of the RealSense Viewer is **realsense-viewer** and the Depth Quality Tool is **rs-depth-quality**

https://github.com/IntelRealSense/librealsense/tree/master/examples