realsenseai/librealsense #10442: Input/output error

kogan-alex, 2022-04-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Linux | 
| Platform                                 | PC |

### Issue Description
Unclear error:
[image]
What might be the cause?
Please note that we replaced the camera and still get an error.

MartyG-RealSense, 2022-04-25:
Hi @kogan-alex  Which Linux kernel version are you using, please?

kogan-alex, 2022-04-25:
5.4.0-109-generic

kogan-alex, 2022-04-25:
There are 4 other cameras that connect successfully on this pc

MartyG-RealSense, 2022-04-25:
On your Linux version of the Viewer, do you have a **Calibration Data** menu option under the 'More' option in the Viewer side-panel that you could use to reset the affected camera's calibration to factory defaults with the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487