realsenseai/librealsense #10444: D405 Disparity Shift

michael-nc, 2022-04-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win (10) | 
| Platform                                 | PC |
| SDK Version                            |  2.50 | 
| Language                            |  python | 

Currently with the D405 camera, I am interested in further decreasing the minimum depth distance below 7cm. Going off of ([https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-move-as-close-as-possible]), it seems like it is possible to further decrease the minimum depth by adjusting the disparity shift parameter. 

Is there a similar figure for D405 like the following: 
[image]

that I can use?

Thanks!

MartyG-RealSense, 2022-04-25:
Hi @michael-nc  I ran depth sensing tests with the D405 model at distances lower than 7 cm, with and without disparity shift.  I found that using disparity shift with D405 provided no benefit and the depth image began progressively breaking up below 7 cm as the camera was moved closer to the observed object whether disparity shift was used or not.  The ideal depth range of the D405 is 7 cm to 50 cm.

There is not an equivalent  of the quoted disparity chart for RealSense camera models other than D415.

michael-nc, 2022-04-25:
Hi @MartyG-RealSense thanks for the quick response. Does that mean the only way to reduce the minZ distance is to reduce the resolution? Is there an alternative?

MartyG-RealSense, 2022-04-25:
Reducing the resolution of the image can normally reduce minimum distance of a 400 Series camera as a simplistic alternative to disparity shift.  

Reducing resolution did provide some small improvement progressively as the resolution lowered.

**848x480**

[image]

**640x480**

[image]

**480x270**

[image]

*******

In regard to alternatives, I am not currently aware of another method for reducing minimum distance below 7 cm without depth image break-up.

MartyG-RealSense, 2022-05-04:
Hi @michael-nc  Do you require further assistance with this case, please?  Thanks!

hakunaMatataHub, 2022-05-05:
@MartyG-RealSense 
In this [link](https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-move-as-close-as-possible)
it says to lower the MinZ we can reduce the resolution of the camera, but wouldn't that compromise the accuracy of the depth estimation. Isn't increasing the disparity shift always the best hing to do?
Also what is the disparity range always 125 pixels in the D435i or any other real sense cameras?

P.S: excuse me if these questions are very fundamental, I am just trying to understand its working better so I use it with best setting for my use case

michael-nc, 2022-05-05:
Hi @MartyG-RealSense , I do not need further assistance with this case. Thank you so much for your quick reply!

closed 2022-05-05 by michael-nc

MartyG-RealSense, 2022-10-11:
Hi @michael-nc  Whilst the D405 model could work with SDK version 2.50.0, **official** support for D405 was introduced in SDK 2.51.1, and a RealSense user at https://github.com/IntelRealSense/librealsense/issues/10963#issuecomment-1274941411 found that disparity shift worked in this SDK version with the now-official support for D405.

michael-nc, 2022-10-11:
Hi @MartyG-RealSense Awesome, thanks for such quick updates! That is super helpful!