realsenseai/librealsense #10448: E: Unable to locate package librealsense2-utils E: Unable to locate package librealsense2-dev in Docker container Jetson

sisaha9, 2022-04-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | N/A | 
| Operating System & Version |  Ubuntu 18.04 (but Docker container running 20.04)  | \                                     | 
| Platform                                 |  NVIDIA Jetson |
| SDK Version                            | Binary                          | 
| Language                            |   N/A                          | 

### Issue Description
```
ARG BASE_IMAGE=arm64v8/ros 
ARG ROS_DISTRO=galactic-ros-base-focal

FROM ${BASE_IMAGE}:${ROS_DISTRO}
LABEL maintainer="Siddharth Saha <omadityasiddharth123@gmail.com>"

RUN apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
RUN apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
RUN apt update && apt install -y --no-install-recommends \
    apt-utils \
    software-properties-common

RUN add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
RUN apt update && apt install -y --no-install-recommends \
    librealsense2-utils \
    librealsense2-dev
```
Running this Docker container results in
```
E: Unable to locate package librealsense2-utils
E: Unable to locate package librealsense2-dev
```

MartyG-RealSense, 2022-04-27:
Hi @sisaha9  There are currently not Arm64 librealsense packages available for 20.04 Focal, though there are for 18.04, as described at https://github.com/IntelRealSense/librealsense/issues/10220#issuecomment-1030919806

Can you confirm whether you are attempting to install the packages on 18.04 or 20.04 please?

sisaha9, 2022-04-27:
The container is on 20.04 but I am building on a 18.04 system @MartyG-RealSense

MartyG-RealSense, 2022-04-27:
Is there any difference if you use **apt-get install** as Intel's official Jetson package installation instructions do instead of 'apt install'

MartyG-RealSense, 2022-04-27:
A potential solution to this error that was discussed with a Docker user at https://github.com/IntelRealSense/realsense-ros/issues/1925#issuecomment-864542325 was to build the librealsense SDK and the RealSense ROS wrapper together at the same time from packages using the ROS wrapper's **Method 1** procedure.

The RealSense user was concerned that this approach may be impractical for Docker due to a risk of a previous installation of librealsense automatically being overwritten by a newer SDK version when it was released.  I suggested that creating a custom GitHub **fork** of the SDK's GitHub may help to exert control over version changes.

MartyG-RealSense, 2022-05-10:
Case closed due to no further comments received.

closed 2022-05-10 by MartyG-RealSense

maxiuw, 2022-11-08:
I follow this and it seems to be working https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md