realsenseai/librealsense #10449: Realsense-Camera Application CPU load rises by 20% if application user is not in administrator-group

Tichy-Ijon, 2022-04-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 (D435i)  | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  -              | 
| Platform                                 | PC |
| SDK Version                            |  2.35.2. Build with default build-settings for Windows10.            | 
| Language                            |  C/C++                          | 
| Segment			|  Robot/machinery               | 

### Additional libs:
OpenCV
Protobuf
Boost beast
omp.h: omp_set_num_threads(1); // using just one CPU for the camera-application.

### Issue Description
We have developed a (Realsense-) camera application for Windows10. It works well in our system
if the application user is assigned or not assigned to the administrator-group.
Our customer uses also Windows10 but with additional security settings. This is a black-box for us.
 
If they running the camera application (with application user isn't inside the administrator-group) the application-CPU load rises by 20%.
(user not in admin-group is the normal use-case in the customer's system.)
 
We have also additional application running on that system which are developed in java and also in c/c++. All these
application uses multi-threading, USB ports and sockets. There are no problems with these application.
All applications have the scheduling priority normal.
 
Investigation:
Running Process-Explorer features a CPU rise for ntdll.dll!RtReleaseSRWLockExclusive+0x2200.
(see image)
[image]

Modifications to encircle the problem:
- Realsense lib w/o debug information.
- unset/set HWM_OVER_XU,FORCE_RSUSB_BACKEND.
- User/Groups manager (lusrmgr.msc). (Set/unset the application user to additional groups.)
- group policies (gpedit.msc) (Set/unset local group policies.)
- changing registry permission for application-user.
- changing camera firmware to 5.13.0.50
all without effect.
 
As mentioned, we don't have this issue on our developing Windows PCs or laptops.
 
The major question: Why do we have different CPU load if an application user (normal user) is assigned or not assigned to the administrator-group.
I don't think that this is an Intel Realsense issue but we have that problem in conjunction with the Realsense API and/or our additional libs.
Maybe the group has similar experience.
 
Does somebody have experience with that behavior?
I know, it is difficult or probably not possible to give a solution for that problem because
we can't describe the difference between our and the customer's system.
 
Every help will be appreciate.
Thanks in advance

MartyG-RealSense, 2022-04-27:
Hi @Tichy-Ijon  I have not seen a past case similar to the problem that you are experiencing.

I will first highlight some potential issues with your build and camera configuration.

Firmware 5.13.0.50 would not be suitable for use with librealsense SDK 2.35.2.  This is because from firmware 5.12.8.200 onwards, the minimum SDK version that should be used is **2.39.0**.  This is due to internal changes in the firmware from 5.12.8.200 onwards.  If you need to use SDK 2.35.2 then I would recommend using a maximum of firmware **5.12.7.100**, the version before the internal changes in the firmware were implemented.

It looks as though you are building the SDK for Windows from source code.  If you are then I would recommend not doing so if possible and instead installing using the quick and easy automated installer program **Intel.RealSense-SDK.WIN10** that is provided in the 'Assets' file list for each SDK version on the SDK 'Releases' page.

https://github.com/IntelRealSense/librealsense/releases

[image]

If the SDK is built from source code then I would recommend setting the build flag **CMAKE_BUILD_TYPE** to 'Release' so that the SDK will be built with optimizations, as the default setting is to build the SDK in 'Debug' mode which may have slower performance.

The default of a CMake build of the SDK is also to use **Microsoft Media Foundation** as its backend (a Linux build uses 'V4L backend' by default.  If **FORCE_RSUSB_BACKEND** is set to True then the SDK will use a backend based on a WinUSB rewrite of the UVC protocol instead.  

To the best of my knowledge there have been a very small number of times during the last several years since the launch of the 400 Series cameras that FORCE_RSUSB_BACKEND has been used with a source code build of the SDK on Windows, so I would recommend not setting FORCE_RSUSB_BACKEND to True and let the SDK use the default Windows Media Foundation backend instead.  As mentioned above though, I would also recommend installing the SDK with the automated installer file instead of CMake if at all possible.

Tichy-Ijon, 2022-04-28:
Hi @MartyG-RealSense,
thanks for the fast answer and the information i.e. max. firmware version for 2.35.2.
I will try the sdk-installation but I think the problem is anchored somewhere in user/group/registries checks or persmission setups. Maybe we can keep that issue open for some days.
I hope somebody comes up with some windows specifics. (but I fear not.).

MartyG-RealSense, 2022-04-28:
Yes, it is fine to keep this case open for a further time period.  Good luck with the SDK installation!

closed 2022-05-21 by MartyG-RealSense