realsenseai/librealsense #10450: Error booting T265

JingyiWang, 2022-04-28:
## An error is reported every time when T265 is started. Need to replug it.

[2022-04-26 00:39:10 /t265/realsense2_camera_manager]
26/04 08:39:10,419 ERROR [139656184891136] (handle-libusb.h:100) failed to claim usb interface: 0, error: RS2_USB_STATU
S_BUSY
[2022-04-26 00:39:10 /t265/realsense2_camera_manager] Device 1/1 failed with exception: failed to set power state
[2022-04-26 00:39:10/t265/realsense2_camera_manager] The requested device with device name containing t265 is NOT found
Will Try again.
26/04 08:39:10,419 ERROR [139656218461952] (sensor.cpp:572) acquire_power failed: failed to set power state
26/04 08:39:10,420 WARNING [139656218461952] (rs.cpp:310) null pointer passed for argument "device"
26/04 08:39:13,398 WARNING [139851843634944] (messenger-libusb.cpp:65) bulk_transfer returned error, endpoint: Ox1, err
or: Resource temporarily unavailable, err. num: 11
26/04 08:39:13,398 ERROR [139851843634944] (tm-boot.h:39) Error booting T265
26/04 08:39:17,441 ERROR [139851843634944] (handle-libusb.h:100) failed to claim usb interface: O, error: RS2_USB_STATL
S_BUSY
26/04 08:39:17, 445 WARNING [139656218461952] (messenger-libusb.cpp:65) bulk_transfer returned error, endpoint: 0x1, err
or: Resource temporarily unavailable, err. num: 11
26/04 08:39:17, 441 ERROR [139851843634944] (tm-boot.h:42) Failed to open T265 zero interface
26/04 08:39:17, 445 ERROR [139656218461952] (tm-boot.h:39) Error booting T265

NemesisX09, 2022-12-19:
Did you find any solutions?

julyfire-wang, 2023-06-21:
I met the same question. Did you have any solutions?

NemesisX09, 2023-06-21:
> I met the same question. Did you have any solutions?

From what I can understand, the OS that I was working on gave all the control of the usb hub to an android app and it wasn't reaching the native code. I had to update the permissions to give the native code priority and it seemed to fix the issue.

julyfire-wang, 2023-06-25:
> From what I can understand, the OS that I was working on gave all the control of the usb hub to an android app and it wasn't reaching the native code. I had to update the permissions to give the [...]

Thank you! I test D435i is OK, just T265 occured this error.

closed 2026-10-23 by MartyG-RealSense