realsenseai/librealsense #10451: Cannot access https://egiintel.jfrog.io/ui/repos/tree/General/librealsense

II-Matto, 2022-04-28:
I am trying to use librealsense to develop an Android app, and followed the instructions of the [Android support](https://github.com/IntelRealSense/librealsense/blob/master/doc/android.md) and the [Java example](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/java_example/readme.md). It seems that I need to download the lib from `https://egiintel.jfrog.io/ui/repos/tree/General/librealsense`. But this link requires a login and I found it nowhere to register an account. How could I obtain the *.so and *.aar files? Many thanks.

MartyG-RealSense, 2022-04-28:
Hi @II-Matto  You should be able to build the RealSense Android wrapper and its AAR with Gradle or with Android Studio using the instructions at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android

II-Matto, 2022-04-28:
@MartyG-RealSense Thanks for your prompt reply. Does this mean that I have to build the AAR myself and cannot directly download one?

MartyG-RealSense, 2022-04-28:
At the moment the only option seems to be to be to build the AAR.

I will raise the issue of the JFrog login with Intel.  Thanks very much for the report!

II-Matto, 2022-04-28:
@MartyG-RealSense Many thanks!

closed 2022-04-28 by II-Matto