realsenseai/librealsense #10453: How big is the error in measuring the distance in water with d435 camera? Has anyone experimented?

afenghandsome, 2022-04-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2022-04-28:
Hi @afenghandsome  There is an extensively detailed discussion on underwater calibration to account for distortion from the water at https://github.com/IntelRealSense/realsense-ros/issues/1723

Because of the length of the discussion, some of it with important details is hidden, so please visit the comment at https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-802741234 and click on the **Load more** link beneath if to expand open the full discussion, where you can also find an image of the underwater custom housing used in that particular project.

[image]

afenghandsome, 2022-04-28:
Thank you for your reply. I'll take a good look at these cases first

closed 2022-05-11 by MartyG-RealSense