realsenseai/librealsense #10454: Question regarding Intel realsense Viewer and Depth units

MartinPedersenpp, 2022-04-28:
Hello

I have googled to try to find an answer to this, but with no luck, so I hope that someone here can give me the answer.

I am using pyrealsense2 for my project and Intel Realsense Viewer to preview setting changes.

I wanted to experiment with depth units, but when I change the depth units from the default 1000um to 500um I don't see a visible difference. However, when I do it in python, all metrics get doubled. I noticed that if i start the depth stream in realsense viewer and then change the depth units, the view metrics will get doubled.
So my question is: Does realsense viewer have some sort of compensation for the depth units, when you change it before starting the stream?

Best regards
Martin

MartyG-RealSense, 2022-04-28:
Hi @MartinPedersenpp  The way that the graphical user interface (GUI) in the C++-based RealSense Viewer program alters a setting may not be exactly the same as changing the setting by using code in a self-created script, though the functions should behave very similarly once a change in a setting is successfully applied by the script's instructions.  

The Viewer could therefore be regarded as a way to test out the effects of different functions on the image output and prototype a configuration before working out how to implement that functionality in your own script.

MartinPedersenpp, 2022-04-28:
Thanks for the response @MartyG-RealSense 
Maybe I need to specify:
In realsense viewer. If I turn on the depth stream and cut my depth unit in half, 60cm will become 120cm.
If i cut it in half when the stream is turned off and then turn it on, the distance is 60cm.
When I cut it in half in python, the distance is 120cm.
I hope this makes my question clearer.

MartyG-RealSense, 2022-04-28:
I understood your meaning.  In my response, I meant that how a function works in the Viewer's user interface may not necessarily be **exactly** the same as how that function is controlled in Python, so you need to work out the programming that produces the desired outcome.

A key example of the difference between Python and the Viewer is that the Viewer applies a range of preset configuration values, color shading settings and post-processing filters by default, whilst in Python those features are not present by default and have to be deliberately programmed into a script in order to produce an image that resembles the Viewer's.

I would not recommend changing the depth units by this method anyway, but instead changing the Depth Unit Scale under the 'Controls' category of the Stereo Module in the Viewer side-panel that is 0.001 by default.

MartinPedersenpp, 2022-04-29:
@MartyG-RealSense
Okay I think I understand it now, I will try to check out the unit scale.

closed 2022-05-05 by MartinPedersenpp