realsenseai/librealsense #10455: Issue: High CPU usage when the camera is disconnected after starting the pipeline

Ahmadgafer, 2022-04-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |   Linux (Ubuntu 20)  | 
| Kernel Version (Linux Only)    |  5.13.0-40-generic                                      | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0                          | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description

While running the following code to start the pipeline if I disconnect the camera cable the cpu usage goes up to 100% on three threads. 

I checked the documentation there's no way to check when the camera gets disconnected so that I can rest the program.

Any ideas on how to prevent this from happening and detect when the camera is disconnected?  

```
import pyrealsense2 as rs
import time

pipeline = rs.pipeline()
config = rs.config()
profile = pipeline.start(config)

while True:
    # do things
    time.sleep(1)
```

MartyG-RealSense, 2022-04-28:
Hi @Ahmadgafer  you could try using **set_devices_changed_callback** to check for camera connection / disconnection events.  A Python script for doing so is at https://github.com/IntelRealSense/librealsense/issues/4212#issuecomment-586262375

When disconnection is detected, you could then trigger a **hardware_reset** instruction to reset the camera automatically.

In regard to the high CPU usage when disconnecting, does this occur when using the introductory Python script from the Python wrapper documentation page at the link below?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#streaming-using-rspipeline

MartyG-RealSense, 2022-04-28:
Could you monitor memory usage whilst your program is running to see whether the high CPU usage conincides with a reduction in available computer memory capacity, please?  If there was a rapid decline in available memory then it could indicate that there is a **memory leak** occurring that leads to degraded performance and even program freeze / exit.

Ahmadgafer, 2022-04-28:
Sure! Here's more info:

**Before disconnecting the usb:**  
process memory usage: 0.5% 
process cpu usage: 0.5% 
system available memory: 22.2 GB

**After disconnecting the usb:**
process memory usage: 0.5% 
process cpu usage:  12.5% cpu (2 cores out of 16 usage goes up to 100%) , 
system available memory 22.2 GB

This is on my PC. When it happens on a less powerful PC (NUC) it causes problems of high temperature as it heats up.

MartyG-RealSense, 2022-04-28:
Are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice, please?

Ahmadgafer, 2022-04-28:
I'm using the short cable that comes with the box. I don't think the problem is related to the cable as the CPU usage goes up when the cable is disconnected.

MartyG-RealSense, 2022-04-28:
I note that you are using kernel 5.13.0-40-generic.  The RealSense SDK currently supports up to kernel **5.11** when building it from source code with CMake and **5.4** when installing from Debian packages.  Whilst newer kernels can work with the SDK, there may be unpredictable consequences in regard to stability.

If you are unable to use a kernel other than 5.13.0-40-generic then you can build the SDK from source code with the **RSUSB backend**.  This installation method bypasses the kernel, so the SDK is not dependent on Linux versions of kernel versions and does not require patching.

Ahmadgafer, 2022-04-29:
I will need until next week to try the build from source option or find a PC with 5.4 kernel. Will keep you updated. Thanks!

MartyG-RealSense, 2022-04-29:
That's no problem at all.  Good luck!

closed 2022-05-22 by MartyG-RealSense