realsenseai/librealsense #10458: Working with RealSense bag file recordings using rosbag instead of pyrealsense2 package

JeffR1992, 2022-05-01:
I'm trying to use the `rosbag` Python package instead of RealSense's `pyrealsense2` library on bag files recorded with a D435i camera. To begin with, I'm trying to simply show with recorded depth image from the `/device_0/sensor_0/Depth_0/image/data` topic. However, if I try the following code:

```
import rosbag
import cv2

bag_file_path = "/path/to/bag/file"
topic = "/device_0/sensor_0/Depth_0/image/data"

bag = rosbag.Bag(bag_file_path)

for (tpc, msg, tmstmp) in bag.read_messages(topics=topic):
    cv2.imshow("Depth Image", msg.data)
```

I get the following error:

Traceback (most recent call last):
  File "c:\Users\jeffr\Desktop\realsense_code\rosbag_test.py", line 111, in <module>
    cv2.imshow("Depth Image", msg.data)
cv2.error: OpenCV(4.5.5) :-1: error: (-5:Bad argument) in function 'imshow'
> Overload resolution failed:
>  - mat is not a numpy array, neither a scalar
>  - Expected Ptr<cv::cuda::GpuMat> for argument 'mat'
>  - Expected Ptr<cv::UMat> for argument 'mat'

I've also tried `cv2.imshow("Depth Image", np.asanyarray(msg.data)), to no avail. Furthermore, if I use `print(dir(msg))` I see the following:

`['__class__', '__delattr__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__getstate__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__setstate__', '__sizeof__', '__slots__', '__str__', '__subclasshook__', '_check_types', '_connection_header', '_full_text', '_get_types', '_has_header', '_md5sum', '_slot_types', '_spec', '_type', 'data', 'deserialize', 'deserialize_numpy', 'encoding', 'header', 'height', 'is_bigendian', 'serialize', 'serialize_numpy', 'step', 'width']`

In particular, I see that there is a `deserialize_numpy` attribute. However, if I try `cv2.imshow("Depth Image", msg.deserialize_numpy) I still get the same error. I'm wondering what I'm missing. Any help would be appreciated

MartyG-RealSense, 2022-05-01:
Hi @JeffR1992   If you are trying to display a rosbag message with cv2, you may first need to convert the ROS message to an OpenCV image with**cv_bridge**

http://wiki.ros.org/cv_bridge/Tutorials/ConvertingBetweenROSImagesAndOpenCVImagesPython#Converting_ROS_image_messages_to_OpenCV_images

JeffR1992, 2022-05-02:
Thanks for this link @MartyG-RealSense I'll give it a try. However, is there a way to convert `msg.data` into something more general like a numpy array without having to use `cv_bridge`? For example, when using the `pyrealsense2` package, I can simply do:

```
config = rs.config()
config.enable_device_from_file(file_path)
pipeline = rs.pipeline()
pipeline.start(config)
frames = pipeline.wait_for_frames()
depth_frame = frames.get_depth_frame()
depth_image = np.asanyarray(depth_frame.get_data())
```

to get the `depth_image` as a numpy array. Do you know if `depth_frame.get_data()` use `cv_bridge` to do this?

MartyG-RealSense, 2022-05-02:
A tutorial at the link below has a Python example of using a bag file to convert msg.data to a numpy array under the section headed **Processing All Images in a Bag**

http://ronny.rest/blog/post_2017_03_30_ros2/

The script is quoted below.

```
import numpy as np
import rosbag
bag = rosbag.Bag(bag_file, "r")
messages = bag.read_messages(topics=["/image_raw"])
num_images = bag.get_message_count(topic_filters=["/image_raw"])

for i in range(num_images):
    # READ NEXT MESSAGE IN BAG
    topic, msg, t  = messages.next()

    # CONVERT MESSAGE TO A NUMPY ARRAY
    img = np.fromstring(msg.data, dtype=np.uint8)
    img = img.reshape(msg.height, msg.width)

    # DO SOME PROCESSING ON THE IMAGE    
    # ...

```

closed 2022-05-15 by MartyG-RealSense