realsenseai/librealsense #10460: L515 light saturation and blinking

ajiko39, 2022-05-02:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f1120935|
|**Firmware Version**|01.05.08.01|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|L500|
|**Asic Serial Number**|0003b66b3842|
|**Firmware Update Id**|0003b66b3842|

Please provide a description of the problem

Hello  
I have purchased 2 units of L515, and one works good, but other blinks with 3D sensing.
Could you confirm attached error logs and provide me some advice ?
Thanks.

[image]

MartyG-RealSense, 2022-05-02:
Hi @ajiko39  In past cases of this error - https://github.com/IntelRealSense/realsense-ros/issues/1750#issuecomment-814329923 and https://github.com/IntelRealSense/librealsense/issues/8866 - it has been recommended that the camera should be returned to the retailer that it was purchased from.

If it was purchased from the official online RealSense Store then please contact the Store using the contact form at the link below, selecting **Return / Warranty** as the topic.

https://www.intel.com/content/www/us/en/forms/support/intel-realsense-store-inquiry.html

If the camera was not purchased from the RealSense Store then please contact the retailer that the camera was bought from to ask about their **Return Material Authorization (RMA)** returns procedure.

ajiko39, 2022-05-02:
Thank you for reply.
I got it via digi-key, so I am asking them whether it can be returned.

MartyG-RealSense, 2022-05-10:
Hi @ajiko39  Do you have an update about this case that you can provide, please?  Thanks!

ajiko39, 2022-05-10:
Hi Marty-san,
I am in conversation with digikey (with e-mail).
I will post here if I have an update.
Thanks.

MartyG-RealSense, 2022-05-10:
Thank you very much!

ajiko39, 2022-05-11:
Hi Marty-san,

I got a message from digi-key that they has shipped another one at their responsibility.
They don't think that it is not needed to be returned.

So, it is OK to close this ticket if there is no problem.
What do you think?

Thanks.

MartyG-RealSense, 2022-05-11:
Yes, if you are satisfied with the outcome of the returns process then it is fine to close this case.  Thanks again!

ajiko39, 2022-05-11:
Thank you for comment.
I will close this ticket, when I get new one and it works well.
Thanks!

MartyG-RealSense, 2022-05-11:
That's no problem at all to wait until you have tested the new unit before closing.

ajiko39, 2022-05-13:
I got new one from supplier, and it works without any error above.
So, I close this issue.

Thank you for your support!

closed 2022-05-13 by ajiko39

MartyG-RealSense, 2022-05-13:
I'm very pleased that Digi-Key were able to resolve your problem.  Thanks very much for the update!