realsenseai/librealsense #10461: uvc streamer watchdog triggered on endpoint: 132

Xue21, 2022-05-02:
I install librealsense , there are some questions as below.
When I open realsense-viewer, it happened as below:
 02/05 01:38:59,605 WARNING [139708932695808] (model-views.cpp:4225) Exception caught trying to detect presets: could not open directory: /home/ubuntu186/Documents/librealsense2/presets/
 02/05 01:39:02,753 WARNING [139708835292928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/05 01:39:03,013 WARNING [139708835292928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/05 01:39:03,066 WARNING [139708835292928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/05 01:39:03,203 WARNING [139708835292928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/05 01:39:03,254 WARNING [139708835292928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/05 01:39:03,306 WARNING [139708835292928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/05 01:39:03,357 WARNING [139708835292928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/05 01:39:07,071 WARNING [139709269981056] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 02/05 01:39:07,417 ERROR [139708659111680] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 02/05 01:39:07,716 ERROR [139708482963200] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 02/05 01:39:09,138 WARNING [139708835292928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/05 01:39:09,189 WARNING [139708835292928] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2022-05-02:
Hi @Xue21  There was a recent case of the **Exception caught trying to detect presets: could not open directory** error at https://github.com/IntelRealSense/librealsense/issues/10361 though the case closed without a clear solution.
 
Could you please first check whether your camera has **Advanced Mode** enabled using the instructions at https://github.com/IntelRealSense/librealsense/issues/10361#issuecomment-1086166630

Also, could you provide information about the following questions.

-  Which RealSense camera model are you using a d what is its firmware version?

-  Which computer or computing device are you using (Raspberry Pi, Nvidia Jetson, desktop/ laptop PC, etc)

-. If you are using Linux, what is your kernel version?

-  Which method did you use to install librealsense?

closed 2022-05-02 by Xue21

monajalal, 2023-02-13:
Have the same exact error:

```
(base) mona@ard-gpu-01:~/Downloads/FRAMOS_D400e_Software_Package$ realsense-viewer
 13/02 09:29:00,014 WARNING [140585705059904] (model-views.cpp:4229) Exception caught trying to detect presets: could not open directory: /home/mona/Documents/librealsense2/presets/
 13/02 09:29:50,088 ERROR [140585209947712] (librealsense-exception.h:52) Unable to execute GevTimestampControlLatch
 13/02 09:30:21,888 ERROR [140585209947712] (librealsense-exception.h:52) Unable to read GevTimestampValue
 13/02 09:31:26,118 ERROR [140585209947712] (librealsense-exception.h:52) Unable to read GevTimestampValue
 13/02 09:31:49,657 ERROR [140585209947712] (librealsense-exception.h:52) Unable to read GevTimestampValue

```

MartyG-RealSense, 2023-02-13:
Hi @monajalal  I note in the first line of your log that it states **FRAMOS_D400e**.  If you are using a FRAMOS industrial e-series RealSense compatible camera then please contact FRAMOS support with your question as Intel cannot provide technical assistance for FRAMOS 'e' cameras because they are not an Intel-manufactured product and have their own SDK and firmware.  

FRAMOS support can be contacted by visiting their support web-page at the link below and clicking on the 'Let's Talk' button on the page.

https://www.framos.com/en/resources/support