realsenseai/librealsense #10462: rs-pointcloud-stitching installation

Testoxe, 2022-05-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 2.50.0 | 
| Operating System & Version |   Ubuntu 18.04  |                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.0  |
| Language                            |  Python  |

### Issue Description
I want to use the rs-pointcloud-stitching in the wrappers.
Its usage is pretty well documented, but it assumes it is already installed.
I can't find a way to install it.
Any insight on how to is welcome.
Thanks in advance.

MartyG-RealSense, 2022-05-02:
Hi @Testoxe  It is possible to perform point cloud stitching with Python, though there is not an official pre-made point cloud example program for doing so.  Whilst stitching can be done via the PCL or Open3D libraries, if you would prefer to do it completely within the RealSense SDK with Python then the instruction **rs2_transform_point_to_point** may meet your needs.  

This instruction is an 'affine transform', where separate sets of coordinates such as a pair of individual point clouds have their position and rotation set to the same values in 3D space.  The instruction is discussed at https://github.com/IntelRealSense/librealsense/issues/5583

***********

Alternatively, if you have two cameras and are able to use ROS and the RealSense ROS wrapper then Intel provide a stitching tutorial here:

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

Testoxe, 2022-05-02:
My priority is to check how accurate is the [rs-pointcloud-stitching](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pointcloud/pointcloud-stitching) program first to see if I can use it in my case, so it's not obligatory for me to use Python. 
I'm just trying to make it work but I can't find how to install it. I don't mind if it uses cpp.
Thank you for you detailed answer though, it will definitely help me in the future.

MartyG-RealSense, 2022-05-02:
As far as I am aware, there is no feedback available from other RealSense users regarding accuracy of the C++ **rs-pointcloud-stitching**  example program.  The instruction guide for the program at the link below indicates though that the **MATLAB** software tool and an additional tool for MATLAB called the **Computer Vision Toolbox** are required by this example program to calibrate two cameras together.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/pointcloud/pointcloud-stitching/doc/pointcloud-stitching-demo.md

If you are able to use C++ then it would be much easier to use the **rs-kinfu** SDK example.  This program only requires a single camera that you move around the scene to progressively build up a pointcloud in real-time by 'fusing' frames together instead of stitching separate point clouds from different views.  Once you are satisfied with the amount of detail in the pointcloud, you can stop adding frames and export the finished cloud as a .ply format pointcloud file.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu

Testoxe, 2022-05-02:
I need to visualize a real-time panorama with 2 cameras, so the **rs-kinfu** is not going to help.

I have already installed the **MATLAB's Stereo Camera Calibrator** software so that's not a problem.
The problem is, when following the [installation doc](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/pointcloud/pointcloud-stitching/doc/pointcloud-stitching-demo.md), I reach a step where I need to run rs-pointcloud-stitching command:
> Now, run rs-pointcloud-stitching:
> rs-pointcloud-stitching C:\pc_stitching_ws

But when I do, I get an error:
> rs-pointcloud-stitching: command not found

MartyG-RealSense, 2022-05-02:
I am not familiar with the **rs-pointcloud-stitching** example program.  I would assume though that it would first have to be built from its source code, and then you would input the **rs-pointcloud-stitching** command from the directory on your computer that it had been built in.

Testoxe, 2022-05-03:
I found a solution.
In order to use the rs-pointcloud-stitching program, BUILD_PC_STITCHING parameter should be true when installing the library.
Small tuto for anyone having the same problem:
- Clone repo
- Open a console in the root folder and execute following commands:
- `mkdir build && cd build`
- `cmake ../ -DBUILD_EXAMPLES=true -DBUILD_PC_STITCHING=true`
- `sudo make uninstall && make clean && make && sudo make install`

You should then be able to use `rs-pointcloud-stitching` command from the console, you can also find it in **usr/local/bin** by default.
Would be nice if this could be added to the [readme](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/pointcloud/pointcloud-stitching/doc/pointcloud-stitching-demo.md).

closed 2022-05-03 by Testoxe

MartyG-RealSense, 2022-05-03:
Thanks so much @Testoxe for sharing the solution that worked for you with the RealSense community!