realsenseai/librealsense #10464: Frame drops when calling  the Infrared streams

makolele12, 2022-05-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 5.13.00.50 | 
| Operating System & Version |   Ubuntu 20.04 VWware Virtual Machine  | 
| Kernel Version (Linux Only)    |  (5.8.0-23                       | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0                          | 
| Language                            |  C++ / Python / ROS                          | 
| Segment			|  Robotics | 

### Issue Description
I am experiencing several frame drops when using the RealSense D435i. I am using the camera for testing out some SLAM algorithms, which are sensitive to frame drops during long periods of time.

I am using Windows 11 and passing-through the camera to my VMWare Guest with Ubuntu 20.04.

I am using the following python code to call both Infrared streams:

```
try:
    # Create a context object. This object owns the handles to all connected realsense devices
    pipeline = rs.pipeline()
# Configure streams
    config = rs.config()
    config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)
    # config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 30)

    # Start streaming
    pipeline.start(config)
    ct = 0

    bFirst = True
    jitter_accel = []
    ros_time_accel = []

    while True:
        # This call waits until a new coherent set of frames is available on a device
        # Calls to get_frame_data(...) and get_frame_timestamp(...) on a device will return stable values until wait_for_frames(...) is called
        frames = pipeline.wait_for_frames()
        infra1 = frames.get_infrared_frame(1)
        
        if not infra1: 
            ct = ct + 1
            continue

        t = infra1.get_timestamp()
        
        # print( str(infra1.get_frame_number()) + " == " + str(ct))
        ct = ct + 1

        if bFirst:
            init_t = t
            last_t = t
            bFirst = False
            continue

        jitter_accel.append( (t - last_t) *1e-3)
        ros_time_accel.append( (t - init_t) *1e-3)
        last_t = t

    exit(0)
#except rs.error as e:
#    # Method calls agaisnt librealsense objects may throw exceptions of type pylibrs.error
#    print("pylibrs.error was thrown when calling %s(%s):\n", % (e.get_failed_function(), e.get_failed_args()))
#    print("    %s\n", e.what())
#    exit(1)
except Exception as e:
    print(e)
    pass 
```

After capturing frames for around 200-300 seconds, I analyze the time difference between the last and current frame. For instance, the following plot depicts this difference for the infrared 1 camera:

[image]

I identify the frame drops because the multiple observed peaks are always multiples of my chosen frequency (30 Hz). Some peaks point to up to 10 simultaneous frame drops.

My first thought was bandwidth. Therefore, I did test the same but only calling one camera at 30 FPS. The result as expected was a really low jittering with no frame drops

[image]

Afterwards, I decided to call both streams but at lower frame rates.
Firstly, with 15 FPS, which looks better than before, but its still dropping some frames.
[image]

Secondly, with 6 FPS. I only observed a frame drop.

[image]

It looked like this was a bandwidth problem. However, I decided to make some tests with the RS-Camera app for Android, just to check if the same problem was replicated, nevertheless this was not the case.

The setup is the following:

```
Android 12
librealsense 2.50.0
camera app version 2.50.0.3697
```

I recorded a ROSBag while calling again both infrared streams at 30 FPS. I performed a timestamp correction of the bag and plotted the same graph. The result is the following:

[image]

Only one frame was dropped in around 200 seconds. Therefore, I am not sure anymore if this is a bandwidth problem.

Any insight on why this is happening with my Virtual Machine setup would be appreciated. Thanks in advance.

Additional Information:

On my Virtual Machine I did the same tests using the realsense-ros-wrapper and recording a rosbag. The Frame-drops were still present just like when using pyrealsense.

MartyG-RealSense, 2022-05-04:
Hi @makolele12   My sincere apologies for the delay in responding.

As you are using two streams and experiencing frame drops as the FPS is increased, it would be worth first trying an increase of the **frame queue size**.  Increasing the frame queue size increases the total number of frames that can circulate in the pipeline at the same time.  By increasing **latency** in this way, the risk of dropped frames reduces.  The default frame queue size is '1'.

This principle is discussed in the RealSense SDK's official **Frame Buffering Management** documentation at the link below.

https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0

The guide suggests a frame queue size of '2' for two streams.  Some RealSense users have had success with '50' though.  An example of Python code for setting the frame queue size is at https://github.com/IntelRealSense/librealsense/issues/6448#issuecomment-633041074

makolele12, 2022-05-05:
Hi @MartyG-RealSense , thanks for your reply!

I've followed the sample script and managed to change the frames_queue_size accordingly. However, I got similar results.

This is the result for frames_queue_size of 2, a bit of improvement but still problematic.

[image]

frames_queue_size of 3 shows no multiple frame drops, but still single ones.

[image]

When trying frames_queue_size of 50 I got the following message:

`set(frame_queue_size) failed! Given value 50 is out of range.`

Therefore, the most I tried was frames_queue_size of 30:

[image]

I also tried using different cables, and made sure that my port is USB3.0. Is there anything else I could try?

MartyG-RealSense, 2022-05-06:
Intel recommend using **VMWare Workstation Player** for a VM if you are not doing so already in order to properly emulate the USB 3 controller, as described at the link below.  My understanding is that VMWare Workstation Pro will also work.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#linux-ubuntu-installation

Having said that, your performance with VMWare Guest seems to be good, aside from occasional frame drops.

In a SLAM system, performing periodic **relocalization** may help to correct any navigation errors that build up over time due to frame drops.

makolele12, 2022-05-09:
I am using VMware Workstation 16 Pro and already made sure that USB 3.1 is chosen in the settings. 

Yes SLAM can recover itself after some dropped frames. However, for my specific use-case I cannot afford loosing Track so easily because of the consecutive multi frame-drops.

I am going to investigate the problem further, if I find something useful, ill report it here. Thanks again!

MartyG-RealSense, 2022-05-09:
Thanks very much for the information, @makolele12 - good luck!

closed 2022-05-26 by makolele12

MartyG-RealSense, 2022-05-26:
No problem at all, @makolele12 - thanks very much for the update!