realsenseai/librealsense #10465: RealSense D405 / D401 official CAD files now available

MartyG-RealSense, 2022-05-04:
Hi everybody,

The official Intel CAD files for the RealSense D405 camera and the D401 depth module are now available at the link below.

https://dev.intelrealsense.com/docs/stereo-depth-camera-d400#section-cad-files

MartyG-RealSense, 2022-05-09:
Hi @dpbnasika  There are currently access problems with the https://www.intelrealsense.com/ website at the time of writing this.  The CAD file downloads use this web address.  Please try again later. Thanks for your patience!

dpbnasika, 2022-05-09:
Hi @MartyG-RealSense i still get the error. I could not able to access it.

MartyG-RealSense, 2022-05-09:
It may come back later today, taking into account the time zone difference between California and the rest of the world.

closed 2022-06-03 by MartyG-RealSense