realsenseai/librealsense #10466: Low FPS using IMU on Camera L515 in Nvidia Jetson Xavier

maximilianxing, 2022-05-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   |01.05.08.01 | 
| Operating System & Version |  Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  4.9.140-tegra      | 
| Platform                                 | NVIDIA Jetson Xavier with JetPack 4.4 |
| SDK Version                            |  2.48.0                | 
| Language                            |  python                    | 
| Segment			|  others                  | 

I have an issue regarding low fps performance of L515 using IMU.

```
process[camera/realsense2_camera_manager-1]: started with pid [17417]
process[camera/realsense2_camera-2]: started with pid [17418]
[ INFO] [1651656240.686239226]: Initializing nodelet with 8 worker threads.
[ INFO] [1651656241.256085383]: RealSense ROS v2.3.1
[ INFO] [1651656241.256314957]: Built with LibRealSense v2.48.0
[ INFO] [1651656241.256419888]: Running with LibRealSense v2.48.0
[ INFO] [1651656241.320069345]:  
[ INFO] [1651656241.413807708]: Device with serial number f0232284 was found.

[ INFO] [1651656241.414024386]: Device with physical ID /sys/devices/3610000.xhci/usb4/4-3/4-3:1.0/video4linux/video0 was found.
[ INFO] [1651656241.414117892]: Device with name Intel RealSense L515 was found.
[ INFO] [1651656241.416133402]: Device with port number 4-3 was found.
[ INFO] [1651656241.416293215]: Device USB type: 3.2
[ INFO] [1651656241.416401025]: Resetting device...
[ INFO] [1651656247.443978202]:  
[ INFO] [1651656247.546270196]: Device with serial number f0232284 was found.

[ INFO] [1651656247.546641278]: Device with physical ID /sys/devices/3610000.xhci/usb4/4-3/4-3:1.0/video4linux/video0 was found.
[ INFO] [1651656247.546823299]: Device with name Intel RealSense L515 was found.
[ INFO] [1651656247.549741330]: Device with port number 4-3 was found.
[ INFO] [1651656247.550146460]: Device USB type: 3.2
[ INFO] [1651656247.562760495]: getParameters...
[ INFO] [1651656247.862833736]: setupDevice...
[ INFO] [1651656247.863011789]: JSON file is not provided
[ INFO] [1651656247.863154961]: ROS Node Namespace: camera
[ INFO] [1651656247.863350262]: Device Name: Intel RealSense L515
[ INFO] [1651656247.863433656]: Device Serial No: f0232284
[ INFO] [1651656247.863511130]: Device physical port: /sys/devices/3610000.xhci/usb4/4-3/4-3:1.0/video4linux/video0
[ INFO] [1651656247.863600701]: Device FW version: 01.05.08.01
[ INFO] [1651656247.864188141]: Device Product ID: 0x0B64
[ INFO] [1651656247.864372466]: Enable PointCloud: On
[ INFO] [1651656247.864509461]: Align Depth: On
[ INFO] [1651656247.864605016]: Sync Mode: On
[ INFO] [1651656247.864867551]: Device Sensors: 
[ INFO] [1651656247.868716902]: L500 Depth Sensor was found.
[ INFO] [1651656247.878174212]: RGB Camera was found.
[ INFO] [1651656247.878896856]: Motion Module was found.
[ INFO] [1651656247.880556932]: Add Filter: pointcloud
[ INFO] [1651656247.882559034]: num_filters: 2
[ INFO] [1651656247.882782880]: Setting Dynamic reconfig parameters.
[ INFO] [1651656248.110075638]: Done Setting Dynamic reconfig parameters.
[ INFO] [1651656248.111821253]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1651656248.112808063]: infra stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1651656248.114534637]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1651656248.120142596]: setupPublishers...
[ INFO] [1651656248.131322064]: Expected frequency for depth = 30.00000
[ INFO] [1651656248.264350403]: Expected frequency for infra = 30.00000
[ INFO] [1651656248.346672772]: Expected frequency for color = 30.00000
[ INFO] [1651656248.427649474]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1651656248.512551432]: setupStreams...
[ INFO] [1651656248.566892187]: insert Depth to L500 Depth Sensor
[ INFO] [1651656248.567413321]: insert Color to RGB Camera
[ INFO] [1651656248.567741234]: insert Infrared to L500 Depth Sensor
[ INFO] [1651656248.594443806]: SELECTED BASE:Depth, 0
 04/05 11:24:08,752 WARNING [546450690432] (l500-depth.cpp:363) UVC metadata payloads are not available for stream . Please refer to installation chapter for details.
[ INFO] [1651656248.795283085]: RealSense Node Is Up!
```
Here is the FPS of /color/image_raw

```
subscribed to [/camera/color/image_raw]
average rate: 25.303
	min: 0.002s max: 0.084s std dev: 0.01484s window: 24
average rate: 25.895
	min: 0.002s max: 0.084s std dev: 0.01322s window: 51
average rate: 25.432
	min: 0.002s max: 0.084s std dev: 0.01391s window: 75
average rate: 25.984
	min: 0.002s max: 0.084s std dev: 0.01300s window: 103
average rate: 25.752
	min: 0.002s max: 0.094s std dev: 0.01374s window: 128
average rate: 25.665
	min: 0.002s max: 0.094s std dev: 0.01321s window: 153
average rate: 25.608
	min: 0.002s max: 0.094s std dev: 0.01351s window: 178
average rate: 25.363
	min: 0.002s max: 0.094s std dev: 0.01366s window: 202
average rate: 25.573
	min: 0.002s max: 0.094s std dev: 0.01362s window: 229
average rate: 25.811
	min: 0.001s max: 0.094s std dev: 0.01464s window: 257
```
Using IMU with 200 fps (enable_gyro = true and enable_accel = true, both 200 fps) the average rate is much less.

```
subscribed to [/camera_l515/color/image_raw]
average rate: 3.522
	min: 0.267s max: 0.299s std dev: 0.01333s window: 4
average rate: 3.304
	min: 0.267s max: 0.330s std dev: 0.02181s window: 7
average rate: 3.191
	min: 0.267s max: 0.339s std dev: 0.02351s window: 10
average rate: 3.084
	min: 0.267s max: 0.375s std dev: 0.03131s window: 12
average rate: 3.061
	min: 0.267s max: 0.375s std dev: 0.02830s window: 15
average rate: 3.054
	min: 0.267s max: 0.375s std dev: 0.02525s window: 19
average rate: 3.049
	min: 0.267s max: 0.375s std dev: 0.02346s window: 22
average rate: 3.043
	min: 0.267s max: 0.375s std dev: 0.02202s window: 25
average rate: 3.035
	min: 0.267s max: 0.375s std dev: 0.02104s window: 28
average rate: 3.013
	min: 0.267s max: 0.375s std dev: 0.02226s window: 30
average rate: 2.987
	min: 0.267s max: 0.378s std dev: 0.02326s window: 33
average rate: 2.965
	min: 0.267s max: 0.402s std dev: 0.02504s window: 36
```
I have also tried to change resolution of color with 1028*720 or other fps values of color, gyro and accel but got same performance.
Has anyone suggestions?
Thanks

MartyG-RealSense, 2022-05-04:
Hi @maximilianxing   It is s known issue with the RealSense ROS wrapper that if pointcloud or depth-color alignment are enabled on Jetson boards then effects such as low FPS and missing color can result.  This does not usually occur on non-Jetson computers and computing devices such as Raspberry Pi and desktop / laptop PCs.  This Jetson issue is described in detail at https://github.com/IntelRealSense/realsense-ros/issues/1967

This issue may not be affecting you in your particular case though as I would expect the FPS to be around 3 instead of 25 if you had this Jetson-specific problem.

Instead, your FPS may be dropping due to a different issue that can occur when having depth and color enabled at the same time.   When auto-exposure is enabled, disabling an RGB option called **auto_exposure_priority** can correct the FPS drop by forcing FPS to remain at a constant rate.  Placing the instruction below in your launch file should disable auto_exposure_priority on both L515 and 400 Series camera models.

```
<rosparam> 
/camera/rgb_camera/auto_exposure_priority: false 
</rosparam>
```

MartyG-RealSense, 2022-05-04:
In regard to IMU performance, please try setting Gyro to 400 and Accel to 250.

MartyG-RealSense, 2022-05-04:
So the color FPS of the topic **/camera/color/image_raw** averages 25 FPS when the IMU is enabled and drops to 3 FPS average when IMU is enabled.

What happens to the color FPS if you add the command below to your roslaunch instruction after enable_gyro and ensble_accel:

`unite_imu_method:=copy`

MartyG-RealSense, 2022-05-04:
Do you require the infra stream?  If you do not, does FPS improve if you add **enable_infra:=false** to the roslaunch instruction to disable it?

maximilianxing, 2022-05-05:
Hi @MartyG-RealSense FPS was not improved, when infra was disabled.

Here is the setup, which i use now.
```
  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="640"/>
  <arg name="confidence_height"   default="480"/>
  <arg name="enable_confidence"   default="false"/>
  <arg name="confidence_fps"      default="30"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="true"/>
  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default="pointcloud"/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="true"/>
  <arg name="unite_imu_method"          default="copy"/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>

  <arg name="stereo_module/exposure/1"  default="7500"/>
  <arg name="stereo_module/gain/1"      default="16"/>
  <arg name="stereo_module/exposure/2"  default="1"/>
  <arg name="stereo_module/gain/2"      default="16"/>
```

maximilianxing, 2022-05-05:
Hi @MartyG-RealSense I think i found where the problem is. Wenn i enable Gyro and Accel and set up `pointcloud_texture_stream := RS2_STREAM_COLOR`, FPS was always low. But with `pointcloud_texture_stream := RS2_STREAM_ANY` or disabling Gyro and Accel, FPS become normal again. But I can not understand how this happens and how to solve it.

MartyG-RealSense, 2022-05-06:
My understanding is that when using **pointcloud_texture_stream :=RS2_STREAM_ANY** it allows the pointcloud to be textured with a color source other than RGB.

xinxing-max, 2022-05-09:
Hi @MartyG-RealSense Thar's right. The RGB information of the pointcloud needs to be used, which means **pointcloud_texture_stream := RS2_STREAM_COLOR**, so I don't know what to do now.

xinxing-max, 2022-05-10:
Hi @MartyG-RealSense Thank you so much for your patience and answers. With librealsense Release 2.48.0 and 2.49.0 there were always low image performance, although i have tried so many setting possibilities, such as **infra_rgb** and **enable_infra**, but not worked. Today i have updated librealsense to 2.50.0 and realsense-ros to 2.3.2 in Jetson Xavier, the image performance was also not perfect but i got 25 FPS. I think that would be better when disabling **auto_exposure_priority**.  I also looked at the upgrade notes, and it doesn't seem to have improved this kind of problem.

xinxing-max, 2022-05-10:
@MartyG-RealSense yes, i got 29~30 FPS after disabling auto_exposure_priority, but FPS of pointcloud was only about 15 FPS. It's ok.

MartyG-RealSense, 2022-05-10:
Did you build the librealsense SDK from source code with CMake?  If so, then when building librealsense with this method on Jetson it is necessary to include the CMake build flag **-DBUILD_WITH_CUDA=true** to enable CUDA support in the SDK.  CUDA can provide computers with an NVidia graphics GPU such as Jetson with automatic acceleration of alignment, pointcloud and color conversion operations by offloading processing work from the CPU onto the GPU.

If librealsense was built from packages on Jetson then CUDA support should already be included in the packages.

xinxing-max, 2022-05-10:
@MartyG-RealSense I built the librealsense SDK always from source code with CMake and in CMake CUDA was also enabled. I use D435i and L515 at same time in Jetson. There was no any problem at D435i but L515. So i think Hardware was not the reason for that.

xinxing-max, 2022-05-10:
@MartyG-RealSense not this case. In our project there are mobil robot with L515 and robot arm with D435i and they are not close to each other. Before i have also tried to only start L515 but doesn't seem to have changed anything.

MartyG-RealSense, 2022-05-11:
Do you get better performance with the L515 if you unplug the micro-sized end of the USB cable from the base of the camera, turn the connector around the opposite direction and re-insert it into the base of the camera?  USB-C cables are two-way insertion on the micro-sized end, and the L515 model can sometimes operate better when a particular one of the two available connector orientations is used.

xinxing-max, 2022-05-11:
@MartyG-RealSense I think the right connector orientation was used. When using another connector orientation, camera was only connected with 2.1 port instead of 3.2 port.

MartyG-RealSense, 2022-05-11:
Are you able to open the **dynamic reconfigure** interface for the L515 with the command **rosrun rqt_reconfigure rqt_reconfigure** after launch has completed and then try different camera presets (such as 'Short Range' instead of the default 'Maximum Range' preset?  If you can, do any of the presets result in an increase of the FPS from 15 when the pointcloud is enabled?

xinxing-max, 2022-05-11:
@MartyG-RealSense Strangely, just after plugging and unplugging the USB Port, the fps of the point cloud reached 21. After selecting 'Short Range', the fps increased to 22. Could it be that the usb port is sometimes in poor contact? But fps of image, imu or infra was normal.

MartyG-RealSense, 2022-05-11:
The reason I asked to try different presets is that the L515 model sometimes does not give optimal results on the default Maximum Range preset.  Changing the camera's configuration values with a different preset can sometimes give better performance depending on what the camera is observing.  

For example, the Short Range preset reduces the values of the Laser Power and the Receiver Gain settings, making the camera better able to deal with surfaces that have reflections or are transparent (like an empty drink bottle).  The Low Ambient Light preset, meanwhile, helps the camera to deal with any natural light that is present in the scene.

Are you using a USB cable with screw-locks on the connector with your L515?  There was a case at https://github.com/IntelRealSense/librealsense/issues/8830 where performance worsened when the connector was screwed tightly into the camera but returned to normal when the connector's screws were loosened.

xinxing-max, 2022-05-11:
@MartyG-RealSense No, the space reserved for the USB cable of the L515 is too small to use the USB cable with screw-locks. I use this kind of cable for my D435i. In addition, there is another problem, currently I am using a USB C to USB C cable to link the Jetson and the camera L515. When I start L515 with the USB C to USB cable, it always prompts that the device can not be found or that it can only recognize USB 2.1 Port.

MartyG-RealSense, 2022-05-11:
Historically, USB C to C cable connections do tend to have more problems with RealSense cameras than USB Type C cables.

xinxing-max, 2022-05-11:
@MartyG-RealSense i got it. Thanks! For now, the performance of the FPS has improved a lot compared to before. I can at least move on with my project.