realsenseai/librealsense #10467: Unable to access accelerometer data for some devices.

Drulludanni, 2022-05-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 18, Windows 10  | 
| Kernel Version (Linux Only)    |  5.4.0-104-generic                                 | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  pyrealsense2==2.49.0.3474   | 
| Language                            |  python 3.7                        | 
| Segment			|  Computer Vision                       | 

### Issue Description
Hi for some reason I'm unable to access the accelerometer data through python on some machines with the error "Couldn't resolve request" however there are other machines where the code runs no problem and I don't understand why, furthermore on the machines where the acceleromter code does not seem to work I'm still able to access the accelerometer data through the realsense-viewer

the code to start the camera looks something like this:

```
self.config.enable_stream(
    rs.stream.depth,
    cfg.FRAME_WIDTH,
    cfg.FRAME_HEIGHT,
    rs.format.z16,
    cfg.FPS)
self.config.enable_stream(
    rs.stream.color,
    cfg.FRAME_WIDTH,
    cfg.FRAME_HEIGHT,
    rs.format.bgr8,
    cfg.FPS)
self.config.enable_stream(
    rs.stream.accel,
    rs.format.motion_xyz32f,
    250) # must be 63 or 250
				
try:
    self.profile = self.pipeline.start(self.config) # Couldn't resolve request error happens here
except Exception as e:
    self.accelerometer = False
    self.config.disable_stream(rs.stream.accel) # disabling the accelerometer "solves" the issue
    self.profile = self.pipeline.start(self.config)
    print("accelerometer error:", e)
```

I added the info page for one that works and one that doesn't
[image]

Any suggestions on how to fix this would be greatly appreciated.

MartyG-RealSense, 2022-05-04:
Hi @Drulludanni   There was a recent case at https://github.com/IntelRealSense/librealsense/issues/10394 of a Python user trying to access Accel and finding that it worked on Ubuntu running on a Mac computer but not in Ubuntu on an Nvidia Jetson.  Their successful solution to correct the problem was to build the RealSense SDK from source code with the **RSUSB backend** installation method.

MartyG-RealSense, 2022-05-04:
There was a past case of two Ubuntu PC computers of the same manufacturer / model and the camera worked normally with one but not the other that belonged to the RealSense user's colleague.  Eventually they found that although the two computers seemed identical, the colleague's computer that the camera worked with had its OS updated to a more recent configuration.  Once the computer with the problem was updated to match the colleague's computer then the camera worked normally.

MartyG-RealSense, 2022-05-11:
Hi @Drulludanni  Do you have an update about this case that you can provide, please?  Thanks!

Drulludanni, 2022-05-12:
Hi, no sorry I didn't get around to testing and currently the device the was not working is unavailable (I have to access it remotely but it wont connect) I'll try to get this done if I get a connection to the device.

MartyG-RealSense, 2022-05-12:
Thanks very much for the update, @Drulludanni

Drulludanni, 2022-05-12:
So I started looking into building from source using [this comment](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049) as a guide however there are a few differences and I'm not sure why (maybe because the comment is old?), using cd into a build directory doesn't do anything because cmake will build it in the "librealsense-2.50.0" directory. I also removed some options that did not seem to apply to my case so my cmake looks like this: `cmake -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python3.7 -DCMAKE_BUILD_TYPE=release .` and everything seems to work fine, however when I do `sudo make install` (skipping the make j4 since I'm not using a jetson) I get the following error:

```
[  2%] Performing update step for 'libusb'
Error copying file (if different) from "/home/vaki/Desktop/librealsense-2.50.0/third-party/libusb/CMakeLists.txt" to "/home/vaki/Desktop/librealsense-2.50.0/third-party/libusb/CMakeLists.txt".
CMakeFiles/libusb.dir/build.make:95: recipe for target 'libusb-prefix/src/libusb-stamp/libusb-update' failed
make[2]: *** [libusb-prefix/src/libusb-stamp/libusb-update] Error 1
CMakeFiles/Makefile2:106: recipe for target 'CMakeFiles/libusb.dir/all' failed
make[1]: *** [CMakeFiles/libusb.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
```

and I'm not sure how to continue.

MartyG-RealSense, 2022-05-12:
The -j number used with make is not specific to Jetson.  It tells CMake how many 'threads' of your computer's CPU processor chip will be allowed to be used in the build process.  The number of threads is X2 the number of cores in the CPU.  So for a dual core CPU, -j4 would be suitable and for a quad core CPU -j8 would be suitable.

Are you able to build successfully if you use the command below to let CMake decide what j-value to use based on your computer's processor?

**make -j$(($(nproc)-1))**

Drulludanni, 2022-05-13:
I get a little bit further but it crashes with a very unhelpful error message:

```
[ 15%] Completed 'libcurl'
[ 15%] Built target libcurl
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

MartyG-RealSense, 2022-05-14:
The RSUSB installation method requires an internet connection.  Was your computer connected to the net when building, please?

If you do have an internet connection, it may be worth trying a simpler RSUSB build method for Linux at the link below that does not include the Python wrapper in order to help to establish whether your computer can build librealsense from source code at all.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Drulludanni, 2022-05-17:
yes, this seems to work

MartyG-RealSense, 2022-05-17:
The link below shows the list of commands that the libuvc backend procedure uses to achieve a successful build.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

MartyG-RealSense, 2022-05-23:
Hi @Drulludanni  Do you require further assistance with this case, please?  Thanks!

Drulludanni, 2022-05-23:
I mean yes, I still cannot get acceloremeter data through python.

Drulludanni, 2022-05-23:
I did uname -r for the computers I could connect to and got these results:

works:
5.4.0-104-generic
5.4.0-97-generic

doesnt work:
5.4.0-107-generic
5.4.0-109-generic

MartyG-RealSense, 2022-05-23:
Thank you.  Are you able to change the kernel version to -104 or -97 on the machines that the code does not work on?

MartyG-RealSense, 2022-06-04:
Case closed due to no further comments received.

closed 2022-06-04 by MartyG-RealSense

Drulludanni, 2023-03-13:
Hi I revisited this issue again and now I have a case where both computers are near identical: they have the same kernel version (5.4.0-144-generic) and they have the exact same cpu (Intel(R) Core(TM) i3-8145UE CPU @ 2.20GHz) and motherboard, is there no way to get a proper error code rather than just "Couldn't resolve request"?

MartyG-RealSense, 2023-03-13:
Looking at the original date of this case (May 2022), if your D435i was recently purchased at that time then you may have a D435i that has a changed USB component.  

Around that time, the IMU component used in the D435i and D455 models was changed.  This IMU in these more recently manufactured camera units does not support 63 accel and 200 gyro, but instead supports **100** accel and 200 gyro.  So if you are requesting 63 accel on the revised modern models, you would get a **Couldn't resolve request** error.

You can check whether your D435i is a modern unit that uses 100 Accel instead of 63 by looking at the drop-down list of available Accel frequencies in the Viewer's Motion Module.

Drulludanni, 2023-03-14:
This appears to solve the issue. Is there a way for me to programmatically determine which mode the IMU supports rather than just guess and check?

MartyG-RealSense, 2023-03-14:
The SDK has a **can_resolve()** instruction to check whether the cfg configuration can be provided, though it is rarely used and so there is not a pre-existing script that demonstrates how to use it.

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.config.html#pyrealsense2.config.can_resolve

Drulludanni, 2023-03-15:
is there no "list available settings" function?

MartyG-RealSense, 2023-03-15:
The **get_stream_profiles()** instruction can be used to check supported stream modes.

The Python code linked to below in the RealSense SDK's IMU calibration tool **rs-imu-calibration.py** demonstrates querying the accel and gyro profiles.

https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-imu-calibration/rs-imu-calibration.py#L240-L247

In that code, I would think that 'pr.format' would be changed to 'pr.fps' to query the frequency instead of the xyz32f format since a config instruction follows the structure ** config.enable_stream(STREAM_TYPE, WIDTH, HEIGHT, FORMAT, FPS)**

Maybe like this:

` if pr.stream_type() == rs.stream.gyro and pr.fps() == 100:`