realsenseai/librealsense #10469: D435i, D455 not supported by Intel.Realsense.CustomRW

mechatheo, 2022-05-04:
Hello, realsense community,

sorry to bother you again. This is a follow-up to https://github.com/IntelRealSense/librealsense/issues/10357. 
We are trying to use the `Intel.Realsense.CustomRW` tool to reset our cameras, however, it seems they are not supported.
Is there a newer version of this tool? Or any other way to perform the golden reset, on our camera models?
[image]

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i, D455 | 
| Operating System & Version |   Linux |

MartyG-RealSense, 2022-05-04:
Hi @mechatheo   There should not be a problem with using those camera models with CustomRW.  You should though be able to reset cameras to their factory-new default calibration in the RealSense Viewer using the alternative method described at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

70m6, 2022-05-09:
Hi, I am also working with @mechatheo on this issue. 

I was able to reset the camera with this method, but it ultimately did not solve our end goal. 

What we are trying to do is prevent this error from cropping up:
```
RealSense error calling rs2_update_firmware_cpp(device:0x7f96a8013420, fw_image:0x1237260):
    Device: 024323050105 is locked for update.
Use firmware version higher than: 5.13.0.50
```
The reason we are trying to do this is creating a testbench for updating our cameras, as it it sometimes unstable.
This entails setting the camera version back and forth several times (our example here is between `Signed_Image_UVC_5_12_15_50` and `Signed_Image_UVC_5_13_0_50`)

Is there anyway to reset this counter?
Thanks

MartyG-RealSense, 2022-05-09:
Hi @70m6  RealSense cameras are limited to a maximum of 20 firmware downgrades before a lock activates and the camera can only be upgraded to a firmware version newer than the currently installed version.  This restriction has been in place for the whole history of the RealSense product range since 2014.  Whilst there is not an official explanation for this policy, my understanding is that it was created to deter frequent malicious changes to the camera from 'malware' attacks.

I once had a case where a RealSense user claimed that they were able to reset the counter to zero by performing a 'gold reset' of the camera to its factory-new configuration with the **CustomRW** tool that your colleague @mechatheo used.  The reason for creating this case though was that CustomRW was not working with your particular camera units.

Are you using USB 2.1 please?  My recollection is that the Dynamic Calibration tool was written at the launch of the 400 Series cameras in late 2017, at which time USB 2.1 support had not been introduced into the RealSense SDK, and so USB 3.1 was required.  I checked the user guide of the tool to confirm this.  In its **Hardware Requirements** section it states "The host processor connection to the camera is through USB 3.1 Gen 1"

70m6, 2022-05-10:
Hi @MartyG-RealSense ,
Yes that did seem to be the issue, we are now able to list and reset all devices using CustomRW version 2.11.1.0.

After performing:
`Intel.Realsense.CustomRW -g`

And getting success back:
```
CustomRW for Intel RealSense D400, Version: 2.11.1.0

  Device PID: 0B3A
  Device name: Intel RealSense D435I
  Serial number: 031522070981
  Firmware version: 05.13.00.50

Calibration on device successfully reset to default gold factory settings.
```

Unfortunately after trying:
`rs-fw-update --serial_number 031522070981 -f ./Signed_Image_UVC_5_12_15_50.bin` 
We still have the same issue:

```
RealSense error calling rs2_update_firmware_cpp(device:0x7f05900086d0, fw_image:0x194a080):
    Device: 024323050105 is locked for update.
Use firmware version higher than: 5.13.0.50
```

MartyG-RealSense, 2022-05-10:
The gold reset is the only method that has been reported to undo the lock, and even then it was only claimed once and never repeated and confirmed by other RealSense users.  So it is likely that the affected camera is permanently only able to be upgraded to newer firmwares from this point onwards, unfortunately.

mechatheo, 2022-05-16:
Hello Guys,

i understand all the points. However can someone from the Realsense Team respond, whether something could be done about this restriction. It's really limiting to us. Imagine you would want to incorporate a relsense cameras into a professional product, which supports over the air updates. Naturally you would test the update process very thoroughly. When we do such test, especially when peripheral updates are part of the update package, we like to run 100-1000 of cycles to detect even rarely occurring issues. Given your current setup it's almost impossible to run automated update tests. 

I wouldn't ask all that, if we could trust the `rs-fw-update` tool blindly. However what we often see, is that a normal update, done following the instruction, will leave the camera in recovery mode, where the recovery option must be used to bring the cam back to live. Sometimes the tools will just freeze mid process. We are trying to automate all such scenarios, however this is really hard to do, if you only have 20 tries.
Alternative question: is there already a maintained wrapper tool which uses `rs-fw-update` which can perform a realsense update automatically?

Thanks

MartyG-RealSense, 2022-05-16:
As this lock has been RealSense policy for the past 8 years, it is unlikely to be changed.

The counter should only be triggered when there is a firmware downgrade and not when only upgrades are performed.  If reliability is critical to your project then a better policy than upgrade-downgrade tests is likely to be to only use firmware versions that Intel has categorized as **Production** (confirmed long-term stable for critical applications and commercial products).  At the time of writing this, the most recent recommended Production firmware version and RealSense SDK version combination is firmware **5.13.0.50** and SDK **2.50.0**.

https://dev.intelrealsense.com/docs/recommended-production-camera-configurations**

There is not a pre-made firmware update tool but you can write your own program in the RealSense SDK to control firmware checking and updates.  RealSense team members provide programming advice for doing so at https://github.com/IntelRealSense/librealsense/issues/4416

closed 2022-05-28 by MartyG-RealSense