realsenseai/librealsense #10470: How to subscribe two realsense image topic through one callback

qin53027, 2022-05-04:
Hi Marty,

In my laptop, I have installed the librealsense SDK so that I can use following python code for D435i camera

import cv2
import pyrealsense2
from realsense_depth import *
# Initialize Camera Intel Realsense
dc = DepthCamera()

while True:
    ret, depth_frame, color_frame = dc.get_frame()    ------- one function call gets two frames 
    ------ --------------------- ---------------------------------
So I can get the color_frame and depth_frame which can be used to detect the distance. 

In my real Robot, I have the D435i as well. 

#following is used to subscribe color_frame
self.image = rospy.Subscriber("/realsense/color/image_raw",sensor_msgs.msg.Image,self.camera_callback)

#following is used to subscribe the depth_frame which is used to get the distance
image_topic = "/realsense/depth/image_rect_raw" #16UC1 format
self.depth_image = rospy.Subscriber(image_topic, sensor_msgs.msg.Image, self.image_callback)

Although, I can get both frames in ROS run, since frames are from two different callbacks: camera_callback and image_callback, the robot moves in on and off base. I think that two callbacks switch back and forth caused the problem.  If I only use one callbacks, then robot moves smoothly and normally.

Do you know how I can use one callback to get two frames: depth_frame and color_frame through subscribe in ROS?

Thanks so much for you help
Jimmy

MartyG-RealSense, 2022-05-05:
Hi @qin53027   My research of your question did not find a reference for using one Python callback for two streams with ROS topics.  The closest I could find was one that apparently uses two callbacks in two separate Python scripts at https://github.com/IntelRealSense/realsense-ros/issues/1711#issuecomment-782631635

Maybe it will provide some ideas for refining your current two callback approach.

qin53027, 2022-05-06:
Hi Marty, thanks for your information.

At this time, I can use ApproximateTimeSynchronizer to combine two image topic into one callback. Although it works, it still is kind of slow.  So for my simple solution, I just used the depth_image only to train and detect the distance since it works

In case someone may have same question, following is just my example python code.

Thanks
Jimmy

from message_filters import TimeSynchronizer, Subscriber, ApproximateTimeSynchronizer
=========================================
        image_sub = Subscriber(
            "/realsense/color/image_raw",
            sensor_msgs.msg.Image,
            queue_size=1
        )

        depth_sub = Subscriber(
            "/realsense/depth/image_rect_raw",
            sensor_msgs.msg.Image,
            queue_size=1
        )

        # Time syncronizer is implimented to make sure that all of the frames match up from all of the topics.
        #ts = TimeSynchronizer([image_sub, depth_sub], 10)
        ts = ApproximateTimeSynchronizer([image_sub, depth_sub], 1, 0.1)
        ts.registerCallback(self.mycallback)

closed 2022-05-06 by qin53027

MartyG-RealSense, 2022-05-06:
Thanks so much @qin53027 for sharing your solution with the RealSense community!