realsenseai/librealsense #10471: What if I use Realsense D435i to work with hight-ension wire closely, will the camera be disturbed by electromagnetic?

jianjuan, 2022-05-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am trying to use D435i to locate object's pose, and my object is hight-ension wire， during the machine's working, the camera will be very close to the camera, like 5cm, I want to know whether if  the depth will be affected by this situation  or not.
 Forwarding for your reply !

MartyG-RealSense, 2022-05-05:
Hi @jianjuan  There have not been any reported problems with magnets during the history of RealSense products.  

The original RealSense R200 stereo depth camera model (the predecessor of the modern 400 Series stereo cameras) was supplied with a magnetic plate resembling a very strong fridge magnet that attached to the rear of the camera and enabled it to be mounted to surfaces such as computer monitors. 

 It would not be practical to sense depth at 5 cm range though as the default minimum depth measurement range of the D435i camera model is 0.1 meters / 10 cm.  Even if the minimum distance of the camera was reduced with the **disparity shift** option the D435i would not be able to detect depth at 5 cm.

However, **RGB** does not have such range limits and can produce a clear image at 5 cm.

jianjuan, 2022-05-05:
Thank you very much for your explanation! 
And I am sorry, I have not described the scene clearly. What  I mean is if there is a hight-ension wire very close ( like 5 cm) to the realsense camera on the top, does this affect the depth ability of the normal range like (0.1~10m) ? I have two wires,  I am not intended to watch depth of the very close wire, but the wire in the normal range. The very close wire will not be appeared in all the cameras (RGB or IR).

MartyG-RealSense, 2022-05-05:
It's no problem at all.  :)

A close range electromagnetic field from the wire should not have an effect on the depth image or the depth sensing range.

jianjuan, 2022-05-05:
> It's no problem at all. :)
> 
> A close range electromagnetic field from the wire should not have an effect on the depth image or the depth sensing range.

Nice !!! :)
Thanks a lot ! I will test it in the real scene！

closed 2022-05-05 by jianjuan