realsenseai/librealsense #10475: When two cameras are used, there will be a discrepancy between the image of one camera and the separation of the object.

lopesnb, 2022-05-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

lopesnb, 2022-05-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                    | D435                                       | 
| Firmware Version                | 05.12.07.150                               | 
| Operating System & Version      | Win10                                      | 
| Kernel Version (Linux Only)     |               1                            | 
| Platform                        | PC                                         |
| SDK Version                     | 2.<?>.<?>                                  | 
| Language                        | C++/opencv                                 | 
| Segment			  | others                                     | 

Hai!
I modified exaple's multicam and algin-advance programs to create a program that separates objects by depth. There is no problem when using one camera. When two cameras are used, there will be a discrepancy between the image of one camera and the separation of the object. Do you have any countermeasures?

[image]

MartyG-RealSense, 2022-05-06:
Hi @lopesnb  The only other case I know of regarding performing depth-color alignment with multicam code is at https://github.com/IntelRealSense/librealsense/issues/1833

May I ask whether changing line 56 to use the default depth scale of 0.001 for the 400 Series cameras (except the D405 model) as a hard-coded value instead of retrieving the depth scale from the camera corrects the alignment issue with the second camera?

`depth_scale = 0.001f;`

lopesnb, 2022-05-07:
Hi! MartyG-RealSense

Thank You! 

I changed it, but it doesn't change. Originally, the value read by both cameras was 0.00100000005.

MartyG-RealSense, 2022-05-07:
Could you provide further information please about what the 'discrepancy' is (which part of the image you think is wrong and why).

lopesnb, 2022-05-07:
Hi! MartyG-RealSense
In the camera image on the left, the line separated by the object and the depth is shifted in the X direction
[image]

MartyG-RealSense, 2022-05-07:
Does this offset still occur if you unplug the camera with the brown object so that only the camera with the offset is displayed when you run the program?  If it does then it may be worth performing a recalibration of the camera to eliminate the possibility that the offset is caused by s miscalibration in that particular camera.

lopesnb, 2022-05-07:
When only one camera is connected, no offset will occur in the image.

MartyG-RealSense, 2022-05-07:
Please open the Windows **Task Manager** interface when using two cameras by using the Ctrl-Alt-Delete keyboard shortcut and then go to the 'Performance' category.  In the **CPU** information display, does it show a very high percentage value when two cameras are being accessed?  If it does then this could indicate that the computer is struggling to handle alignment of two cameras simultaneously.

Even if the two cameras work normally in the RealSense Viewer, alignment is a processing intensive operation that is twice the normal burden on the CPU in this particular case with two alignments happening at the same time.

lopesnb, 2022-05-07:
Hi! MartyG-RealSense
Thank you. I will try tomorrow and report the results.

lopesnb, 2022-05-09:
Hi! MartyG-RealSense

I started the task manager.
When the phenomenon occurred, the CPU was 100%.
When I changed the OpenMP setting to no, the CPU became 50%.
However, the phenomenon does not change.
The first recognized D435 will not be offset until it recognizes the second D435, but it will be offset when it recognizes the second.

[image]
[image]

MartyG-RealSense, 2022-05-09:
Hi @lopesnb  I see that you closed this case.  Can you confirm if you no longer require assistance with this case, please?  Thanks!

lopesnb, 2022-05-09:
I accidentally closed it. I need support.

MartyG-RealSense, 2022-05-09:
When BUILD_WITH_OPENMP is set to true, it allows the SDK to use multiple cores of the CPU to process depth-color alignment, with the disadvantage that it can increase the percentage of the CPU that is utilized.

Fluorescent lights such as the ceiling strip lights in your pictures can cause interference to the images output by 400 Series cameras.  As your results are correct when only one camera is used though, this suggests that the ceiling lights are not the cause of your particular problem.

Next, could you try covering over the projector of the other camera to eliminate the possibility that the light emissions from its projector are negatively affecting the view of the second camera that has the offset problem.  Normally, multiple 400 Series cameras do not interfere with each other.  On D435 the projector is the large circle with a green center on the front of the camera.  

Please try putting your finger over this green sensor on the camera that is looking at the brown object to see whether the offset on the other camera image disappears.

lopesnb, 2022-05-09:
Hi! MartyG-RealSense

Thank You!

The two D435s are 90 degrees in different directions.
I try to cover the projector and stop the light emission.

lopesnb, 2022-05-10:
I covered the projector and stopped the light emission. But there was no change.

I tried to find out where in the program the offset occurs.
The offset occurs in the program's aligin.process ().

auto processed = aligin.process (fs);

① No1 No offset
② No2 No offset
③ No1 offset occurs

[image]

[image]

lopesnb, 2022-05-10:
Hi!　MartyG-RealSense

thank you for the advice. Actually, I didn't know what was different, so I tried various things.

lopesnb, 2022-05-11:
Hi!　MartyG-RealSense

thank you for the advice. 
The offset issue has been resolved.
I was wrong with the program.
When importing an image by leaving a profile separately for RealSense
It was solved by importing scale, align_to, align for each profile.

[image]

[multicam_non_offset_error](https://gist.github.com/lopesnb/c5b12b9270c4268bb4a59bde621fae2c)

MartyG-RealSense, 2022-05-11:
It's excellent to hear that you achieved a solution.  Thanks very much for the update and for sharing your solution with the RealSense community!

closed 2022-05-11 by lopesnb