realsenseai/librealsense #10476: Why is there always a difference of 4mm between the depth of the same object in the morning and in the afternoon  ?

AtariaLee, 2022-05-07:
Measurements of the depth of the same object in the morning and afternoon will always be off by about 4mm under almost identical lighting conditions.  I mean the difference between the average depth in the morning and the average depth in the afternoon is 4 mm. Whether it is cloudy or sunny, the depth of each time period is similar.  But the difference between morning and afternoon is always four millimeters .What causes this situation?

 Camera Model :                                   D435I
 Firmware Version  :                            05.12.15.50
 Operating System & Version:              ubunru18.04                                
 Platform :                                             PC  
 SDK Version :                                       2.48.0                     
 Language:                                            C++,opencv                          
 Segment	:		                             Robot                          

Depth measured at noon（938mm）:
[image]

Depth values measured in the morning(942mm):
[image]

MartyG-RealSense, 2022-05-07:
Hi @AtariaLee   If lighting conditions do not appear to be a factor in the completely consistent daily pattern of depth measurement behaviour then the external **temperature** of the environment that the camera is operating in may be an influence (for example, if the location that the camera is operating in is typically warmer at noon than at morning).  This is discussed at https://github.com/IntelRealSense/librealsense/issues/8434 in a case where depth varied by 15 mm over a 12 hour period as the room temperature changed from 25 to 35 degrees C.

If the environmental temperature is consistent over the course of the day then another possibility is that the changing time of day affects how visible the camera's infrared dot pattern projection is.  The dots can become easier or harder for the camera to see during different phases of the day, and the camera uses this for pattern that it casts onto objects to analyze the surfaces for depth information.  

For example, in good lighting conditions the pattern might be visible on some types of material but not others.  And then later in the day when the light level may be dimmer, the behaviours seem reversed - the dots are visible on the materials they were previously invisible on, and invisible on the materials they were earlier visible on.

If your depth variation is related to the dot pattern projection then maximizing the value of the **Laser Power** setting from its default '150' to its '360' maximum may help to improve dot visibility when the pattern becomes more difficult to see, as dot visibility increases as Laser Power increases.

AtariaLee, 2022-05-08:
Hi @MartyG-RealSense ,thanks for your quick reply. Because I need to work in an outdoor environment, the light intensity cannot be completely consistent. But the same thing happens in both high and low light, so I don't think it's the light that's causing the problem.  And I tried to increase the LaserPower from 150 to 360,but it doesn't seem to be working. 

> This is discussed at https://github.com/IntelRealSense/librealsense/issues/8434 in a case where depth varied by 15 mm over a 12 hour period as the room temperature changed from 25 to 35 degrees C.

Sometimes the external temperature of the camera is over 35°, and the difference between morning and noon can be as much as 6° or 7°.  So I think it's probably the temperature that's causing this problem. 
 At #8434 ,I saw that Intel provided a thermal compensation mechanism for the D455. I wonder if similar functionality is now available for the D435I.Or is there a better way to solve this problem?  (The camera is installed at the end of the robot arm, so no cooling device can be added)  .Thanks for your help!

AtariaLee, 2022-05-08:
With the outside temperature around 25 degrees, the camera temperature increased from 35 degrees to 52 degrees in one hour.  Is this normal?  At 52°, does it begin to affect the camera's accuracy in obtaining depth values?  
[image]
After one hour of use:
[image]

AtariaLee, 2022-05-08:
> With the outside temperature around 25 degrees, the camera temperature increased from 35 degrees to 52 degrees in one hour. Is this normal?

When the camera temperature dropped to 35°, I tested the depth value of the same object again, which was still 3~4mm different from the average value in the morning.  This makes me very confused.

MartyG-RealSense, 2022-05-08:
52 degrees for ASIC and 50 for Projector are not good temperatures.  The official recommended maximum for the Projector is 35 degrees, though in real world conditions it can likely go to 42 degrees before problems may begin occurring in the image beyond that temperature.

When the temperature reaches 60 degrees, a safety mechanism in the camera firmware driver will first halve the current Laser Power value, resulting in a depth image with less detail.  If the temperature continues to be 60 or above for several seconds after that then the projector's laser shuts off.  The camera can continue operating but the depth image detail will likely degrade even further.

Intel lab tests found that if the temperature of the outer casing of the camera does not exceed 44 degrees C then the camera's projector temperature should not go beyond 60 degrees.

The camera may also be overheated if there is a hardware problem with the computer USB port that that the camera is attached to or the USB cable has a fault.  There was also a past case where a camera was overheated because it was attached to a dingle-board computing device such as a Raspberry Pi and the power component on the board was overheating.

AtariaLee, 2022-05-08:
I'm not sure if the USB cable is causing this problem.  I'll replace it with a new extension cord to test it out. 

> At#8434 ,I saw that Intel provided a thermal compensation mechanism for the D455.

 I wonder if similar functionality is now available for the D435I.Or is there a better way to solve the problems caused by fever?(The camera is installed at the end of the robot arm, so no cooling device can be added) .Thanks for your help!

MartyG-RealSense, 2022-05-08:
The thermal compensation feature is only on the D455 due to a thermal monitoring mechanism that this camera model has.

Does the robot arm have any heat passing through it that might be being transferred onto the camera through its mounting point?  If so then inserting some kind of insulation material on the mount in-between the arm and the camera might help to reduce the camera's temperature.

Is there anything obstructing the D435i's ventilation grilles on the top and bottom of the camera casing?

Is there any possibility of adding a strong cooling fan to the area directly around the robot arm instead of on the arm itself to try to reduce the camera casing's external temperature and give some protection from natural environmental heat increase? Air should not be blown directly into the camera's ventilation grilles as this could disrupt internal airflow in the casing and make overheating worse.

AtariaLee, 2022-05-08:
I don't think the robot will transfer heat to the camera, because the robot arm is much cooler than the camera.  
And I have an iron bracket on top of the camera that connects the camera to the robot arm.  Is it possible that the heating problem of the camera is caused by the bracket?

MartyG-RealSense, 2022-05-08:
The saying "Like a hot iron" (a heated iron rod, not a clothes iron) makes me think that there may be the possibility that the iron could absorb heat from the camera and retain it next to the camera instead of the heat being able to dissipate away.

AtariaLee, 2022-05-08:
I used a stent like this：
[image]

MartyG-RealSense, 2022-05-08:
There is still the question you raised in https://github.com/IntelRealSense/librealsense/issues/10476#issuecomment-1120374574 of why the depth reading was still 4mm off when the temperature reduced to 35 degrees C.  

If the camera experiences a severe temperature increase event then its imaging sensors may become mis-calibrated and need to be recalibrated with a software tool.

AtariaLee, 2022-05-08:
After I use on_chip Calibration, it is still 4mm error .The error is based on my calibration.

MartyG-RealSense, 2022-05-08:
Does a change of 4mm provide a problem to your robot arm application or is it still capable of carrying out its function successfully?

AtariaLee, 2022-05-10:
Sorry, I was distracted by other things. I tested the distance from the camera to the object using the laser rangefinder and found that the laser rangefinder read 398mm.  However, According to page 68 of the manual:
[image]
The actual distance should be plus 4.2mm. I also used on-chip calibration and Tara calibration to correct the depth values, but that didn't work either.  
The current situation is that the camera was accurate when it started up, the depth value was 942mm, but after a short period of work it began to drop to 938mm

AtariaLee, 2022-05-10:
Now there is a strange problem.  When the robot is at the same height, the depth detected by the camera directly above the object is 949mm.  I obtained the current depth (949mm) by hand-eye calibration of the camera and the robot, and the coordinates converted to the robot coordinate system are accurate.  In addition, I compared the distance between the aruco plate detected by the camera and the distance between the coordinate system of the robot tool and the Aruco plate, and determined that the calibration result was accurate.  
[image]
[image]
But the laser rangefinder still shows 939mm or 938mm, the actual distance should be 942mm or 943mm according to the manual.  I'm not sure which of these two depth values is correct and why this is happening.

MartyG-RealSense, 2022-05-10:
I would recommend forgetting about adding -4.2 (mm) to the depth measurement to find the 'ground truth' measurement at the camera lenses instead of the normal depth measurement taken from the front glass of the camera.  Very few people adjust their RealSense depth measurement for ground truth.

AtariaLee, 2022-05-12:
I'm still trying to stabilize the camera's depth information. Unfortunately, every time I turn the camera on, the depth values are always different and vary greatly. Now, I just opened the camera and measured the same object, and it now shows a different depth value of 895mm.  I really don't know what caused it right now.  SAD

MartyG-RealSense, 2022-05-12:
I went through this case from the beginning in search of fresh insights.  My understanding of the facts are:

*  The correct distance should be 942 mm, measured in the morning.  But by afternoon the measurement has dropped by 4 mm to 938 mm, which is inaccurate.

*  Lighting is mostly identical throughout.  However, there can be an external environmental temperature shift of around 6 or 7 degrees C between morning and afternoon.

*  The camera's operating temperature can become up to 52 degrees C but the measurement error remains even when the camera temperature drops back to 35 degrees C.  Performing on-chip calibration makes no difference.

**************

Is my understanding of the facts correct, please?

I note that you stated "I found that the center point depth values measured directly above the target, above the left, and above the right were also significantly different".

What method is your program using to measure the distance, please?  The significant error that you describe in distance measurement for coordinates that are further out from the center can occur if using depth-color alignment in combination with the **get_distance()** instruction.  Is that your program's approach?

Using Aruco image tags also introduces the possibility for error if something occurs to make the tag more difficult to read, such as the tag being obscured by IR dots from the camera's projector being overlaid on the image.  The dots can become more visible or less visible depending on the time of day, which could affect how distance is read from an Aruco tag depending on the time of day.

I also note that liberealsense 2.48.0 was being used at the start of this case and it has not yet been suggested to try version 2.50.0 to see whether it behaves better.  I can appreciate that you may want to explore other options first though before changing your liberealsense version.

AtariaLee, 2022-05-12:
@MartyG-RealSense  I am sorry for describing the problem so muddled. And yes, Your understanding is correct.

> What method is your program using to measure the distance, please?
I used the camera in the `ros1-Melodic` environment. So I use `OpenCV` to calculate the center point position. In simple terms, it is to find the outer rectangle of the object, and then find the position of the center point of the object in the pixel coordinate system according to the definition of the rectangle.  
And use
`camera_info_sub=nh.subscribe("camera/aligned_depth_to_color/camera_info",5000,&color_depth_node::camerainfoCallback,this);` to obtain the internal reference information of the camera.
And then I convert the point information from pixel coordinates to camera coordinates  
```
            temp.z = 0.001*depth_ptr->image.at<uint16_t>(y,x);    
            temp.x = (x-camera_info.K.at(2))/camera_info.K.at(0)*temp.z; 
            temp.y = (y-camera_info.K.at(5))/camera_info.K.at(4)*temp.z;
```

> Using Aruco image tags also introduces the possibility for error if something occurs to make the tag more difficult to read,
I used the Aruco board in order to use the `Easy Hand-eye` to convert point information to robotic coordinates.  I think the transformation matrix is correct, but I will also try to recalibrate it.  

> I also note that liberealsense 2.48.0 was being used at the start of this case 
If other work does not improve the situation, I will try to use version 2.50.0 of LiberealSense.  But I didn't actually use the software, I simply called the depth value at a point in the camera.  

Thank you so much for your constant help.

MartyG-RealSense, 2022-05-12:
You are very welcome!

When writing a ROS node script for finding distance, it is common for **cv_bridge** to be used, like the script at https://github.com/IntelRealSense/realsense-ros/issues/807#issuecomment-511337064 and the ROS1 wrapper's official Python-based distance finding example at the link below.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/scripts/show_center_depth.py

Does your own OpenCV script use cv_bridge?

AtariaLee, 2022-05-12:
yes, I use CV Bridge for image format conversion. And I wrote the program in C++, but the basic flow is similar to the program in the connection.  

> https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/scripts/show_center_depth.py

AtariaLee, 2022-05-12:
```
  color_sub=it.subscribe("camera/color/image_raw",1,&image_detection::RectCallback,this);
    depth_sub=it.subscribe("camera/aligned_depth_to_color/image_raw",1,&image_detection::DepthCallback,this);
    camerainfo_sub=nh.subscribe("camera/aligned_depth_to_color/camera_info",1,&image_detection::camerainfoCallback,this);
	void camerainfoCallback(const sensor_msgs::CameraInfo &msg)
    {
        cameraInfo=msg;
    }
	//Color image Callback 
    void RectCallback(const sensor_msgs::ImageConstPtr &msg)
    {
        try
        {   
            
            color_ptr=cv_bridge::toCvCopy(msg,"bgr8");
        }
        catch(cv_bridge::Exception &e)
        {
            ROS_ERROR("error:%s",e.what());
            return ;
        }
        
    }
    //Depth image Callback
    void DepthCallback(const sensor_msgs::ImageConstPtr &msg)
    {
        try
        {
           
           depth_ptr = cv_bridge::toCvCopy(msg,"16UC1");
        }
        catch(cv_bridge::Exception &e)
        {
            ROS_ERROR("error:%s",e.what());
        }   
    }
    //Transform
   geometry_msgs::Point pix2cam(cv::Point cv_points)
    {
        geometry_msgs::Point temp_cam_point;
        float x=cv_points.x;
        float y=cv_points.y;
        temp_cam_point.z=0.001*depth_ptr->image.at<uint16_t>(y,x);
        ROS_INFO("z=%f at cam_frame",temp_cam_point.z);
        temp_cam_point.x=(x-cameraInfo.K.at(2))/cameraInfo.K.at(0)*temp_cam_point.z;
        temp_cam_point.y=(y-cameraInfo.K.at(5))/cameraInfo.K.at(4)*temp_cam_point.z;
        return temp_cam_point;
    }
```
This is part of my program for subscribing to image and coordinate transformations.

MartyG-RealSense, 2022-05-12:
I note that you are using ROS1 Melodic.  Can you confirm if you are also using the RealSense ROS wrapper for Melodic?  If you are then another quick test you could perform is to open the **dynamic_reconfigure** graphical interface with the command **rosrun rqt_reconfigure rqt_reconfigure** and set the camera visual preset as **High Accuracy**.  This should provide a confidence check on the depth values that helps to exclude coordinates from the depth image that have doubtful accuracy.

AtariaLee, 2022-05-13:
Yes, i m using the RealSense ROS wrapper for Melodic. And I tried to set it to high Accuracy ,but it doesn't work.
In this case, I used Tara-calibration yesterday to calibrate the depth, which was useful and worked well all day.  But when I re-used it today, it was back to its uncalibrated state.
Then, after waiting for about 30 minutes with the camera working, the measurement results slowly changed back to the calibrated value (with an error of about 4mm).  After waiting another 20 minutes or so, the depth value started to drop by nearly 1cm again, so I had to recalculate (Tara-calibration).  I was really confused.  What causes this?

AtariaLee, 2022-05-15:
I tested it for two days, and it was the same thing.  So I had to recalibrate it at the beginning of each day.

MartyG-RealSense, 2022-05-15:
The section of the calibration white-paper linked to below states that the High Accuracy preset should be set when using tare in order to help to prevent significant error.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-b-incorrect-result

I wonder whether you used High Accuracy before performing tare on the day that it worked well, and then the next day didn't set High Accuracy before doing tare (because the preset had seemed to have no effect) and therefore overwrote the previous day's good tare calibration with an bad one.

AtariaLee, 2022-05-17:
I used High Accuracy before using Tare-calibration.  In addition, this depth change occurred before recalibration the next day. Because of this situation, I re-used the Tare-calibration.  It wasn't a depth error that occurred the next day after recalibration.  Is Tare calibration required every time it is used?

MartyG-RealSense, 2022-05-17:
No, you should not need to repeatedly perform tare calibration to maintain depth measurement accuracy in every session that the camera is used.

AtariaLee, 2022-05-20:
Thank you for your continued help!

MartyG-RealSense, 2022-06-12:
Hi @AtariaLee  Do you require further assistance with this case, please?  Thanks!

AtariaLee, 2022-06-16:
@MartyG-RealSense   I am very sorry for the late reply.  I've been busy with some other work.  I will close the issue and reopen it when I encounter the same problem again. Thank you again for your help!!!!

closed 2022-06-16 by AtariaLee