realsenseai/librealsense #10477: How to do Registration between RGB-D captured point cloud and CAD-modeled point cloud

yiyangd, 2022-05-07:
### Issue Description

I would like to continue the discussion of [#9479](https://github.com/IntelRealSense/librealsense/issues/9479) and[#5082](https://github.com/isl-org/Open3D/issues/5082) 

My goal is to do Registration between the a bottle point cloud captured by D435i RGB-D and the point cloud created by CAD tool with this bottle's actual size (e.g. 28 cm height).

Here is the result I use ICP algorithm **WITHOUT** manually scaling: the **SINGLE BLUE point** is the *RGB-D captured point cloud* and the **yellow point cloud** is the *CAD model*  (this is because the significant large different scale)
[image]

After I scaled manually (increase the RGBD point cloud size by 500, and decrease the CAD point cloud size by 1000), the registration result is still not matched.
[image]

My question is: is it possible to get point cloud information (the distance of the points) so that we could scale down the CAD model by dist_CAD/dist_RGBD rather than put the scaling number manually?

Thanks for helping!

MartyG-RealSense, 2022-05-08:
Hi @yiyangd  Instead of trying to correct the scaling difference by getting the distance of the points, it may be more effective to correct the alignment problem by using Open3D's **Multi-Scale ICP** instead of ordinary ICP.  A tutorial for this can be found at the link below.

http://www.open3d.org/docs/release/tutorial/t_pipelines/t_icp_registration.html#Multi-Scale-ICP-Example

If you are unable to change to Multi-Scale ICP, if you scroll up the tutorial page a little to slightly above the section that I linked to then it suggests that poor ICP convergence due to bad alignment may be compensated for by using a large **max_correspondence_distance**

If you would still prefer to retrieve the distance of the points in Open3D instead of using the methods above then I will be happy to explore that possibility for you.

yiyangd, 2022-05-08:
Thanks @MartyG-RealSense for discussing this problem for one year! I will try the method you shared soon and give the feedback!

closed 2022-05-31 by MartyG-RealSense