realsenseai/librealsense #10478: D405 Failed to enable ros_Wrapper

8umpk1n, 2022-05-08:
ubuntu 18  
ros melodic
realsense d405
Error when I run realsense2_camera node rs_camera：
[ WARN] [1652005345.580662362]: Device 126122270685 is connected using a 2.1 port. Reduced performance is expected.
[FATAL] [1652005345.580839992]: Unsupported device! Product ID: 0x0B5B
[camera/realsense2_camera_manager-2] process has died [pid 25914, exit code 1, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/a/.ros/log/c0d4f87e-ceb8-11ec-b9fc-380025c31919/camera-realsense2_camera_manager-2.log].
log file: /home/a/.ros/log/c0d4f87e-ceb8-11ec-b9fc-380025c31919/camera-realsense2_camera_manager-2*.log

MartyG-RealSense, 2022-05-08:
Hi @8umpk1n   Only the RealSense **ros2_beta** wrapper (the 4.x versions) for ROS2 supports the D405 camera model.  There are no plans to add it to the ROS1 wrapper (Kinetic, Melodic, Noetic).

MartyG-RealSense, 2022-05-14:
Hi @8umpk1n  Bearing in mind the advice provided in the comment above, do you require further assistance with this case please?  Thanks!

closed 2022-05-20 by MartyG-RealSense