realsenseai/librealsense #10479: To measure the length of an object

HyeYumKim, 2022-05-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC |
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 

Hello.
I want to measure the volume of an object using the depth sensor of the realsense camera.
I've seen a lot of examples that are provided in sdk.
What I've looked at is box_dimensioner_multicam in Python and rs-measure in c++.
I have a question here, unlike Python code calibration with chessboard, c++ code could measure the length without chessboard calibration. How do you know that's possible?
Also, is the method of obtaining the depth value a stereo method or a TOF method?
Thank you always.

MartyG-RealSense, 2022-05-09:
Hi @HyeYumKim  Are you referring to the RealSense SDK's **rs-measure** example that measures between two points, please?  

It is possible to use **box_dimensioner_multicam.py** with a single camera without the chessboard by editing its source code.  The purpose of the chessboard is to enable the positions of multiple cameras to be calibrated relative to one another, as you  tested in your past case at https://github.com/IntelRealSense/librealsense/issues/9951

**rs-measure** is a single camera application though and so does not require a chessboard, and neither does the similar built-in tool in the RealSense Viewer's 3D point cloud mode.

A RealSense user created a Python conversion of rs-measure at https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

RealSense 400 Series cameras such as D455 use stereo depth.  The L515 lidar camera's method of obtaining depth is similar to Time of Flight (ToF).

HyeYumKim, 2022-05-09:
Then, in the example of rs-measure, does it measure the length without using algorithms such as rigid transformation when using a single camera?

MartyG-RealSense, 2022-05-09:
rs-measure only measures between two points, whilst the Viewer version of the tool allows selection of more than 2 points for calculating volumetric measurements.

The script notes say that the program is deprotecting 2D pixels to 3D points and then measuring the **Euclidean distance** between two points, as shown in the highlighted code at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/examples/measure/rs-measure.cpp#L286-#L295

HyeYumKim, 2022-05-10:
So when you run a single camera in Python code, the measurement of volume is carried out with the same algorithm as the c++ code?

MartyG-RealSense, 2022-05-10:
Volume measurement is usually not performed by a built-in camera algorithm or by a RealSense SDK instruction, but instead by maths written by the RealSense user in their script.  So in that case, there are likely to be differences in the approaches taken in C++ and in Python to achieve that measurement.

closed 2022-05-22 by MartyG-RealSense