realsenseai/librealsense #10480: Get Frame metadata of IntelRealSense camera with Python

CherifB98, 2022-05-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 }                                                  | 
| Operating System & Version |   { Ubuntu server 20.04 }                        | 
| Platform                                 | { Raspberry Pi 4 }                                   | 
| Language                              |  { Python }                                               | 
----------------------------------------------------------------------------------------------------

### Issue Description

Hello everyone, 

I'm currently working on a project to collect video frames from 3 IntelRealsense cameras (each connected to a Raspberry Pi 4) and send them to a server PC. Timestamps are an important factor in what I am doing in order to have a precise synchronization of the system. For instance, getting the timestamp of each frame is mandatory. 

While I was checking the IntelRealsense site, I found this interesting article : https://dev.intelrealsense.com/docs/frame-metadata . 
But there's not enough information about how to efficiently use this frame metadata object. I tried to look up for documentation about the object/class/functions but have found nothing on the Github. 

So I was wondering if there's anyone who knows more than I do about this tool. I would like to know more about how to use it and the functions of it on order to get a maximum information about each frame (timestamps mainly).

I thank you in advance ;) 

Have a nice day

MartyG-RealSense, 2022-05-09:
Hi @CherifB98  You should be able to access frame metadata if support for hardware metadata is enabled and you use the instruction **get_frame_metadata**.  For example for finding the metadata parameter **frame_laser_power_mode** with Python:

`emitter = rs_frame.get_frame_metadata(rs.frame_metadata_value.frame_laser_power_mode)`

And for getting the **frame_counter** metadata parameter:

`counter = rs.frame.get_frame_metadata(frame, rs.frame_metadata_value.frame_counter)`

From these two examples you can see a pattern for constructing retrieval commands for particular metadata parameters.

1.  Define a descriptive variable name to store the metadata value in, such as 'emitter' or 'counter'.

2.  Set up the instruction for getting the metadata.

`[chosen variable name] = rs.frame.get_frame_metadata(frame, rs.frame_metadata_value.`

3.  After the full stop after **value.** add the name of the metadata parameter that you want to retrieve.

You can read more about this subject at https://github.com/IntelRealSense/librealsense/issues/3179

At https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803 a RealSense team member provides detailed descriptions for a range of metadata parameters.

frame_laser_power_mode

CherifB98, 2022-05-10:
Thank you @MartyG-RealSense  for the answer and for clearing out this issue. 

As I can see in the examples you gave, the frame metadata are restored in and from a 'frame' object. Otherwise, I'm trying to get the metadata from an color-image object. Is there a way to get or store the metadata in a color_image object ?

MartyG-RealSense, 2022-05-10:
A difference between depth and RGB metadata is that when RGB auto-exposure is enabled, RGB exposure metadata is not accessible.  So you would have to set RGB exposure to manual to retrieve exposure metadata.  A RealSense team member advises about this at https://github.com/IntelRealSense/librealsense/issues/2549#issuecomment-431863920

CherifB98, 2022-05-11:
I have succeeded to get frame metadata and have no problem with exposure. My main goal for now is to get the metadata (or at least the timestamps) directly from the color or depth image (not from rs_frame as for the given examples). Is there any chance to get the color or depth image ? Or is there a way to store the frames metadata in the color and depth images ? 
I'm currently working on sending those frames from the Raspberry Pi to a Server PC for further treatment of the images and where the metadata (frames timestamps) are really important to have.
Thank you

MartyG-RealSense, 2022-05-11:
I cannot think of a way that the metadata or timestamps could be embedded into an image.  If you have a Raspberry Pi that you want to connect to a networked computer, Intel's ethernet networking system **rs-server** that was designed with Pi in mind might meet your needs.  It enables data from a RealSense camera attached to a Pi to be accessed on a central host computer by treating the camera as a network device.

https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras

MartyG-RealSense, 2022-05-17:
Hi @CherifB98  Do you require further assistance with this case, please?  Thanks!

CherifB98, 2022-05-17:
Hello @MartyG-RealSense and thank you again for your effort. 
My main goal is to stream/record a video with an IntelRealsense camera and have per frame metadata.
After some long researches these are some potentially possible ways to do it : 
 - Make a text file with the metadata 
 - Use python libraries that potentially can help get per-frame metadata (ffmpeg, exif ...)
 - Use a specific video codec to have per-frame metadata (MKV, CinemaDNG ...)

MartyG-RealSense, 2022-05-17:
I would think that if you wanted to be certain that you were obtaining camera metadata then making a text file export of the metadata would be the most appropriate approach, as the **rs-save-to-disk** C++ SDK example program does by saving "the metadata attributes available for each stream into a comma-separated text file".  It is not saving to a **.txt** file though but instead to a textual format called **csv** that is compatible with spreadsheets and databases.

https://github.com/IntelRealSense/librealsense/blob/master/examples/save-to-disk/rs-save-to-disk.cpp

I have looked before at the subject of extracting metadata to csv in Python in https://github.com/IntelRealSense/librealsense/issues/6536 but did not find an equivalent solution.  I researched again today and still could not find a method, unfotunately.

closed 2022-05-23 by CherifB98