realsenseai/librealsense #10485: Save the depth map and read the depth value of specific pixel position with OpenCV

afenghandsome, 2022-05-10:
hello，I want to ask two questions,
1. We use the align function to align the depth map with the color map, and then use get_ Distance() obtains the depth value from the depth map. Now save the depth map with the type of mat depth_ image(Size(depth_w,depth_h), CV_ 8UC3, (void*)depth_ frame. get_ data(), Mat::AUTO_ STEP); This picture is color. Read this picture with OpenCV. How to read the depth value of this picture.

[image]

3. I use an alignment function written by myself to align the depth map to the color map. The mat type makes 16 bit data, mat depth_ image(Size(depth_w, depth_h), CV_ 16U, (void*)depth_ frame. get_ data(), Mat::AUTO_ STEP); The image is like this. It's gray. I open this image with OpenCV. How can I get the depth value of a specific pixel.
[image]

MartyG-RealSense, 2022-05-10:
Hi @afenghandsome  For your first question about obtaining a depth value from an OpenCV mat, I hope that the script references in the links below will be helpful.

https://github.com/IntelRealSense/librealsense/issues/6239#issuecomment-615249398

https://support.intelrealsense.com/hc/en-us/community/posts/4402828074899-intel-realsense-object-center-pixel-to-coordinate-coverstion

In the second question, if you are using cv_16u in that image then the code suggested by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-412804368 may help you to obtain the depth value.

afenghandsome, 2022-05-10:
@MartyG-RealSense 
Thank you for your reply. I use opencv's imwrite to save the picture to my computer, and then create a new project to load my picture, so I can't use get without turning on the camera_ Distance() function to get the depth value.

Now my idea is to use the method in OpenCV to obtain the depth value from these two images respectively. Do you have any good suggestions?

closed 2022-05-22 by MartyG-RealSense