realsenseai/librealsense #10486: Can not open D405 in ROS melodic.

S1aoXuan, 2022-05-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 5.13.0.50(Latest version&Min version for D405) | 
| Operating System & Version |   Ubuntu 18.04  | 
| Platform                                 | Raspberry Pi 4b |
| Language                                | python                          | 
| Segment			         |  Robot                         | 

### Issue Description
Hi, I am very new to using Realsense and would like some guidance on this problem. I used method 1 to install realsense2_camera that is  sudo apt-get install ros-melodic-realsense2-camera, and then use roslaunch realsense2_camera rs_camera.launch. Then error comes out: Unsupported device! Prodect ID: 0X0B5B
Realsense ROS v2.3.0
Build with libRealSense V2.45.0
Running with  libRealSense V2.45.0
Device USB type 3.2

And I also run these in VM, it reports same error.

MartyG-RealSense, 2022-05-10:
Hi @S1aoXuan  You have not made an error in your installation.  Only the RealSense **ros2_beta** wrapper (the 4.x versions such as 4.0.4) for ROS2 supports the D405 camera model.  There are no plans to add D405 support to the ROS1 wrapper (Kinetic, Melodic, Noetic).

S1aoXuan, 2022-05-10:
> Hi @S1aoXuan You have not made an error in your installation. Only the RealSense **ros2_beta** wrapper (the 4.x versions such as 4.0.4) for ROS2 supports the D405 camera model. There are no plans to add D405 support to the ROS1 wrapper (Kinetic, Melodic, Noetic).

Thanks for replying. So that means for now I can only use D405 in **ROS2**?

MartyG-RealSense, 2022-05-10:
You can only access the RealSense-specific functions provided by the RealSense ROS wrapper with the ROS2 **ros_beta** wrapper.

S1aoXuan, 2022-05-10:
> You can only access the RealSense-specific functions provided by the RealSense ROS wrapper with the ROS2 **ros_beta** wrapper.

Thanks again!

closed 2022-05-10 by S1aoXuan