realsenseai/librealsense #10490: D435i disconnected. Failed to set power state.

zaksel, 2022-05-10:
| Required Info                         |                                                                |
|---------------------------------|-------------------------------------------|
| Camera Model                       |D435i | 
| Firmware Version                   |05.13.00.50| 
| Operating System & Version |Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |5.13.0-40-generic                                         | 
| Platform                                 |PC|
| SDK Version                            |v2.50.0                        | 
| Language                            |Python and ROS                         | 
| Segment			|Robot                          | 

### Issue Description
I have an application using a Realsense D435i The Images are Published on a ROS-Topic with the Wrapper provided [here](https://github.com/IntelRealSense/realsense-ros). For two weeks the app was running without Problems but today I get random disconnects after taking a few Images.
[image]

If the Camera has disconnected it seems to be invisible to the PC. See the attached lsusb screenshot.
[image]
Unplugging and Replugging sometimes helps. Sometimes I need to restart the PC.

dmesg shows the following Output:
[image]
[image]

I have tried different USB-Ports, running on 2.1 and 3.2 Speeds and running the application outside my Docker Container. The disconnects keep coming. Is my Camera damaged?
 
Has someone seen similar behaviour. Does anybody konw a fix for that Problem?
I would really appreciate any help, because I can't figure it out.

Kind regards
Hannes

MartyG-RealSense, 2022-05-10:
Hi @zaksel   Are you setting **initial_reset** to true in your launch, please?  There was a case reported yesterday at https://github.com/IntelRealSense/librealsense/issues/10482 where a RealSense ROS user was using **initial_reset:=true** and the camera was apparently not being re-detected after disconnecting during the reset.  The issue did not occur if initial_reset:=true was not used in the launch instruction.

zaksel, 2022-05-10:
Hi Marty,
initial_reset is not set by me and uses the default value "false".

zaksel, 2022-05-10:
I have tried the usbreset tool from [here](https://github.com/themancalledjakob/usbreset), which did not work in my case.

Then I used the command `rtcwake -m off -s 30` from https://github.com/IntelRealSense/librealsense/issues/5913#issuecomment-590936871. After the boot i was able to see the camera in lsusb. So I started realsense-viewer and tried streaming the color image. I got the following errors:
[image]

zaksel, 2022-05-10:
Tried to disable autosuspend in ubuntu with `echo -1 >/sys/module/usbcore/parameters/autosuspend`. Now the camera stays visible but keeps connecting/disconnecting like crazy and does not provide any Images
[image]

What does "failed to set power state" mean? My Problem seems similar to that described in #5828 but I dont have a multi camera setup

MartyG-RealSense, 2022-05-10:
**Failed to set power state** could cover anything related to being unable to access the camera, from being undetectable due to disconnection, or insufficient power being provided to the camera by the USB port, or the camera being already active and in a busy state (which is unlikely due to the disconnections).

zaksel, 2022-05-10:
Ok thank you for this explanation. It seems to me like the camera can't fully initialize before it gets disconnected. I will check the power delivery tomorrow. At the moment I am not able to access the camera physically...

MartyG-RealSense, 2022-05-10:
Yes, a RealSense camera may be able to be detectable and listable by queries such as **rs-enumerste-devices** but disconnect as soon as a stream is enabled as the camera's power draw on the USB port increases if there is insufficient power provided by the port or a power instability on the port.

zaksel, 2022-05-11:
I've checked the camera connection tody and it seems to be the Cable got Damaged by Bending on the Robot. If the Wrist exceeded a certain angle the connection was lost. I replaced the cable and the Connection seems to be stable.

@MartyG-RealSense Thanks for helping and explaining.

closed 2022-05-11 by zaksel