realsenseai/librealsense #10493: How to solve the noise when using Intel RealSense Viewer to shoot video

L-xn, 2022-05-10:
[image]

| Camera Model                       | {  D455 } | 

| Operating System & Version |   {Win (10)   |

L-xn, 2022-05-11:
Thank you very much!

L-xn, 2022-05-11:
There is a hole-filling Filter inside the Intel RealSense Viewer.However, after each recording, the result of bag is not the result of hole-filling Filter. How can I save the result of hole-filling Filter .

MartyG-RealSense, 2022-05-11:
The effects of post-processing filters are not saved to bag files in the RealSense SDK.  The bag is loaded into the computer's memory and then the filters are applied to the bag data in real-time.

L-xn, 2022-05-11:
> 后处理过滤器的效果不会保存到 RealSense SDK 中的包文件中。袋子被加载到计算机的内存中，然后过滤器实时应用于袋子数据。

I want to perform a series of operations on the filtered depth map, what should I do?

MartyG-RealSense, 2022-05-11:
Does your project have the ability to use ROS and the **RealSense ROS wrapper** and apply the post-processing filters to streams in the wrapper instead of the RealSense Viewer? The **rosbag record** function of ROS can handle bag recording in different ways to the RealSense SDK (such as allowing the saving of aligned data to a bag, which the RealSense SDK's bag recording does not).

MartyG-RealSense, 2022-05-11:
There was a mouse analysis case several years ago where a RealSense team member provided detailed technical advice at https://github.com/IntelRealSense/librealsense/issues/1171#issuecomment-365055104 for improving the image.  A D415 camera was being used in that particular case.  It may be worth trying to apply that advice to your project first before exploring use of ROS.

L-xn, 2022-05-13:
> 几年前在#1171有一个鼠标分析案例，RealSense 团队成员为改善图像提供了详细的技术建议。在这种特殊情况下使用的是 D415 相机。在探索 ROS 的使用之前，首先尝试将该建议应用到您的项目中可能是值得的。

I have carefully read the mouse analysis case you sent, and I have tried to debug the parameters, but the depth map at the bottom of the barrel fluctuates greatly and is not one color. How can I improve the image quality?  
[image]

MartyG-RealSense, 2022-05-14:
There was a past case at https://github.com/IntelRealSense/librealsense/issues/8799 of this fluctuating depth phenomenon when depth-sensing a bucket.

The RealSense user in that case was also performing mouse analysis, as described at https://github.com/IntelRealSense/librealsense/issues/8799#issuecomment-818383037

In that case, I suggested that better light illumination into the bucket might improve results, as the center of the bottom of the bucket was stable but the areas towards the sides of the bucket were not.

Using a **Temporal Filter** post-processing filter and changing its **alpha** value from its default of '0.5' to a lower value such as '0.1' can also reduce the depth fluctuation rate.

L-xn, 2022-05-15:
> [在#8799](https://github.com/IntelRealSense/librealsense/issues/8799)过去有一个案例，即在对桶进行深度感应时出现这种波动的深度现象。
> 
> 在这种情况下，RealSense 用户也在执行鼠标分析，如[#8799（评论）中所述](https://github.com/IntelRealSense/librealsense/issues/8799#issuecomment-818383037)
> 
> 在这种情况下，我建议更好的光照到桶中可能会改善结果，因为桶底部的中心是稳定的，但桶侧面的区域却不是。
> 
> 使用**Temporal Filter**后处理过滤器并将其**alpha**值从默认的“0.5”更改为较低的值，例如“0.1”，也可以降低深度波动率。

Thank you very much for your reply. Any operation in intelrealsense viewer will not change the content in the recorded bag. I will consider the rosbag method you mentioned before.  In addition, whether it can parse the. Bag package through Python, and filter the quality of the picture after parsing.  Like the images I shared above, I can successfully set the bottom of the barrel to the same color in depth Visualization with Visual Preset set to NEAR, I wonder how this can be done in Python.

L-xn, 2022-05-15:
> [在#8799](https://github.com/IntelRealSense/librealsense/issues/8799)过去有一个案例，即在对桶进行深度感应时出现这种波动的深度现象。
>
> 在这种情况下，RealSense [...]

Thank you very much for your reply. Any operation in intelrealsense viewer will not change the content in the recorded bag. I will consider the rosbag method you mentioned before.  In addition, whether it can parse the. Bag package through Python, and filter the quality of the picture after parsing.  Like the images I shared above, I can successfully set the bottom of the barrel to the same color in depth Visualization with Visual Preset set to NEAR, I wonder how this can be done in Python.

MartyG-RealSense, 2022-05-15:
You can set the Near depth visualization color preset in Python with the code below.

```
colorizer = rs.colorizer()
colorizer.set_option(rs.option.visual_preset, 2) # 0=Dynamic, 1=Fixed, 2=Near, 3=Far
```

MartyG-RealSense, 2022-05-16:
If you want to set minimum and maximum distance for the colorizer:

colorizer = rs.colorizer()
colorizer.set_option(rs.option.visual_preset, 2) # 0=Dynamic, 1=Fixed, 2=Near, 3=Far
colorizer.set_option(rs.option.min_distance, value_min)
colorizer.set_option(rs.option.max_distance, value_max)

depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())

https://github.com/IntelRealSense/librealsense/issues/3363 has a discussion that contains a script in the Python language that includes setting the colorizer color scheme with rs.option.color_scheme.

The colorizer color schemes are selected with a number code (Jet is '0'). The full list of numbers is in the link below:

https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_processing.hpp#L764-L773

L-xn, 2022-05-17:
Thank you very much. I have now implemented some operations in The RealSense Viewer using Python, and the results are good.  What can I do to extract the color and depth maps from the .bag.  
[image]

MartyG-RealSense, 2022-05-17:
You can use a RealSense SDK tool called **rs-convert** to extract depth and color frames from a bag file in a range of file formats.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

The Windows version of the SDK has a pre-built .exe version of the tool so that you do not need to build its code.  You can launch it with the following procedure.

1.  Launch the Windows **Command Prompt** interface.  You can do this by typing the word **command** into the text box at the bottom of your Windows screen, next to the corner button where the PC is shut down.

2.  Navigate to the Windows folder location where the pre-built rs-convert.exe program resides and paste a copy of your bag file into it.

**C: > Program Files (x86) > Intel RealSense SDK 2.0 > Tools**

3.  Input the command below to navigate to the **Tools** folder location where the pre-built rs-convert.exe program resides.

**cd C:\Program Files (x86)\Intel RealSense SDK 2.0\tools**

4.  Input a command to launch rs-convert.exe and perform a particular conversion operation on the bag file that was placed in the Tools folder.  For example to extract 
  
For example, to extract into the Tools folder all the depth and color frames from a bag file called test.bag and give the PNG files the filename 'test' plus a number:

**rs-convert.exe -p test -i test.bag**

L-xn, 2022-05-17:
I have tried this method and the resulting color map and depth map are inconsistent in number. And the transformed image is not processed by filtering.  I have now filtered and aligned the bag package in Python.  I want to use Python to extract the image and verify that the color and depth images are not missing.  
[image]

MartyG-RealSense, 2022-05-17:
A RealSense Python user at https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225 took the approach of extracting frames from a bag file and saving color to PNG and depth as an array of scaled matrices in an **.npy** file.

https://github.com/IntelRealSense/librealsense/issues/2731#issuecomment-529938267 instead saves depth and color to video **.avi** files, which could work as well as PNG extraction if the purpose is to visually verify whether frames are missing.  The script is for a live camera but could likely be easily edited to use a bag file as the data source instead.

https://github.com/IntelRealSense/librealsense/issues/2877#issuecomment-446102880 suggests Python code for saving both depth and color frames to PNG.

L-xn, 2022-05-20:
Thank you very much. I successfully extracted the AVI video from bag file.  I found a problem that when extracting depth, I could only use raw or aligned depth frames.  I failed in my attempt to get depth using a depth stream filled with holes.  
       `frames = pipeline.wait_for_frames()
        aligened_frames=align.process(frames)
        aligened_depth_frame=aligened_frames.get_depth_frame()
        color_frame=frames.get_color_frame()

        spatial=rs.spatial_filter()
        spatial_depth=spatial.process(aligened_depth_frame)
        temporal = rs.temporal_filter()
        temporal.set_option(rs.option.filter_smooth_alpha, 0.1)
        temporal.set_option(rs.option.filter_smooth_delta, 20)
        tem_depth = temporal.process(spatial_depth)
        hole_filling = rs.hole_filling_filter()
        filled_depth = hole_filling.process(tem_depth)

        filled_depth_frame = filled_depth.get_depth_frame()
        w=filled_depth_frame.get_width()
        h=filled_depth_frame.get_height()
        dis=round(filled_depth_frame.get_distance(int(w/2),int(h/2))*100,2)`

MartyG-RealSense, 2022-05-20:
In your **filled_depth** definition, please try placing **aligened_depth_frame** in the brackets instead of 'tem_depth'.

**filled_depth = hole_filling.process(aligened_depth_frame)**

L-xn, 2022-05-21:
`align_to = rs.stream.color
    align = rs.align(align_to)

    # Streaming loop
    while True:
        # Get frameset of depth
        frames = pipeline.wait_for_frames()
        aligened_frames=align.process(frames)
        aligened_depth_frame=aligened_frames.get_depth_frame()

        color_frame=frames.get_color_frame()

        spatial=rs.spatial_filter()
        spatial_depth=spatial.process(aligened_depth_frame)

        temporal = rs.temporal_filter()
        temporal.set_option(rs.option.filter_smooth_alpha, 0.1)
        temporal.set_option(rs.option.filter_smooth_delta, 20)
        tem_depth = temporal.process(spatial_depth)

        hole_filling = rs.hole_filling_filter()
        filled_depth = hole_filling.process(aligened_depth_frame)

        
        # Colorize depth frame to jet colormap
        depth_color_frame = colorizer.colorize(filled_depth)

        # Convert depth_frame to numpy array to render image in opencv
        depth_color_image = np.asanyarray(depth_color_frame.get_data())
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_color_image, alpha=0.41), cv2.COLORMAP_JET)
        color_image=np.asanyarray(color_frame.get_data())

        w=filled_depth.get_width()
        h=filled_depth.get_height()

        dis=round(filled_depth.get_distance(int(w/2),int(h/2))*100,2)
        print("the dis are",dis,"cm away")
`
I modified it but it ran wrong.
[image]

MartyG-RealSense, 2022-05-21:
The script may be having a problem with the filled_depth definition:

**filled_depth = hole_filling.process(aligened_depth_frame)**

This instruction may not provide useful values that can be used for 'filled_depth' by w=filled_depth.get_width() and h=filled_depth.get_height() as it is essentially just saying to activate the application of the hole_filling filter to the depth frame.

The link below has a Python script in the section headed **get the depth distance of a point** that provides an example of similarly storing the contents of w and h variables, setting up its equivalent of filled_depth with a **def** instruction.

https://codestudyblog.com/cs2112pyb/1215213230.html

[image]

L-xn, 2022-05-21:
My original depth map was less than ideal, so the depth distance extracted was also less than ideal.My idea is to filter the depth flow, and the filtered depth map is very good.  In this way, it is impossible to extract the filtered depth.  
[image]
[image]

MartyG-RealSense, 2022-05-21:
Is it possible for you to pause the stream with the space bar on a good image and then use the **Snapshot** icon on top of the depth panel of the Viewer to extract that paused frame as csv and raw files that preserve the depth values?  Or does it need to be a moving recording like a bag?

L-xn, 2022-05-26:
The method of using Snapshot was not suitable for my project.I recorded a five-minute video, and I wanted to extract the depth value of the mouse.The depth values of mice were used for behavioral analysis.

MartyG-RealSense, 2022-05-26:
If the problem is that the original depth that is saved to the bag file in the RealSense Viewer is of insufficient quality, please try going to the **Depth Units** option under 'Stereo Module > Controls' and changing the default '0.001' value to **0.01** using the pencil icon next to the option to enter the value with the keyboard manually.  This can result in a better depth image at close range, such as the distance of the camera from the bucket.

 
[image]

MartyG-RealSense, 2022-06-07:
Hi @L-xn  Do you require further assistance with this case, please?  Thanks!

davidmartinez13, 2022-06-09:
hey @L-xn There is a whole [guide](https://dev.intelrealsense.com/docs/depth-post-processing) for improving the depth quality from the realsense documentation, including trade-offs of changing certain parameters. If you would like to try someting more experimental, it is possible to solve this problem by using a denosing autoencoder neural network (Unet) based on this [guide](https://dev.intelrealsense.com/docs/tensorflow-with-intel-realsense-cameras) also prepared by realsense. I have personally tried a similar solution based on deep learning and it seems to work.

L-xn, 2022-06-10:
> 嘿@L-xn有一个完整的[指南](https://dev.intelrealsense.com/docs/depth-post-processing)，用于从 realsense 文档中提高深度质量，包括更改某些参数的权衡。如果您想尝试更多的实验，可以通过使用基于本指南的去噪自动编码器神经网络 (Unet) 来解决这个问题，该[指南](https://dev.intelrealsense.com/docs/tensorflow-with-intel-realsense-cameras)也是由 realsense 编写的。我个人尝试过基于深度学习的类似解决方案，它似乎有效。

Thank you very much, I will try this method.  It may be the only way.

L-xn, 2022-06-10:
I have now recorded a video of the mouse through the IntelRealsense Viewer. The video was saved as BAG.The first thing I need to do is extract the exact distance of the mouse.As discussed above, a series of post-processing operations through the intelrealsense viewer does not change the video content in the BAG. This is a bad situation!Tensorflow is the only way to train depth data.  But I saw that the realsense document needed to use the infrared image, and I did not turn on the red outflow when I took the data. In that case, can we still train?

MartyG-RealSense, 2022-06-10:
It does look as though the RealSense Tensorflow guide depends heavily on infrared images.

You can obtain 3D world depth from a bag by drag and dropping it into the center panel of the RealSense Viewer to play it and then moving the computer mouse cursor over coordinates on the image in real-time to obtain the depth in meters in the corner of the panel.

[image]

You can also use the 'Snapshot' option on top of the Depth panel in the Viewer to export a depth frame from the bag as PNG, csv and raw format files.

You would not be able to obtain infrared frames from the bag if they are not recorded within it though.

L-xn, 2022-06-11:
This's a good way to do it.But I want to extract the depth data of each mouse frame and there are a lot of frames.

MartyG-RealSense, 2022-06-11:
Thanks very much for your patience.  I researched your problem carefully, but the available options are limited if you have to work with a bag file that does not have infrared frames and you are not able to record another bag with the mouse in the bucket.

One possibility might be to use a Python script at https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225 to use the bag as the data source instead of a live camera and export its depth stream values to an **.npy** file.  This file could then be imported into a Python tool such as **Pytorch** for data training purposes.

That script also exports a color stream in the bag, so you would have to edit out the code relating to the color stream if your mouse bag-file does not contain color stream data.

The link below has a tutorial for loading an .npy file into Pytorch.

https://towardsdatascience.com/beginners-guide-to-loading-image-data-with-pytorch-289c60b7afec

If you google for the term **pytorch training** then you can find resources for performing data training and image classification with Pytorch, such as this Microsoft guide for doing so:

https://docs.microsoft.com/en-us/windows/ai/windows-ml/tutorials/pytorch-train-model

L-xn, 2022-06-12:
Thank you very much! I will try to do it.

closed 2022-06-12 by L-xn