realsenseai/librealsense #10494: roslaunch rs_camera.launch fails for a sr300 with ros noetic

rioyoan55, 2022-05-10:
Hi, I'm trying to connect my sr300 camera but when I run roslaunch realsense2_camera rs_camera.launch the following error appears: [ERROR] [1652196634.176027378]: Region-of-interest is not implemented for this device!
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300 
| Firmware Version                   |  3.21
| Operating System & Version |   ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi 4B  |
[image]

| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
            | 

### Issue Description
<Describe your issue / question / feature request / etc..>
[image]

MartyG-RealSense, 2022-05-10:
Hi @rioyoan55   There is an example of the **Region-of-interest is not implemented for this device!** error in the RealSense ROS wrapper with SR300 at https://github.com/IntelRealSense/librealsense/issues/5406

In that case, a RealSense team member advised that "It's the combination of SDK, ROS-RealSense and SR300 firmware version that makes the wrapper think it should support RGB auto-exposure ROI, when in fact it should not".

The RealSense team member who developed the RealSense ROS wrapper advised at a later date that "using librealsense2 version 2.48.0 and realsense-ros version 2.3.1 and an SR305, I can no longer reproduce the Region-of-interest is not implemented for this device! error".

The SR305 was an updated version of the SR300 that was an SR300 on the inside but had a new outer casing.  It is possible to convert an existing SR300 to be recognized as an SR305 by updating the SR300's firmware in the **RealSense Viewer** program.  

Whilst the camera will likely be treated as an SR305 by the RealSense SDK after firmware update, this update may make the camera incompatible with very old SR300-compatible software packages from years ago.

rioyoan55, 2022-05-11:
so it must downgrade firmware version or downgrade librealsense2 version?

MartyG-RealSense, 2022-05-11:
It would likely be a firmware version **upgrade** from your camera's currently installed firmware.  In the RealSense Viewer, you can find the option to update the firmware by going to the **More** option near the top of the Viewer's options side-,panel and selecting the firmware update option from the drop-down menu.

rioyoan55, 2022-05-11:
maybe have link for download update firmware sr300?

rioyoan55, 2022-05-11:
are the newer firmware version is 3.26 ?

rioyoan55, 2022-05-11:
in realsense viewer when click check for update it give message feedback sw/fw version up to date

MartyG-RealSense, 2022-05-11:
Thank you for checking your firmware in the Viewer.

Does the error **Region-of-interest is not implemented for this device!** cause the launch to fail or are you able to continue to use the ROS wrapper normally after it occurs?

MartyG-RealSense, 2022-05-17:
Hi @rioyoan55  Do you require further assistance with this case, please?  Thanks!

rioyoan55, 2022-05-19:
i want to ask maybe have link for install package realsense for open roslaunch realsense_camera realsense_rviz.launch 
because i have an error 

RLException: [realsense_rviz.launch] is neither a launch file in package [realsense_camera] nor is [realsense_camera] a launch file name
The traceback for the exception was written to the log file

rioyoan55, 2022-05-19:
Hi @MartyG-RealSense  i want to ask when i re installing realsense ros i got a problem like this 
Building CXX object realsense_camera/CMakeFiles/realsense_camera_node.dir/src/realsense_camera.cpp.o
/home/ubuntu/catkin_ws/src/realsense_camera/src/realsense_camera.cpp:19:10: fatal error: pcl/ros/conversions.h: No such file or directory
   19 | #include <pcl/ros/conversions.h>
[image]
how to resolve this problem?

MartyG-RealSense, 2022-05-19:
It looks as though you are installing the original, years old **realsense_camera** ROS wrapper that was compatible with SR300 rather than the current **realsense2_camera** wrapper (which is also compatible with SR300).  Is that correct, please?

If you are using the original wrapper then it would have to be paired with the original generation of the librealsense SDK, now known as **Legacy Librealsense** as the wrapper would not work with the current librealsense2 SDK.  That original-generation wrapper would also not be compatible with ROS Noetic.

https://github.com/IntelRealSense/librealsense/tree/v1.12.1

If possible, I would recommend using the current librealsense2 SDK **2.50.0** and ROS wrapper version **2.3.2** with your SR300.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0
https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.2

rioyoan55, 2022-05-20:
i want to ask how to get run , roslaunch realsense_camera realsense_rviz.launch ? because i get error like this when run in terminal
[image]

MartyG-RealSense, 2022-05-20:
I cannot find a realsense_camera launch file called realsense_rviz.launch.  An appropriate launch instruction in that original wrapper for an SR300 would be:

**roslaunch realsense_camera sr300_nodelet_default.launch**

Then after launch has completed, open RViz with the command **rosrun rviz rviz**

closed 2022-06-01 by MartyG-RealSense