realsenseai/librealsense #10495: Camera Matrix of RealSense L515 and D415

harezhh, 2022-05-10:
Hello RealSense-Community,

i need the camera matrix of the realsense L515 and D415. In past month I worked a lot with booth cameras for a project and used the camera to calculate the reald world coordinates of points in the image plane. Now I need the camera matrix values for some theoretical calculations. Unfortunately I dont have accesss to booth cameras anymore. 

It's possible to get the typical camera matrix values provides by `get_intrinsics()`. I need `intr.ppx`, `intr.ppy`, `intr.fx`, `intr.fy`

It would be a very great help to me if someone could provide me with these values.

Thanks.

MartyG-RealSense, 2022-05-11:
Hi @harezhh  I located an example of D415 extrinsics at the link below to get you started.  I should be able to provide an L515 example later this week.

https://support.intelrealsense.com/hc/en-us/community/posts/4406673626259/comments/4406687440915

MartyG-RealSense, 2022-05-14:
As promised, here are examples of intrinsics from an L515 camera.

[image]

[image]

[image]

[image]

[image]

closed 2022-05-26 by MartyG-RealSense