realsenseai/librealsense #10496: pyrealsense2 cannot find serializable_device() function

JeffR1992, 2022-05-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    | 5.4.0-110-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0 | 
| Language                            |  python | 
| Segment			|  N/A | 

### Issue Description
I'm trying to load a custom json parameter file onto my L515 lidar using `pyrealsense2` but am having some issues. I'm following the solutions given in #6902, but whenever I try and call `rs.serializable_device(dev)` I get the following error:

`AttributeError: module 'pyrealsense2' has no attribute 'serialize_device'`

I've updated `librealsense` to the latest version and am still getting this error. In #6902, no one seemed to mention the above `AttributeError`, and doing some Googling hasn't given me any useful info either, which has left me scratching my head. Any help would be appreciated.

MartyG-RealSense, 2022-05-11:
In the line `dev = find_device_that_supports_advanced_mode(config)` in the partial script at https://github.com/IntelRealSense/librealsense/issues/6902#issuecomment-781364057 it is using a **config** reference that is not defined in the code snippet.  Could you try removing config from the brackets, please?

`dev = find_device_that_supports_advanced_mode()`

JeffR1992, 2022-05-12:
I'm using the code that was given by GreatScherzo [here](https://github.com/IntelRealSense/librealsense/issues/6902#issuecomment-793433287) but am unfortunately still getting the same error when trying to call `serialize_device(dev)`:

```
import pyrealsense2 as rs
import numpy as np
import cv2
import json

#%%
DS5_product_ids = ["0AD1", "0AD2", "0AD3", "0AD4", "0AD5", "0AF6", "0AFE", "0AFF", "0B00", "0B01", "0B03", "0B07", "0B3A", "0B5C", "0B64"]

def find_device_json_input_interface() :
    ctx = rs.context()
    ds5_dev = rs.device()
    devices = ctx.query_devices();
    for dev in devices:
        if dev.supports(rs.camera_info.product_id) and str(dev.get_info(rs.camera_info.product_id)) in DS5_product_ids:
            if dev.supports(rs.camera_info.name):
                print("Found device", dev.get_info(rs.camera_info.name))
            return dev
    raise Exception("No product line device that has json input interface")

jsonDict = json.load(open("LCustom.json"))
jsonString= str(jsonDict).replace("'", '\"')

#%%
# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()

try:
    dev = find_device_json_input_interface()
    ser_dev = rs.serializable_device(dev)  
    ser_dev.load_json(jsonString)
    print("loaded json")

except Exception as e:
    print(e)
    pass
#%%
```

MartyG-RealSense, 2022-05-12:
Another way to test preset loading is to load one of the L515's built-in presets instead of a custom JSON file, as applying built-in presets uses different code.  If a built-in configuration also failed to load then this could indicate that the problem is not in your script but elsewhere.  An example of Python code for built-in preset loading is at https://github.com/IntelRealSense/librealsense/issues/9748#issuecomment-927675832

Changing the number '5' in **set_option(rs.option.visual_preset, 5)** to a value lower than that should change the preset that is applied.  The numbers generally correspond to the order that presets are listed in the preset selection drop-down in the RealSense Viewer, with '0' being the first in the list.

JeffR1992, 2022-05-13:
I tried the following which works:

```
pipeline = rs.pipeline()
config = rs.config()
profile = pipeline.start(config)

device = profile.get_device()
depth_sensor = device.query_sensors()[0]
depth_sensor.set_option(rs.option.visual_preset, 0)

...
```

But yes, ideally I'd like to be able to load custom json parameter files onto my L515.

MartyG-RealSense, 2022-05-14:
I researched the subject again carefully but it does indeed appear that if the solutions in https://github.com/IntelRealSense/librealsense/issues/6902 do not work then nothing else will, unfortunately.

Can you confirm whether the json still will not load if you unplug the micro-sized end of the USB cable from the base of the L515, turn the connector around the other way and re-insert it?  USB-C cables are two-way insertion on the micro-sized end, and this camera model sometimes performs better when the connector is in one of the two possible insertion orientations.

closed 2022-05-26 by MartyG-RealSense