realsenseai/librealsense #10497: Inter D435i can't working with other  Plug and play USB  Camera?

Lawrencewzx, 2022-05-11:
hi,all 
I am working on a project where I am using a realsense D435 sensor and another three Plug and play USB  Camera to get data.
**but , Code that can works separately doesn't work together**
@[zivsha](https://github.com/zivsha)
@[RealSense-Customer-Engineering](https://github.com/RealSense-Customer-Engineering)
@[afdaniele](https://github.com/afdaniele)

- PC
- Windows 10
- python 3.8

[image]
btw, D435i data can be collected if the USB camera resource is released.

```
 import pyrealsense2 as rs
 import numpy as np
 import cv2
 def get_image():
     pipeline = rs.pipeline()
     config = rs.config()
     config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
     pipeline.start(config)
    for ii in range(0, pic_num):
        data = pipeline.wait_for_frames()
        color = data.get_color_frame()
        D435I_image = np.asanyarray(color.get_data())
        ret0, img0 = cap0.read()
        ret1, img1 = cap1.read()
        ret2, img2 = cap2.read()
        img0 = cv2.flip(img0, 1)
        img1 = cv2.flip(img1, 1)
        img2 = cv2.flip(img2, 1)
        img0_list.append(img0)
        img1_list.append(img1)
        img2_list.append(img2)
        img3_list.append(D435I_image)
    for ij in range(0, pic_num):
        imgname0 = r"D:\11\img" + str(ij) + r"0.png"
        imgname1 = r"D:\11\img" + str(ij) + r"1.png"
        imgname2 = r"D:\11\img" + str(ij) + r"2.png"
        imgname3 = r"D:\11\img" + str(ij) + r"3.png"
        cv2.imwrite(imgname0, img0_list[ij])
        cv2.imwrite(imgname1, img1_list[ij])
        cv2.imwrite(imgname2, img2_list[ij])
        cv2.imwrite(imgname3, img3_list[ij])
 if __name__ == "__main__":
     cap0 = cv2.VideoCapture(0)
     cap1 = cv2.VideoCapture(1)
     cap2 = cv2.VideoCapture(2)
     cap3 = cv2.VideoCapture(3)
     pic_num = 300
     img0_list = []
     img1_list = []
     img2_list = []
     img3_list = []
     get_image()
     cap0.release()
     cap1.release()
     cap2.release()
```

MartyG-RealSense, 2022-05-11:
Hi @Lawrencewzx   A case at https://github.com/IntelRealSense/librealsense/issues/9325 where the kernel version was found to be responsible for a D435i not working when a USB camera was also used seems to fit your symptoms.

MartyG-RealSense, 2022-05-11:
I am wondering whether this is an issue related to a need to recognize the USB webcams as a **platform camera** (non-RealSense cameras, such as the built-in webcams in laptops).

An example of platform camera detection code is shown in the **realsense_device_manager.py** script at the link below that is part of the RealSense SDK's 'box_dimensioner_multicam' multiple camera example project.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/realsense_device_manager.py#L45

Lawrencewzx, 2022-05-12:
> I am wondering whether this is an issue related to a need to recognize the USB webcams as a **platform camera** (non-RealSense cameras, such as the built-in webcams in laptops).
> 
> An example of code for detecting whether or not a camera is a platform camera is shown in the **realsense_device_manager.py** script at the link below that is part of the RealSense SDK's 'box_dimensioner_multicam' multiple camera example project.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/realsense_device_manager.py#L45

**thank you for your answer** 
According to your solution, part of the problem has been solved.
The loop can be repeated about 15 times, but the same error is reported later 
i'm tying to solve the problem completely, any other suggestions ?

MartyG-RealSense, 2022-05-12:
If a liberealsense Python script stops functioning after 15 frames it is often because an **append** instruction has been used.  The fix that usually works is to store frames in the comouter's memory with the SDK's **Keep()** instruction.  Information about this can be found at https://github.com/IntelRealSense/librealsense/issues/8652

Lawrencewzx, 2022-05-13:
> If a liberealsense Python script stops functioning after 15 frames it is often because an **append** instruction has been used. The fix that usually works is to store frames in the comoputer's memory with the SDK's **Keep()** instruction. Information about this can be found at #8652

**thank you for your suggestion**

closed 2022-05-25 by MartyG-RealSense