realsenseai/librealsense #10499: Intel Realsense D4XX USB2 Recovery Connected

xin-Dream, 2022-05-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |2.0                          | 

Question：

I have a D455 camera that when opened with the Realsense Viewer reads: `notifications. CPP :511 - Intel Realsense D4XX USB2 Recovery Connected`, as shown below
[image]

I tried to upgrade in the Realsense Viewer, but the connection timed out after the upgrade，as shown below
[image]

And then I looked at the other answers,[Intel ResalSense D4xx Recovery #4744](https://github.com/IntelRealSense/librealsense/issues/4744)，So I tried using rs-fw-update.exe  to upgraded firmware,but it did not work . It says timeout
[image]

I looked in device Manager and it showed RS400 in device Manager.
[image]

So could you tell what could I do to solve it?

MartyG-RealSense, 2022-05-12:
Hi @xin-Dream   Your rs-fw-update log's message that update progress is at 100% suggests to me that firmware update completed successfully and it failed to detect the camera when trying to restart it afterwards.

If the camera is still listed as Recovery in the RealSense Viewer after the firmware update, pease first try a **complete** uninstall of the RealSense camera drivers in the Windows Device Manager using the instructions at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4419989666323/comments/4431239847443

If you do not have RealSense **Depth** and **RGB** drivers listed under the 'Cameras' category of the Device Manager then please try uninstalling 'Intel RealSense (TM) RS400 Device' instead using the uninstall method in the instructions.

closed 2022-05-25 by MartyG-RealSense

MartyG-RealSense, 2022-06-20:
Hi @anujpatel1608  The message **Firmware update progress: 100% ... firmware update done** suggest that the firmware update process completed successfully and the camera is just not being detected when being re-initialized after the update has completed.

Is the camera able to be detected successfully in the **RealSense Viewer** tool, please?  If it is, what is the name of the camera that is shown at the top of the Viewer's options side-panel?

anujpatel1608, 2022-06-20:
Actually, I get 98% ,the viewer says it will disconnect & try to connect again. The notification on the right pops up saying it got disconnected and another notification pops up a sec later indicating realsense did  connect again(in recovery). 

In the viewers panel I get 'Intel Realsense D4XX Recovery'

anujpatel1608, 2022-06-20:
Is it ok for serial no. ,firmware no. & usb type to be unknown?

anujpatel1608, 2022-06-20:
Using rs-fw-update tool I got 100% but in the viewer i get 98%.After this showing 100% I did check it still went into recovery

anujpatel1608, 2022-06-20:
I have realsense l515 i'm using cable that I got with that...its a 1m USB-c to USB3.0 Cable

MartyG-RealSense, 2022-06-20:
If you have a D420 caseless camera board and a Vision Processor D4 Board joined together by an 'interposer' rigid board or flexible cable, the recent case at the link below of a firmware update gone bad may be helpful.

https://support.intelrealsense.com/hc/en-us/community/posts/1500000617041-D455-firmware-upgrade-gone-bad-Hardware-reset-?page=2#community_comment_6783126912403

anujpatel1608, 2022-06-20:
yes, i'm using the configuration you mentioned. So the solution on that link suggests that I should disconnect interposer from vision processor board and then perform firmware upgrade by just connecting vision processor board to pc, is that right?

anujpatel1608, 2022-06-21:
Tried that, disconnected camera module from vision board and then connected to pc. When only vision board is connected everything works fine. I click on check for updates and it says all sw up to date and i'm out of recovery too. As soon as i unplug and attach camera module back to Vision Board I again get into recovery mode. Also one error i receive when connecting without camera module(the image i attached)
[image]
 
Also I have a question regarding l515 which we've been using for last 8 months in indoor conditions, the l515 is very sensitive to indoor lightings as well as sunlight from windows. There is lot of noise and this noise gets detected as obstacle and the robot stops completely. I checked and there is a 'ambience' preset  ,but I want to use that preset in ros,so how do i do that? Also is there any other solution for this lighting condition problem?

anujpatel1608, 2022-06-21:
I'm also getting,
'Started Firmware Update process
Recovery device connected, starting update

ERROR: UNKNOWN in rs2_update_firmware_cpp(device:000001B1ADD2C6C0, fw_image:000001B1AE6C1060):
Device:xxxxxxx is locked for update.
Use firmware version higher than: 5.13.0.50'

MartyG-RealSense, 2022-06-21:
Once the depth module is reconnected, are you able to reset the camera to its default factory-new calibration settings in the 'More' setting at the top of the Viewer's options side-panel using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

***************

The L515 model works best indoors in controlled lighting and without the presence of infrared light sources or sunlight.  The L515 can handle the presence of some sunlight though when the **Low Ambient Light** preset is used.  

In ROS you can set presets at launch by loading a **json** camera configuration file.  You can create a json that represents the Low Ambient Light settings by going to the RealSense Viewer, setting the Low Ambient Light preset from the drop-down list of presets and then clicking the json save option to save the configuration to a json.

[image]

You can then load the json in the roslaunch instruction.  For example:

`roslaunch realsense2_camera rs_camera.launch json_file_path:=/path/to/config/DefaultPreset_viewer.json``

Alternatively, you can do so within the launch file using the **json_file_path** parameter to set the location where the ROS wrapper can find the preset.

**ROS1**
https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_camera.launch#L5

**ROS2**
https://github.com/IntelRealSense/realsense-ros/blob/ros2-beta/realsense2_camera/launch/rs_launch.py#L20

A third approach that is available in ROS1 would be to set the L515 preset in the **dynamic reconfigure** settings interface panel after launch has completed.  The dynamic reconfigure interface can be opened with this command:

**rosrun rqt_reconfigure rqt_reconfigure**

[image]

***********

RealSense devices have a limit of 20 firmware downgrades.  This mechanism has been used since the original RealSense camera models in 2014.  After 20 firmware downgrades have been performed, the camera locks downgrades and from that point onwards only allows the installation of a newer firmware version than the currently installed one.  

This lock is permanent for the lifetime of that particular camera.  Although there is not an official reason for the lock, my understanding is that it is to prevent multiple changes to the camera by malicious updates such as 'malware'.

anujpatel1608, 2022-06-27:
[image]
No, write table options is disabled

MartyG-RealSense, 2022-06-27:
In order for the Write Table option to be usable, you need to first enable the **Stereo Module** by left-clicking on the red icon to turn it to blue (On) before opening the Camera Calibration window.

[image]

anujpatel1608, 2022-07-01:
I get this when I try to enable Stereo Module
[image]

MartyG-RealSense, 2022-07-01:
Please click the 'i' information button to display information about the camera and post a picture of it in a comment here please.

anujpatel1608, 2022-07-01:
[image]

anujpatel1608, 2022-07-01:
[image]

MartyG-RealSense, 2022-07-01:
Thank you very much for the image.  I requested it in order to check the camera's serial number because in previous cases of this **invalid value in rs2_open_multiple** error at https://github.com/IntelRealSense/librealsense/issues/7337 and https://github.com/IntelRealSense/librealsense/issues/10093 the camera serial number was incorrectly displayed as 'ffffffffffff'.  The image shows that your camera has the same issue.

The RealSense user in https://github.com/IntelRealSense/librealsense/issues/7337 eventually concluded at https://github.com/IntelRealSense/librealsense/issues/7337#issuecomment-693120103 that their particular problem was caused by the USB cable having been bent.  Is this a possibility for your own situation, please?

anujpatel1608, 2022-07-01:
I think yes, but now I tried with a different usb3 cable which is perfect but error persists.
Also I should mention that when if i plug realsense with camera module attached i go into recovery.So I first plug only vision processor card and after it is detected in viewer correctly I then attach camera module to vision processor card and do whatever is to be done

MartyG-RealSense, 2022-07-01:
I will conduct further research into your symptoms.  Thanks very much for your patience in the meantime!

anujpatel1608, 2022-07-02:
I clicked on Hardware Reset and landed in Recovery Mode. And now same error of realsense locked for update comes up
[image]
ense locked for update.

MartyG-RealSense, 2022-07-02:
If you are updating the firmware in rs-fw-update, can you confirm please whether you have used the '-r' flag to try to recover a camera that is in Recovery mode?  For example when using the downloaded formware bin file 'Signed_Image_UVC_5_13_0_50.bin':

**rs-fw-update -r -f Signed_Image_UVC_5_13_0_50.bin**

anujpatel1608, 2022-07-04:
I did, check this image https://github.com/IntelRealSense/librealsense/issues/10499#issuecomment-1160335754 .
I'll try again & let you know what happens

MartyG-RealSense, 2022-07-04:
Thanks very much, @anujpatel1608 -I look forward to your next update.  Good luck!

anujpatel1608, 2022-07-05:
Now I'm getting this
[image]

MartyG-RealSense, 2022-07-05:
Could you confirm which version of the Vision Processor D4 Board you are using, please?  There are currently 4 versions from V1 to V4, but only the original D4 V1 board is compatible with the D420 depth module.  

This is because V2 is for the D430 module to give it IMU capability, V3 is for the D450 module and V4 is for the D401 module.

MartyG-RealSense, 2022-07-05:
The Mouser product code is for the original V1 board, so that does not appear to be the problem.

I will discuss your situation with my RealSense colleagues.  Thanks again for your patience!

MartyG-RealSense, 2022-07-06:
Do you have a Non Disclosure Agreement (NDA) with Intel, please?  If so then you can submit a support ticket on Intel's **Zendesk** support system.

anujpatel1608, 2022-07-06:
we dont' have NDA. But if we buy from 3rd party sites like mouser(depth module & interposer) & evelta(we brought camera module from evelta.com which is based in India) why would we need to sign an NDA / How do we sign an NDA with the product manufacturer Intel? How do we raise a support ticket urgently as it is causing delay to our deployment?

anujpatel1608, 2022-07-06:
OK,Great, Thank you!

anujpatel1608, 2022-07-07:
anuj@coboticca.com or sudarshan@coboticca.com

MartyG-RealSense, 2022-07-07:
@anujpatel1608  Thanks very much, I have passed your email address to my colleagues.

anujpatel1608, 2022-07-07:
Thanks for the support!

anujpatel1608, 2022-07-29:
Hi @MartyG-RealSense ,We haven't received any communication yet.

MartyG-RealSense, 2022-07-29:
Thanks very much @anujpatel1608 for the update.  I have contacted my Intel RealSense colleagues to follow up on your case.

MartyG-RealSense, 2022-08-18:
Hi again @anujpatel1608  I have been advised that you have signed an NDA agreement.  Please submit a ticket about your issue on the Zendesk support channel.  Thanks again!