realsenseai/librealsense #10501: D435 now frame arrives when using USB hub

GavinGaoYan, 2022-05-12:
Ubuntu 18
direct connect OK.
through usb3.0 hub not OK.

GavinGaoYan, 2022-05-12:
In the viewer will show:
device disconnected.
device connected.
or
dismiss.

GavinGaoYan, 2022-05-12:
may be power problem?

GavinGaoYan, 2022-05-12:
Hub: anker a7522

MartyG-RealSense, 2022-05-12:
Hi @GavinGaoYan  All of the web searches for Anker a7522 state that it has an **ethernet convertor**.  Are you connecting the camera to the hub with a USB 3 cable like the one supplied with the camera or by another connection method, please?

GavinGaoYan, 2022-05-12:
I didn't use the ethernet converter. 
Connect the camera directly using the origin cable to the hub.

MartyG-RealSense, 2022-05-12:
Is the Anker a7522 a mains electricity powered hub that connects to a wall plug socket?  If so then RealSense cameras that are attached to the hub may not function unless the hub is switched on at the mains socket.

GavinGaoYan, 2022-05-12:
OK I see. There may be not enough power supply to the camera.  I will buy a new hub with extra power input.

MartyG-RealSense, 2022-05-12:
If the camera can be detected with the **lsusb** command or with **rs-enumerate-devices** but disconnects almost immediately when streaming is enabled then that would be consistent with insufficient power being supplied to the camera or a power instability on the USB port, yes.

As the camera can function normally when attached directly to the computer (so that it draws off the computer's power supply) then that would suggest that there is an issue with the particular hub that you are using.

closed 2022-05-18 by GavinGaoYan