realsenseai/librealsense #10502: D455 error margin on x,y coordinates

bricemiramont, 2022-05-12:
Hi, I am looking for the error margin for the d455 on the x,y axis. The depth error margin is known but I need a precise localisation of depth points on the y axis. Required precision would be 2 to 3 mm.
Resolution is 1280 x 270

Example:
at 2m with a 1280x720 resolution, I calculated that a pixel represents a 2,97mm x 3,08 mm zone (width x height). 

How accurate are these coordinates ?

thank you,

Brice

MartyG-RealSense, 2022-05-12:
Hi @bricemiramont   There are not estimates available for XY accuracy in the same way that there are for Z-depth.  In general though, you are likely to have less accurate results if you use depth-color alignment with the **align_to** SDK instruction and then deproject with **rs2_deproject_pixel_to_point** to obtain XYZ values for an image.  

Generating a pointcloud with **pc.calculate** and then obtaining its XYZ coordinates with the **points.get_vertices** instruction is likely to provide more accurate results.

These two approaches are compared  at https://github.com/IntelRealSense/librealsense/issues/4315 with advice provided by RealSense team members.

You can also accurately obtain XYZ for a single specific coordinate instead of the entire image by converting a color pixel to a depth pixel with the **rs2_project_color_pixel_to_depth_pixel** instruction, referred to at https://github.com/IntelRealSense/librealsense/issues/9749#issuecomment-917664627

MartyG-RealSense, 2022-05-19:
Hi @bricemiramont  Do you require further assistance with this case, please?  Thanks!

bricemiramont, 2022-05-19:
Hi Marty, no that's ok thank you, I'll use your instructions. thank you,
Brice

closed 2022-05-19 by bricemiramont

MartyG-RealSense, 2022-05-19:
Thanks very much for the update!

5204338, 2023-08-06:
> Hi Marty, no that's ok thank you, I'll use your instructions. thank you, Brice

dear sir , hello, 

> Hi Marty, no that's ok thank you, I'll use your instructions. thank you, Brice

May I ask which scheme you use specifically, and how accurate are the X and Y coordinates of a specific point?