realsenseai/librealsense #10503: Get splitted objects from point cloud

Allius27, 2022-05-12:
Hi there.

Does someone have ready to use algorithm for splitting point cloud from camera (d435/d415) by objects?
In my case camera doesn't see the floor or wall, just different "flying" objects.

MartyG-RealSense, 2022-05-12:
Hi @Allius27  It may be better to look at the reasons why the floor and walls are not appearing and try to include them on the image.  If you could provide an RGB image of the scene that the camera is observing then it would help to diagnose possible causes.

If you are using the RealSense Viewer, please also check whether a post-processing filter called the **Threshold Filter** is enabled and disable it if it is.  This filter sets the maximum observable depth distance to 4 meters by default when enabled and excludes depth coordinates further from the camera than 4 m from the depth image.

Allius27, 2022-05-12:
Hi Marty
Obviously my request was unclear.

I will try to give some details. 
Our robot has a construction when camera is catching objects without any additional objects such as "wall" or "floor".

Example. 
Left - colorized depth
Right - color image
[image]

And question is. 
Where can I find an algorithm which is able to split objects on the cloud of points.
E.g. 
1) got PC from librealsense
2) put PC into split algorithm
3) get list of objects on the PC, like on attached image

closed 2022-05-19 by Allius27