realsenseai/librealsense #10504: CMAKE python wrapper fails with Error 2

scottrpaterson, 2022-05-12:
Camera: D435
OS: Ubuntu 20.04
CMAKE Command: cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=Release

Fails with: 
[  4%] Built target realsense-file
[  4%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/linux/backend-v4l2.cpp.o
[  4%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/linux/backend-hid.cpp.o
[  4%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/third-party/easyloggingpp/src/easylogging++.cc.o
[  4%] Linking CXX shared library pybackend2.cpython-310-x86_64-linux-gnu.so
[  4%] Built target pybackend2
make: *** [Makefile:156: all] Error 2

MartyG-RealSense, 2022-05-12:
Hi @scottrpaterson  Does the make command below enable you to successfully complete the build process, please?

`sudo make uninstall && make clean && make -j$(($(nproc)-1)) && sudo make install`

scottrpaterson, 2022-05-13:
Running that command results in the same error message.

MartyG-RealSense, 2022-05-14:
I could not find a previous case of the '156' type of CMake build error occurring with RealSense.  

Which kernel version are you using, please?

scottrpaterson, 2022-05-15:
I was able to get it this working by using a different Linux kernel - 5.11.10-051110-generic

closed 2022-05-15 by scottrpaterson

MartyG-RealSense, 2022-05-15:
That's great news, @scottrpaterson - thanks very much for the update  :)